mobizt/Firebase-ESP-Client #232: support ESP32-C3 chips?

strid3r21, 2022-01-16:
do you have any plans to support the C3 chips?

mobizt, 2022-01-16:
What issue you've found on that device?
Why do you think it's not supported.

closed 2022-01-16 by mobizt

strid3r21, 2022-01-16:
i cant get the firestore examples to work on my C3-dev module. they work fine on a regular esp32

i get a wall of errors when trying to compile it.

`In file included from C:\Users\ubers\Documents\Arduino\libraries\Adafruit_BusIO\Adafruit_SPIDevice.h:4,
                 from C:\Users\ubers\Documents\Arduino\libraries\Adafruit_BusIO\Adafruit_SPIDevice.cpp:1:
C:\Users\ubers\Documents\Arduino\libraries\Adafruit_BusIO\Adafruit_SPIDevice.cpp: In constructor 'Adafruit_SPIDevice::Adafruit_SPIDevice(int8_t, int8_t, int8_t, int8_t, uint32_t, BusIOBitOrder, uint8_t)':
C:\Users\ubers\Documents\Arduino\hardware\espressif\esp32\cores\esp32/Arduino.h:106:66: error: 'GPIO_OUT1_REG' was not declared in this scope
 #define portOutputRegister(port)    ((volatile uint32_t*)((port)?GPIO_OUT1_REG:GPIO_OUT_REG))
                                                                  ^~~~~~~~~~~~~
C:\Users\ubers\Documents\Arduino\libraries\Adafruit_BusIO\Adafruit_SPIDevice.cpp:53:29: note: in expansion of macro 'portOutputRegister'
   csPort = (BusIO_PortReg *)portOutputRegister(digitalPinToPort(cspin));
                             ^~~~~~~~~~~~~~~~~~
C:\Users\ubers\Documents\Arduino\hardware\espressif\esp32\cores\esp32/Arduino.h:106:66: note: suggested alternative: 'GPIO_OUT_REG'
 #define portOutputRegister(port)    ((volatile uint32_t*)((port)?GPIO_OUT1_REG:GPIO_OUT_REG))
                                                                  ^~~~~~~~~~~~~
C:\Users\ubers\Documents\Arduino\libraries\Adafruit_BusIO\Adafruit_SPIDevice.cpp:53:29: note: in expansion of macro 'portOutputRegister'
   csPort = (BusIO_PortReg *)portOutputRegister(digitalPinToPort(cspin));
                             ^~~~~~~~~~~~~~~~~~
C:\Users\ubers\Documents\Arduino\hardware\espressif\esp32\cores\esp32/Arduino.h:107:66: error: 'GPIO_IN1_REG' was not declared in this scope
 #define portInputRegister(port)     ((volatile uint32_t*)((port)?GPIO_IN1_REG:GPIO_IN_REG))
                                                                  ^~~~~~~~~~~~
C:\Users\ubers\Documents\Arduino\libraries\Adafruit_BusIO\Adafruit_SPIDevice.cpp:60:33: note: in expansion of macro 'portInputRegister'
     misoPort = (BusIO_PortReg *)portInputRegister(digitalPinToPort(misopin));
                                 ^~~~~~~~~~~~~~~~~
C:\Users\ubers\Documents\Arduino\hardware\espressif\esp32\cores\esp32/Arduino.h:107:66: note: suggested alternative: 'GPIO_IN_REG'
 #define portInputRegister(port)     ((volatile uint32_t*)((port)?GPIO_IN1_REG:GPIO_IN_REG))
                                                                  ^~~~~~~~~~~~
C:\Users\ubers\Documents\Arduino\libraries\Adafruit_BusIO\Adafruit_SPIDevice.cpp:60:33: note: in expansion of macro 'portInputRegister'
     misoPort = (BusIO_PortReg *)portInputRegister(digitalPinToPort(misopin));
`

mobizt, 2022-01-17:
I see the error you previous post is related to Adafruit library about the SPI.