mobizt/Firebase-ESP-Client #237: Firebase OTA Update crash

bhupiister, 2022-01-27:
**IDE and its version:**
 - PlatformIO
Board: M5 Core2 

I was testing the OTA update feature and while it downloads the file successfully (upto 98%) it crashes after that.
I feel the crash is related to applying the new firmware. No clue in this domain
Here is the output
```

Download firmware file OTA with Google Cloud Storage JSON API...

Downloading firmware firmware.bin (1629728)
Downloaded 0%
Downloaded 2%
~~~~~~~~~~~
Downloaded 96%
Downloaded 98%
[1124326][E][ssl_client.cpp:36] _handle_error(): [data_to_read():343]: (-76) UNKNOWN ERROR CODE (004C)
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x40121b6d  PS      : 0x00060030  A0      : 0x80123afa  A1      : 0x3ffda7c0
A2      : 0x0018de20  A3      : 0x3ffc5d30  A4      : 0x3f803d8c  A5      : 0x000001ab
A6      : 0x00001000  A7      : 0x001127aa  A8      : 0x80121b6d  A9      : 0x3ffda7a0
A10     : 0x00000000  A11     : 0x00000e20  A12     : 0x000001ab  A13     : 0x3ffd2c44  
A14     : 0x0018d000  A15     : 0x00000e20  SAR     : 0x00000014  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000000  LBEG    : 0x4008ad04  LEND    : 0x4008ad0f  LCOUNT  : 0x00000000

Backtrace:0x40121b6a:0x3ffda7c00x40123af7:0x3ffda9c0 0x40124246:0x3ffdaa70 0x401247cb:0x3ffdaad0 0x400d4c17:0x3ffdaba0 0x400d4c8c:0x3ffdac00 0x400e141a:0x3ffdac20 0x400e14ab:0x3ffdac50 0x40103dcd:0x3ffdac70 0x40105006:0x3ffdac90 0x401050f5:0x3ffdacb0 0x400f0799:0x3ffdace0 0x400f0891:0x3ffdad00 0x400d4634:0x3ffdad20 
  #0  0x40121b6a:0x3ffda7c00 in GG_CloudStorage::handleResponse(FirebaseData*, fb_esp_gcs_req_t*) at .pio/libdeps/m5stack-core2/Firebase Arduino Client Library for ESP8266 and ESP32/src/gcs/GCS.cpp:1866

```
Partition table
```
# Name,   Type, SubType,  Offset,   Size,  Flags
nvs,      data, nvs,      0x9000,  0x4000,
otadata,  data, ota,      0xd000,  0x2000,
phy_init, data, phy,      0xf000,  0x1000,
factory,  app,  factory,  0x10000,  2M,
ota_0,    app,  ota_0,    ,         2M,
ota_1,    app,  ota_1,    ,         2M,
spiffs,   data, spiffs,   ,         0x370000,
```
Can you help in knowing what caused this issue?

mobizt, 2022-01-27:
Your bin file is bug.

The library ignored this space check during begin update and continue to write the update and may crash.

This will reject update and report the error in the next library update.

You should try to adjust the partition scheme in Arduino IDE to increase the app space and minimize the spiffs.

closed 2022-01-27 by mobizt

bhupiister, 2022-01-27:
I tried with a small bin file of 300 kb. It still gives same error. I tried same bin file and updated from SD card and it worked fine.

mobizt, 2022-01-27:
The library was tested for all downloadOTA functions in Google Cloud Storage, Firebase Storage and RTDB. All are ok.

bhupiister, 2022-01-27:
I did update the library and tested the new SDFAT support from you. Thank you for adding. I will try to upload the 1.3 MB file from SD card. If it is successful. I will report back.

bhupiister, 2022-01-27:
So what i tried was to download firmware file separately and store in SD card and update the firmware. 
While downloading file the progress callback has weird output

```
Downloading file firmware.bin (1627888) to /firmware.bin
Downloaded 0%
Downloaded 2%
Downloaded 4%
Download completed
```

File is not downloaded completely and the progress shows as download complete. My upload and everything is working fine, just the download/downloadOTA is not working as it should.

If i manually download 1.6 mb file in SD card and update firmware from it, it works fine.

mobizt, 2022-01-27:
It strange and it never happened in my hundred of times testing.

bhupiister, 2022-01-27:
@mobizt Just for your information, i am using the latest library, this is in the readme section
`Google's Firebase Arduino Client Library for ESP8266 and ESP32 v2.8.3`

mobizt, 2022-01-27:
I test it for ESP8266 too and never found the issue.

bhupiister, 2022-01-28:
I was trying the old library 2.7.7. I feel the issue is at these two lines in both versions of the library

```
Available: 1383
~~~~~~~~~~
Available: 1383
Available: 954
[ 24764][E][ssl_client.cpp:36] _handle_error(): [data_to_read():343]: (-76) UNKNOWN ERROR CODE (004C)
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
```

Backtrace:0x400dcf26:0x3ffb24500x400de99f:0x3ffb2640 0x400def61:0x3ffb26e0 0x400df057:0x3ffb2740 0x400d3576:0x3ffb27d0 0x400e87f1:0x3ffb2820
  #0  0x400dcf26:0x3ffb24500 in GG_CloudStorage::handleResponse(FirebaseData*, fb_esp_gcs_req_t*) at .pio/libdeps/m5stack-core2/Firebase Arduino Client Library for ESP8266 and ESP32/src/gcs/**GCS.cpp:1830**

```
if (fbdo->tcpClient.stream())
{
  available = fbdo->tcpClient.stream()->available();
  Serial.print("Available: "); Serial.println(available);
}
```
When the file is about to end, i get error on this line in stack trace
**available = fbdo->tcpClient.stream()->available();**
When the above line checks for available, it crashes.

mobizt, 2022-01-28:
No, it ever checked with this
if (fbdo->tcpClient.stream())

stream() returns the pointer to Client derived class object which will be NULL (0) when it disposed.

mobizt, 2022-01-28:
Before accessing the memory by the pointer(address), we check the valid address other than 0 before use it.

bhupiister, 2022-01-28:
So i kept a check of nullptr 
```
WiFiClient* cli = fbdo->tcpClient.stream();
cli == nullptr ? Serial.println("yes"):Serial.println("No");
```
It always returns No, that  means it never gets null and thus it still tries to get to this line after file has been fully read.
`available = fbdo->tcpClient.stream()->available();
`

mobizt, 2022-01-28:
I never check for specific NULL (c) or nullptr(C++) in my all programs.
nullptr or NULL is evaluate to false in if conditional checking.

mobizt, 2022-01-28:
The stack trace result is not accurate.

Can you test in Arduino IDE and decode the trace with ESP Exception Decoder which gives more information?

This may inaccurate trace result too and I have to investigate what's wring in the last execution function.

bhupiister, 2022-01-28:
Adding this to the condition has solved the issue for me and file gets downloaded
```
if (fbdo->tcpClient.stream() && payloadRead < response.contentLen)
{
 available = fbdo->tcpClient.stream()->available();
}
```

mobizt, 2022-01-28:
Not really to do that. It checked from  if (available) already.

bhupiister, 2022-01-28:
You may be right but i am saying that adding this check **payloadRead < response.contentLen** has temporarily solved the issue for me.
**fbdo->tcpClient.stream()** does not get 0 at the end of file and for this reason it still tries to read the available data which is not there and thus it crashes. Problem may be something else, this is just what i think. 

```
Serial.print("Payload");Serial.print(payloadRead);Serial.print(" Total:");Serial.println(response.contentLen);
WiFiClient* cli = fbdo->tcpClient.stream();
cli == nullptr ? Serial.println("yes"):Serial.println("No");
if (cli  && payloadRead < response.contentLen)
{
    available = fbdo->tcpClient.stream()->available();
    Serial.print("This gets printed: ");
    Serial.println((int)available);
}
```
I made a check by above code and it was till going inside if condition, which should not happen.

bhupiister, 2022-01-28:
I cannot run this code in Arduino IDE as my setup and library dependency are as per platformIO. Ill have to download everything to make a stack trace in Arduino

mobizt, 2022-01-28:
You may help to reproduce the issue by providing debug, stack traces and information about your device and sdk version instead of pointing to the code.

Nothing is related to file, it is just Client receive buffer.

mobizt, 2022-01-28:
You can open Arduino IDE and open Exception Decoder and browse to the elf file at **.pio/build/your_board/firmware.elf**

bhupiister, 2022-01-28:
Let me do this. Ill open the file in Exception Decoder

bhupiister, 2022-01-28:
in filters i have kept this, do i need to add anything else to this? 
```
monitor_filters = 
                ;log2file
                default
                esp32_exception_decoder
```

mobizt, 2022-01-28:
That PIO options is for filter and are not related to the ESP Exception Decoder.

bhupiister, 2022-01-28:
But ill have to copy the stack trace from somewhere? or it will get if from firmware.elf?

mobizt, 2022-01-28:
It reads elf file directly. just try it.

bhupiister, 2022-01-28:
This is the full Stack trace

```
PC: 0x400dcf51: GG_CloudStorage::handleResponse(FirebaseData*, fb_esp_gcs_req_t*) at .pio/libdeps/m5stack-core2/Firebase Arduino Client Library for ESP8266 and ESP32/src/gcs/GCS.cpp line 1832
EXCVADDR: 0x00000000

Decoding stack results
0x400dcf4e: GG_CloudStorage::handleResponse(FirebaseData*, fb_esp_gcs_req_t*) at .pio/libdeps/m5stack-core2/Firebase Arduino Client Library for ESP8266 and ESP32/src/gcs/GCS.cpp line 1832
0x400de9c7: GG_CloudStorage::gcs_sendRequest(FirebaseData*, fb_esp_gcs_req_t*) at .pio/libdeps/m5stack-core2/Firebase Arduino Client Library for ESP8266 and ESP32/src/gcs/GCS.cpp line 772
0x400def89: GG_CloudStorage::sendRequest(FirebaseData*, fb_esp_gcs_req_t*) at .pio/libdeps/m5stack-core2/Firebase Arduino Client Library for ESP8266 and ESP32/src/gcs/GCS.cpp line 148
0x400df07f: GG_CloudStorage::mDownload(FirebaseData*, char const*, char const*, char const*, fb_esp_mem_storage_type, fb_esp_gcs_get_options_t*) at .pio/libdeps/m5stack-core2/Firebase Arduino Client Library for ESP8266 and ESP32/src/gcs/GCS.cpp line 172
0x400d357a: loop() at .pio/libdeps/m5stack-core2/Firebase Arduino Client Library for ESP8266 and ESP32/src/gcs/GCS.h line 102
0x400e8829: loopTask(void*) at C:/Users/Bhupi/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp line 46
```

bhupiister, 2022-01-28:
One more interesting find. If i just consume `fbdo->tcpClient.stream()`  once before the if condition, it does not go inside if condition at the end of the file. So it gets null when it is read twice at the. Doing this also does not crash my code.
```
WiFiClient* cli = fbdo->tcpClient.stream(); //Just consuming and not using it                                   
if (fbdo->tcpClient.stream())
{
    available = fbdo->tcpClient.stream()->available();
}
```

mobizt, 2022-01-28:
I'll remove all the direct Client pointer access and all can access from the tcpClient wrapper class instead for safer.

bhupiister, 2022-01-28:
Ok, i will wait for update. Till then ill use the workaround.

bhupiister, 2022-01-28:
@mobizt Can you help in identifying this warning? Does google cloud storage has some cap on downloading?

```
Download failed, response payload read timed out due to network issue or too large data size
Download failed, response payload read timed out due to network issue or too large data size
response payload read timed out due to network issue or too large data size
```

mobizt, 2022-01-28:
It's payload response reading timed out.

Due to slow or high traffic or unstable internet connection.

mobizt, 2022-01-28:
The timeout is 10 seconds by default and can be changed via the config
https://github.com/mobizt/Firebase-ESP-Client/blob/a41b2031272baa3995c1cefb4502247a625980c8/examples/RTDB/Basic/Basic.ino#L111-L112

bhupiister, 2022-01-28:
Some more edits. I know this is not the appropriate location. let me know where to put them.
For example, in file MB_FS.h, bool remove function does not have remove option for **MBFS_ESP32_SDFAT_ENABLED**

```
bool remove(const MB_String &filename, mbfs_file_type type)
{
#if defined(MBFS_ESP32_SDFAT_ENABLED)//Added function
                  MBFS_SD_FILE file;
                  return file.remove(filename.c_str());
#endif
}
```

mobizt, 2022-01-28:
The remove function is never used.

I skip this and overwrite or create or append the file for SdFat.

There a issue found from testing, may be incompatibility when using in ESP32.  I can't remove file and check the file existent with sdfat.

bhupiister, 2022-01-28:
The problem i noticed was that instead of file to be overwritten, it was being appended. So 10 kb file was showing 20kb after multiple downloads.

mobizt, 2022-01-28:
You can test delete file or check the existent.

I have not much experiences on this sdfat library and some are not work as my expectation which I try to solve this removing file issue.

And you said true about the same file download will be appended.

mobizt, 2022-01-28:
The esp8266 version of sdfat works for checking file existent and removing.

mobizt, 2022-01-28:
May be changing the file writing mode help but I can't check the old file existed.

bhupiister, 2022-01-28:
I am testing few things. Ill let you know if its successful.

bhupiister, 2022-01-28:
So far what i have read is that if the file is in the root directory then it has to be deleted by `SdFs `

SdFs sd;
```
  // Initialize the SD.
  if (!sd.begin(SD_CONFIG)) {
    sd.initErrorHalt(&Serial);
    return;
  }
  // Remove any existing file.
  if (sd.exists("ReadCsvDemo.csv")) {
    sd.remove("ReadCsvDemo.csv");
  }
```

mobizt, 2022-01-28:
Thanks for this information.

bhupiister, 2022-01-28:
@mobizt I would really like to thank you for your time today and for this library. I have successfully implemented what i wanted.