mobizt/Firebase-ESP-Client #248: Is there any Ethernet Connectivity?

GauravMall, 2022-02-02:
I'm trying to create a device with an Arduino and ENC28J60 module. Is there a way to get the ethernet connection and use it for the firebase library instead of the wifi? If the code is simple, please post it, as I have no idea how to do it plus I'm pretty new to ethernet shields on Arduino. If it can't be done, please recommend me another solution. Is it even possible?

mobizt, 2022-02-02:
Please look at examples.
https://github.com/mobizt/Firebase-ESP-Client/tree/main/examples/RTDB/BasicEthernet

closed 2022-02-02 by mobizt

mobizt, 2022-02-02:
The library is only for ESP8266 and ESP32 devices. 
https://github.com/mobizt/Firebase-ESP-Client#unsupported-at-command-and-mobile-modem-bridge

GauravMall, 2022-02-02:
Thank you for your response. So to confirm I can't really do anything if I have just my Arduino Nano and an ENC28J60 module connected to it? I must have the listed devices above?

mobizt, 2022-02-02:
AVR is not suitable to use in this field of application (database) due to too little memory and flash space.

AVR is for signal and sensor application only.

GauravMall, 2022-02-02:
Basically, I want to read some data from a sensor and send it to Firebase through the ethernet. How can I do it?

Sulomus, 2022-02-02:
> Basically, I want to read some data from a sensor and send it to Firebase through the ethernet. How can I do it? (Sorry, If I'm being stubborn, I'm new to this so I don't really understand some things well)

You just can't do it with your Arduino setup right now.

Buy ESP32 and connect it to Arduino via serial (if you really need Arduino) and use ESP32 as a IoT device which will connect to Firebase using this library :)

GauravMall, 2022-02-02:
Thank you for your response! Okay, will try it out.