mobizt/Firebase-ESP-Client #250: FireSense Addon folder contains functions in header files

Erik-Koning, 2022-02-04:
The bug:
Specifically Firebase-ESP-Client-main/src/addons/FireSense/TokenHelper.h causes the builds to fail becuase of repetitive definitions of functions since they are defined in a header file.

**To Reproduce**
Build like expected with esp-idf and arduino core for esp32

**Expected behavior**
A successful build

**IDE and its version:**
 - ESP-IDF 4.4 w/ arduino core as component
 
 Ill submit a pull request to resolve

mobizt, 2022-02-04:
There is no file "src/addons/FireSense/TokenHelper.h"

It is at "src/addons/TokenHelper.h" instead.

Do you accidently move it in your IDE?

The library compilation is ok.

mobizt, 2022-02-04:
The compilation in PlatformIO is ok using Github Action too.

closed 2022-02-04 by mobizt

mobizt, 2022-02-04:
The addons are not parts of library. 

It is optional for demonstration. User can use library without it and it can include in his sketch as in example as helper functions.

Erik-Koning, 2022-02-04:
For my case building with ESP-IDF, and this library as a component it fails becuase the TokenHelper.h file has function definitions in it.

For those experiencing this it is fixed by making a new TokenHelper.cpp file and placing the definitions there.

mobizt, 2022-02-04:
As I told you that file is not part of library. Only use in examples, user can go to that file and copy or see what is it.

For header file, it can be include functions in it and not need source files.

mobizt, 2022-02-04:
There are many header files that contain functions in this library.

mobizt, 2022-02-04:
I will define all functions in that header file as static to avoid multiple definitions error.