mobizt/Firebase-ESP-Client #265: RTDB - Multipath stream not working

kumbhat10, 2022-02-28:
**Just ask something**
If you have questions, use [Discussions](https://github.com/mobizt/Firebase-ESP-Client/discussions) instead. 

**Describe the bug**
In the below code, Multipath streaming doesn't start as it used to before. I was using multipath stream before and it was working fine but suddenly stopped working now. I don't remember which previous version of Firebase-ESP-Client I was using

**To Reproduce**
#include <WiFi.h>
#include <Firebase_ESP_Client.h>
#include <addons/TokenHelper.h>
#include <addons/RTDBHelper.h>

#define WIFI_SSID "xxxxxxxx"
#define WIFI_PASSWORD "xxxxxxxxx"
#define API_KEY "xxxxxxxxxsl4hxxxxxxxxxI"
#define DATABASE_URL "https://xxx-xxx-default-rtdb.europe-west1.firebasedatabase.app"
#define USER_EMAIL "xxxxx@gmail.com"
#define USER_PASSWORD "xxxxx"

FirebaseData fbdo;
FirebaseData stream;
FirebaseAuth auth;
FirebaseConfig config;
unsigned long sendDataPrevMillis = 0;

String parentPath = "/test/stream/data";
String childPath[2] = {"/node1", "/node2"};

int count = 0;
volatile bool dataChanged = false;

void streamCallback(MultiPathStream stream)
{
  size_t numChild = sizeof(childPath) / sizeof(childPath[0]);
  for (size_t i = 0; i < numChild; i++)
  {
    if (stream.get(childPath[i]))
    {
      Serial.printf("path: %s, event: %s, type: %s, value: %s%s", stream.dataPath.c_str(), stream.eventType.c_str(), stream.type.c_str(), stream.value.c_str(), i < numChild - 1 ? "\n" : "");
    }
  }
  Serial.println();
  Serial.printf("Received stream payload size: %d (Max. %d)\n\n", stream.payloadLength(), stream.maxPayloadLength());
  dataChanged = true;
}

void streamTimeoutCallback(bool timeout)
{
  if (timeout)
    Serial.println("stream timed out, resuming...\n");
  if (!stream.httpConnected())
    Serial.printf("error code: %d, reason: %s\n\n", stream.httpCode(), stream.errorReason().c_str());
}

void setup()
{
  Serial.begin(115200);
  WiFi.begin(WIFI_SSID, WIFI_PASSWORD);
  Serial.print("Connecting to Wi-Fi");
  while (WiFi.status() != WL_CONNECTED)
  {
    Serial.print(".");
    delay(300);
  }
  Serial.println();  Serial.print("Connected with IP: ");
  Serial.println(WiFi.localIP());  Serial.println();
  Serial.printf("Firebase Client v%s\n\n", FIREBASE_CLIENT_VERSION);

  config.api_key = API_KEY;
  auth.user.email = USER_EMAIL;  auth.user.password = USER_PASSWORD;
[... 111 lines cut ...]
11:34:16.930 -> [V][ssl_client.cpp:197] start_ssl_client(): Setting hostname for TLS session...
11:34:16.977 -> [V][ssl_client.cpp:212] start_ssl_client(): Performing the SSL/TLS handshake...
11:34:17.937 -> [V][ssl_client.cpp:233] start_ssl_client(): Verifying peer X.509 certificate...
11:34:17.937 -> [V][ssl_client.cpp:242] start_ssl_client(): Certificate verified.
11:34:17.937 -> [V][ssl_client.cpp:257] start_ssl_client(): Free internal heap after TLS 151692
11:34:17.937 -> [V][ssl_client.cpp:295] send_ssl_data(): Writing HTTP request with 32 bytes...
11:34:17.937 -> [V][ssl_client.cpp:295] send_ssl_data(): Writing HTTP request with 902 bytes...
11:34:17.937 -> [V][ssl_client.cpp:295] send_ssl_data(): Writing HTTP request with 256 bytes...
11:34:17.937 -> [V][ssl_client.cpp:295] send_ssl_data(): Writing HTTP request with 61 bytes...
11:34:17.985 -> [V][ssl_client.cpp:295] send_ssl_data(): Writing HTTP request with 32 bytes...
11:34:17.985 -> [V][ssl_client.cpp:295] send_ssl_data(): Writing HTTP request with 902 bytes...
11:34:17.985 -> [V][ssl_client.cpp:295] send_ssl_data(): Writing HTTP request with 256 bytes...
11:34:17.985 -> [V][ssl_client.cpp:295] send_ssl_data(): Writing HTTP request with 61 bytes...
11:34:17.985 -> JSON set ok
11:34:17.985 -> 
11:34:31.877 -> Set json...
11:34:31.877 -> [V][ssl_client.cpp:295] send_ssl_data(): Writing HTTP request with 32 bytes...
11:34:31.877 -> [V][ssl_client.cpp:295] send_ssl_data(): Writing HTTP request with 902 bytes...
11:34:31.877 -> [V][ssl_client.cpp:295] send_ssl_data(): Writing HTTP request with 256 bytes...
11:34:31.877 -> [V][ssl_client.cpp:295] send_ssl_data(): Writing HTTP request with 61 bytes...
11:34:31.877 -> [V][ssl_client.cpp:295] send_ssl_data(): Writing HTTP request with 32 bytes...
11:34:31.920 -> [V][ssl_client.cpp:295] send_ssl_data(): Writing HTTP request with 902 bytes...
11:34:31.920 -> [V][ssl_client.cpp:295] send_ssl_data(): Writing HTTP request with 256 bytes...
11:34:31.920 -> [V][ssl_client.cpp:295] send_ssl_data(): Writing HTTP request with 61 bytes...
11:34:31.920 -> JSON set ok

kumbhat10, 2022-02-28:
I have tried using https://github.com/mobizt/Firebase-ESP32 library also with provided multipath and  callback example.  Multipath example doesn't work but callback example works like a charm

mobizt, 2022-02-28:
Thanks for reporting this bug.

The library update [v3.1.3](https://github.com/mobizt/Firebase-ESP-Client/releases/tag/v3.1.3) is available with this bug fixed.

closed 2022-02-28 by mobizt

kumbhat10, 2022-02-28:
I have tested with v3.1.3 and it is working again. Thanks for fixing this issue. However I had to manually download the .zip code and then use it. Any idea how long will it take for Arduino library manager to show update available. As I am having CI/CD pipeline, and want to build and update firmware of all my ESP32's automatically