mobizt/Firebase-ESP-Client #289: ENHANCEMENT

amitrao17, 2022-04-07:
The config.time_zone parameter is not documented; it would be helpful if it was.

The parameter takes a float with unit hours which is fine, but the official Arduino libraries take a quantity in seconds, so my suggestion is for the Firebase-ESP-Client and Firebase-ESP8266 libraries to also use the unit of seconds:

https://github.com/arduino-libraries/NTPClient

> // You can specify the time server pool and the offset, (in seconds)
> // additionally you can specify the update interval (in milliseconds).
> // NTPClient timeClient(ntpUDP, "europe.pool.ntp.org", 3600, 60000);

mobizt, 2022-04-07:
As I said time zone is hour in the standard terminology.
The example and its comment are enough for how to use this config.

amitrao17, 2022-04-07:
ok, sure leave it in hours if that is your view.

the problem is still that there is no documentation about the config.time_zone parameter unless I'm missing it?

closed 2022-04-07 by mobizt

amitrao17, 2022-04-07:
So, no documentation is fine?