mobizt/Firebase-ESP-Client #300: Device lost connection after few hours

rvbc1, 2022-04-25:
**Describe the bug**
Now I'm using version 2.4.1, and everything works fine, but after update to  v3.1.7 my devices permanent  lost connection after 2-3 hours. 

**To Reproduce**
[FreeRTOS task]

Firebase.begin(&config, &auth);

Firebase.reconnectWiFi(true);

firebaseDataInputStream.setResponseSize(INPUT_STREAM_RESPONSE_SIZE);
firebaseDataOutputStream.setResponseSize(OUTPUT_STREAM_RESPONSE_SIZE);

if(Firebase.RTDB.beginStream(&firebaseDataInputStream, getInNode().c_str())){
  while(true){
       if (!Firebase.RTDB.readStream(&firebaseDataInputStream)) {
        if (firebaseDataInputStream.streamAvailable()) {
            if (firebaseDataInputStream.dataType() == "string") {
                parseData(firebaseDataInputStream.stringData());
            }
        }
    }
   delay(100);
  }
}

**Expected behavior**
Firebase stream will be working all the time. 

**IDE and its version:**
 PlatformIO

**ESP8266 Arduino Core SDK version**
 platform = espressif32 @ 3.5.0

mobizt, 2022-04-25:
Did you see this
https://github.com/mobizt/Firebase-ESP-Client/blob/f808c1d57257ef8ad91cedd06d46a64d73b125f4/examples/RTDB/DataChangesListener/Callback/Callback.ino#L168-L170

closed 2022-04-25 by mobizt

mobizt, 2022-04-25:
Call Firebase.ready() function repeatedly to handle authentication tasks.

rvbc1, 2022-04-25:
I modify the loop to: (But on version 2.4.1 previous loop worked fine) 
    if (Firebase.ready()) {
        if (Firebase.RTDB.readStream(&firebaseDataInputStream)) {
            if (firebaseDataInputStream.streamAvailable()) {
                if (firebaseDataInputStream.dataType() == "string") {
                    parseData(firebaseDataInputStream.stringData());
                }
            }
        }

        if (firebaseDataInputStream.streamTimeout()) {
        }
    }
    
But after 3 hours my devices stops responding and from Firebase cli (https://firebase.google.com/docs/cli) I get:

{"name":"listener-listen","timestamp":1650884463248,"client":{"remoteAddress":{"address":"89.171.50.181"},"userAgent":{"browser":"unknown","os":"unknown","platform":"unknown","userAgentString":"ESP","version":"unknown"}},"path":["a3698ff9-55a4-4339-b805-665945786185","58:BF:25:EF:50:B0","in"],"unIndexed":false,"pendingTime":1,"millis":0,"bytes":0,"allowed":false}

So device is connected to Firebase but getting 0 bytes. It makes big Firebase data transfer (about 25Mb/h, on version 2.4.1 device make these transfer in month). 

It looks that device sends empty request to the Firebase.

mobizt, 2022-04-25:
Based on your code post above that's the improper usage of library. Please follow the examples.

mobizt, 2022-04-25:
The latest version of library is 3.2.0. You should update and follow the [examples](https://github.com/mobizt/Firebase-ESP-Client/tree/main/examples/RTDB).