mobizt/Firebase-ESP-Client #321: Partial document being fetched from firestore

Ramanujam-RS, 2022-05-21:
**Description**
While fetching document from firestore using getDocument smaller documents are read fine but when the document size is bit large complete JSON is not returned from fdao.payload() on ESP32D. Appreciate the help!

**To Reproduce**
void getDoc() {
  if (Firebase.ready()){
    String documentPath = "devices/device1";
    if (Firebase.Firestore.getDocument(&fbdo, FIREBASE_PROJECT_ID, "", documentPath.c_str(), "")) {
      Serial.print("Expected payload size ");
      Serial.println(fbdo.payloadLength());
      Serial.print("Actual payload size ");
      Serial.println(strlen(fbdo.payload().c_str()));
      Serial.printf("ok\n%s\n\n", fbdo.payload().c_str());
      Serial.println("End of payload");
    } else
      Serial.println(fbdo.errorReason());
  }
}

**Expected behavior**
02:10:15.642 -> Expected payload size 3715
02:10:15.642 -> Actual payload size 2039

**Screenshots**
[image]

**IDE and its version:**
 - Arduino - 1.8.15, PlatformIO - Ubuntu 20
 - Library - 3.3.5

**ESP32 Arduino Core SDK version**
 - v4.4-beta1-189-ga79dc75f0a

mobizt, 2022-05-22:
You should increase the response size limit, by calling [fbdo.setResponseSize](https://github.com/mobizt/Firebase-ESP-Client/tree/main/src#set-the-http-response-size-limit).

closed 2022-05-22 by mobizt

Ramanujam-RS, 2022-05-22:
@mobizt wonderful, works like a charm. Thanks for the timely response!