mobizt/Firebase-ESP-Client #336: PlatformIO install issue

41linea45, 2022-06-16:
Hi

I tried added to platformio firebase library but i cant build
Always gives error in Arduino framework. ESPressif 4.0.4 framework
platform ini:

`[env:esp32dev]
platform = espressif32
board = esp32dev
framework = arduino
lib_deps = mobizt/Firebase ESP32 Client@^4.0.0

monitor_speed = 115200`
error:
` #include <SPIFFS.h>
          ^~~~~~~~~~
compilation terminated.
*** [.pio\build\esp32dev\lib468\Firebase ESP32 Client\signer\Signer.cpp.o] Error 1
*** [.pio\build\esp32dev\lib468\Firebase ESP32 Client\rtdb\QueryFilter.cpp.o] Error 1
In file included from .pio/libdeps/esp32dev/Firebase ESP32 Client/src/session/FB_Session.cpp:33:
.pio/libdeps/esp32dev/Firebase ESP32 Client/src/FirebaseFS.h:26:10: fatal error: SPIFFS.h: No such file or directory

****************************************************************
* Looking for SPIFFS.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:SPIFFS.h"
* Web  > https://registry.platformio.org/search?q=header:SPIFFS.h
*
****************************************************************

 #include <SPIFFS.h>
          ^~~~~~~~~~
compilation terminated.
*** [.pio\build\esp32dev\lib468\Firebase ESP32 Client\session\FB_Session.cpp.o] Error 1
In file included from .pio/libdeps/esp32dev/Firebase ESP32 Client/src/FB_Net.h:10,
                 from .pio/libdeps/esp32dev/Firebase ESP32 Client/src/wcs/esp32/FB_TCP_Client.h:36,
                 from .pio/libdeps/esp32dev/Firebase ESP32 Client/src/wcs/esp32/FB_TCP_Client.cpp:36:
.pio/libdeps/esp32dev/Firebase ESP32 Client/src/FirebaseFS.h:26:10: fatal error: SPIFFS.h: No such file or directory

****************************************************************
* Looking for SPIFFS.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:SPIFFS.h"
* Web  > https://registry.platformio.org/search?q=header:SPIFFS.h
*
****************************************************************

 #include <SPIFFS.h>
          ^~~~~~~~~~
compilation terminated.
*** [.pio\build\esp32dev\lib468\Firebase ESP32 Client\wcs\esp32\FB_TCP_Client.cpp.o] Error 1
`

maxgerhardt, 2022-06-16:
(Cross posted to https://community.platformio.org/t/firebase-library-add-issue-arduino-mobitz/28369)

mobizt, 2022-06-17:
This library includes `SPIFFS.h` since its first publish

This library developed in PlatformIO and I test for both VSCode IDE and Arduino IDE.

The problem is your esp32 PlatformIO core sdk installation.

If you're familiar with ESP32 or ESP8266 Arduino, you should know that `SPIFFS.h` comes with the core sdk installation.

Missing the core library headers files means you have error in core sdk installation.

Adding compiler flags for library dependency finder e.g. `lib_ldf_mode` can't help.

The header file `SPIFFS.h` is in the core library at 
`C:\Users\UserAccount\.platformio\packages\framework-arduinoespressif32\libraries\SPIFFS\src\SPIFFS.h`

You should delete `C:\Users\UserAccount\.platformio\packages\framework-arduinoespressif32` or `C:\Users\UserAccount\.platformio\packages` and compile the project again.

closed 2022-06-17 by mobizt