mobizt/Firebase-ESP-Client #341: Code Feasibility for FCM Message Receiving

kaiavi, 2022-06-28:
Is the code given in this repo suffiecient to "receive" notifications from FCM server? 
If I want to receive notifications from the server, can I achieve it?

And something a little bit out of the concept, from your experience, can we integrate FCM to Linux such that we can write a client that will be able to receive notifications from the sever?

Thanks for this project and support.

mobizt, 2022-06-28:
There are examples for FCM in this library.

Which this library us FCM admin REST API to work for the Google server (as a backend) to process the notification message.

I don't understand what you mentioned about Linux server. If you mean use Linux server as a FCM admin, you can use it to send http request via REST API to Google FCM server using your code runs on your server

closed 2022-06-28 by mobizt

kaiavi, 2022-06-28:
My question is, ideally, an Android device receives push notifications from the server using FCM. Can this library be used to achieve the same in Arduino?

mobizt, 2022-06-28:
https://github.com/mobizt/Firebase-ESP-Client/blob/6a5d44d605bc2553d37c1b0691ed5a0de5766ff3/examples/Messaging/Send/HTTPv1/SendMessage/SendMessage.ino#L15-L18