mobizt/Firebase-ESP-Client #342: Sometimes i get this exception (28)

Gcii, 2022-06-30:
## Version library:
 firebase arduino client library for esp8266: 4.0.0 and also the exception ocurrs in 3.3.7 . 
Device manufacturer / model: NodeMCU 1.0 (ESP-12E Module)
## IDE and its version:
 - Arduino, PlatformIO
 - Version 1.8.19

## To reproduce
Steps to reproduce the behavior:

I use this code:

```
//WIFI
#include<ESP8266WiFi.h>
//FIREBASE
#include <Firebase_ESP_Client.h>
#include <addons/TokenHelper.h>
//Obtener la hora de NTP
#include <NTPClient.h>
#include <WiFiUdp.h>
//Firebase credentials
#define API_KEY "ddd"
#define FIREBASE_PROJECT_ID "ddd"
#define USER_EMAIL "ddd"
#define USER_PASSWORD "ddd"
///FireBase data
FirebaseData fbdo;
FirebaseAuth auth;
FirebaseConfig config;
//Obtener la hora
WiFiUDP ntpUDP;
NTPClient timeClient(ntpUDP, "pool.ntp.org");
// WIFI
String sSsid = "ddd";
String sPasswordWifi = "ddd";
byte bCounterWifi = 0;
byte bMaxAttempts = 50;
bool _bThereIsDocument = false;
bool bWiFi = false;
// TIEMPO
unsigned long dataMillis = 0;
float lMillisTime = 0;
float lMillisTimeCopy = 0;
float lMillisTimePerPower = 0;
String sCurrentDate = "";
byte byCurrentHour = 0;
byte byCurrentMinutes = 0;
byte byMinuteDifference = 0;
int iYear = 0;
byte byMonth = 0;
byte byDay = 0;
// FIREBASE
String documentPath = "DDD/ddd/ddd/ddd";
bool bTimeIsUpToCheck = false;
bool bSendDataToServer = false;
//PARA DEBUG
bool bDataChangedInRead = false;
float fRam = 0;
float iTime[1300];
float iEne[1300];
void setup() {
  bTimeIsUpToCheck = false;
  bDataChangedInRead = false;
  bSendDataToServer = false;
  _bThereIsDocument = false;
  Serial.begin(115200);
  for (int i = 0; i < 1300; i++) {
    iTime[i] = 0.0;
    iEne[i] = 0.0;
[... 290 lines cut ...]
Decoding stack results
0x40208e30: FB_TCP_Client_Base::readLine(char*, int) at c:\users\galarzaa\documents\arduino\libraries\firebase_arduino_client_library_for_esp8266_and_esp32\src\wcs\base/fb_tcp_client_base.h line 289
0x4020825e: MB_FS::newP(unsigned int) at c:\users\galarzaa\documents\arduino\libraries\firebase_arduino_client_library_for_esp8266_and_esp32\src\mbfs/mb_fs.h line 788
0x40209549: FB_Firestore::handleResponse(FirebaseData*, fb_esp_firestore_req_t*) at C:\Users\Galarzaa\Documents\Arduino\libraries\Firebase_Arduino_Client_Library_for_ESP8266_and_ESP32\src\firestore\FB_Firestore.cpp line 1079
0x4020a358: FB_Firestore::firestore_sendRequest(FirebaseData*, fb_esp_firestore_req_t*) at C:\Users\Galarzaa\Documents\Arduino\libraries\Firebase_Arduino_Client_Library_for_ESP8266_and_ESP32\src\firestore\FB_Firestore.cpp line 939
0x40100d82: free(void*) at C:\Users\Galarzaa\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266\umm_malloc\umm_malloc.cpp line 595
0x4020a3f6: FB_Firestore::connect(FirebaseData*) at C:\Users\Galarzaa\Documents\Arduino\libraries\Firebase_Arduino_Client_Library_for_ESP8266_and_ESP32\src\firestore\FB_Firestore.cpp line 970
0x4020a4b2: FB_Firestore::sendRequest(FirebaseData*, fb_esp_firestore_req_t*) at C:\Users\Galarzaa\Documents\Arduino\libraries\Firebase_Arduino_Client_Library_for_ESP8266_and_ESP32\src\firestore\FB_Firestore.cpp line 653
0x40208948: MB_String::setPtr(mb_string::mb_string_ptr_t) at C:\Users\Galarzaa\Documents\Arduino\libraries\Firebase_Arduino_Client_Library_for_ESP8266_and_ESP32\src/json/MB_String.h line 790
0x4020a6a9: FB_Firestore::mPatchDocument(FirebaseData*, mb_string::mb_string_ptr_t, mb_string::mb_string_ptr_t, mb_string::mb_string_ptr_t, mb_string::mb_string_ptr_t, mb_string::mb_string_ptr_t, mb_string::mb_string_ptr_t, mb_string::mb_string_ptr_t, mb_string::mb_string_ptr_t) at C:\Users\Galarzaa\Documents\Arduino\libraries\Firebase_Arduino_Client_Library_for_ESP8266_and_ESP32\src\firestore\FB_Firestore.cpp line 193
0x40100d82: free(void*) at C:\Users\Galarzaa\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266\umm_malloc\umm_malloc.cpp line 595
0x40217fd5: String::invalidate() at C:\Users\Galarzaa\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266/WString.h line 295
0x40205820: loop() at C:\Users\Galarzaa\Documents\Arduino\libraries\Firebase_Arduino_Client_Library_for_ESP8266_and_ESP32\src/json/MB_String.h line 446
0x40218974: loop_wrapper() at C:\Users\Galarzaa\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266\core_esp8266_main.cpp line 201
0x40218974: loop_wrapper() at C:\Users\Galarzaa\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266\core_esp8266_main.cpp line 201
0x40100260: ets_post(uint8, ETSSignal, ETSParam) at C:\Users\Galarzaa\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266\core_esp8266_main.cpp line 181
0x40218974: loop_wrapper() at C:\Users\Galarzaa\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266\core_esp8266_main.cpp line 201
```
## Expected behavior
I don't want this exception because the nodeMCU restarts after this and create new problems with mi code

## Additional context
Every day i got this problem, i think maybe is a problem with the RAM, because i have two big arrays, but i check the "heap" and all its okay, maybe i have other problem. 

I hope that you can help me, greetings.

mobizt, 2022-06-30:
https://github.com/mobizt/Firebase-ESP-Client#memory-options-for-esp8266

mobizt, 2022-06-30:
You should debug for WiFi and SSL and your code.

The library itself works fine from intensive test for years.

closed 2022-06-30 by mobizt