mobizt/Firebase-ESP-Client #348: Firebase Storage Upload Error from SD Card File using SdFat or Standard SPI

rt2gaurav, 2022-07-10:
### Discussed in https://github.com/mobizt/Firebase-ESP-Client/discussions/347

<div type='discussions-op-text'>

<sup>Originally posted by **rt2gaurav** July 10, 2022</sup>
**Description**
I am using ESP32 Wroom 4MB and I am trying to upload a file to Firebase Storage from SD card. I am getting fcs callback error as "response payload read timed out due to network issue or too large data size" which looks similar to the issue [#315](https://github.com/mobizt/Firebase-ESP-Client/discussions/315). I have two Firebase libraries ([Firebase-ESP-Client](https://github.com/mobizt/Firebase-ESP-Client) & [Firebase-ESP32](https://github.com/mobizt/Firebase-ESP32))which was installed using Arduino's Library Manager.

I am using hspi pins for reading and writing the files from my SD card. I am sharing the example code to upload a file to Firebase Storage with slightly modified code for my hspi pins which used to work earlier but after that I updated the arduino core and Firebase libraries but since then it stopped working. I was using SPI call function to access the SD files which used to work. In the sample code below I have commented the hspi part to use [SdFat library](https://github.com/greiman/SdFat) but that also gives me same error. I am not sure if I have configured my FirebaseFS.h and SDHelper.h library. I have attached my [FirebaseFS.txt] and [SDHelper.txt] library for your reference let me know if I need to modify anything else. I haven't modified anything in SdFat library. This is the [example code from the library for Firebase Storage File Upload which I am referring to](https://github.com/mobizt/Firebase-ESP-Client/blob/main/examples/Storage/FirebaseStorage/UploadFile/UploadFile.ino)

I have tried using ESP core 2.0.1,2.0.3 & latest 2.0.4 versions. For Firebase library I have tried using 4.0.0 for both libraries. I have also tried Firebase ESP client with v3.3.3. Since I am using Firebase32 example for cloud storage upload I am not sure what version of Firebase32 library I should use as you have shared v3.3.3 of Firebase ESP Client to be working. Since there are two libraries it gets little bit more confusing and complicated for me as it might be causing conflicting issues due to different versions and I don't know what files are used during compilation of the code.

**Questions**

1. What version of library I should use for Firebase32, Firebase ESP Client, SdFat as well as ESP Core.
2. To be on safe side I updated FirebaseFS and SdHelper files in both library location. Do I need to update anything else? Does my file configuration look good to use SdFat with hspi pins [Library Files.zip](https://github.com/mobizt/Firebase-ESP-Client/files/9079603/Library.Files.zip)
3. What should I comment out in library and in my code below if I want to use HSPI calls from my code. Currently HSPI part of code is commented below
4. I have disabled OTA to get extra memory on my ESP32 and currently having 2MB for APP and 2MB for SPIFFS. Could this cause an issue?

```
#if defined(ESP32)
#include <WiFi.h>
#elif defined(ESP8266)
#include <ESP8266WiFi.h>
#endif

#include <Firebase_ESP_Client.h>
#include <FirebaseESP32.h>
#include "SdFat.h"
// Provide the token generation process info.
#include <addons/TokenHelper.h>

// Provide the SD card interfaces setting and mounting
#include <addons/SDHelper.h>

/* 1. Define the WiFi credentials */
#define WIFI_SSID "WIFI_AP"
#define WIFI_PASSWORD "WIFI_PASSWORD"

/* 2. Define the API Key */
#define API_KEY "API_KEY"

/* 3. Define the user Email and password that alreadey registerd or added in your project */
#define USER_EMAIL "USER_EMAIL"
#define USER_PASSWORD "USER_PASSWORD"

/* 4. Define the Firebase storage bucket ID e.g bucket-name.appspot.com */
#define STORAGE_BUCKET_ID "BUCKET-NAME.appspot.com"

// Define Firebase Data object
FirebaseData fbdo;

FirebaseAuth auth;
FirebaseConfig config;

bool taskCompleted = false;
char user_id[24] = "abc";
char workout_id[32] = "123";
const String Board_id = "LL02";
char FB_filename[30] = "";
char Ffilename[30] = "testgb";
String FB_url;

void setup()
{

    Serial.begin(115200);
    Serial.println();
    Serial.println();

[... 119 lines cut ...]
    }
}
```

**IDE and its version:**
Arduino IDE
Version [1.8.19]

**ESP32 Arduino Core SDK version**
Version [2.0.1, 2.0.3, 2.0.4]

**esptool.py**
Version [3.1]

**Firebase ESP client**
Version [3.3.3, 4.0.0,4.0.3]// I don't remember which version I was using earlier when my code used to work with hspi calls

**Firebase32**
Version [4.0.0,4.0.3] // I don't remember which version I was using earlier when my code used to work with hspi calls

**SdFat**
Version [2.2.0]
[image]

</div>

mobizt, 2022-07-11:
These two libraries i.e. [Firebase-ESP32](https://github.com/mobizt/Firebase-ESP32) and [Firebase-ESP8266](https://github.com/mobizt/Firebase-ESP8266), are the libraries that developed to solve the difficulty fingerprint update problem of this original [Firebase library](https://github.com/FirebaseExtended/firebase-arduino/issues/369) for ESP8266.

The [Firebase-ESP32](https://github.com/mobizt/Firebase-ESP32) and [Firebase-ESP8266](https://github.com/mobizt/Firebase-ESP8266) libraries are originally accepted only database secret for Firebase REST API access.

Both libraries, [Firebase-ESP32](https://github.com/mobizt/Firebase-ESP32) and [Firebase-ESP8266](https://github.com/mobizt/Firebase-ESP8266) are for RTDB and FCM (legacy API) only.

And [more authentication methods are requested](https://github.com/mobizt/Firebase-ESP8266/issues/160) by some user and then the new unified library, [Firebase-ESP-Client](https://github.com/mobizt/Firebase-ESP8266/issues/183) was published with more authentication methods supported and more Firebase services are supported e.g. Firestore, Cloud Functions, Google Cloud Storage, Firebase Storage and FCM with legacy and HTTPv1 APIs. 

For better libraries development and maintenance, the libraries [Firebase-ESP32](https://github.com/mobizt/Firebase-ESP32) and [Firebase-ESP8266](https://github.com/mobizt/Firebase-ESP8266) are later used the same source code of authentication, RTDB and legacy FCM as in as [Firebase-ESP-Client ](https://github.com/mobizt/Firebase-ESP-Client) library.

These libraries [Firebase-ESP32](https://github.com/mobizt/Firebase-ESP32) and [Firebase-ESP8266](https://github.com/mobizt/Firebase-ESP8266) are kept updated when  [Firebase-ESP-Client ](https://github.com/mobizt/Firebase-ESP-Client) is updated because of share source code.

As I mentioned about the share code of RTDB and legacy FCM among these three libraries, when you installed this [Firebase-ESP-Client ](https://github.com/mobizt/Firebase-ESP-Client) and [Firebase-ESP32](https://github.com/mobizt/Firebase-ESP32) libraries on the same IDE, the unexpected behavior during runtime or error during compile time due to compiler takes files from different libraries into compilation process may happen depending on the differences of share source code that can be affected with other Firebase service functions not only RTDB and FCM if it related to the utility classes or functions.

Then you should remove [Firebase-ESP32](https://github.com/mobizt/Firebase-ESP32) and [Firebase-ESP8266](https://github.com/mobizt/Firebase-ESP8266) from Arduino libraries folder if you want to use [Firebase-ESP-Client](https://github.com/mobizt/Firebase-ESP-Client).

I test your code together with [your three header files](https://github.com/mobizt/Firebase-ESP-Client/files/9079699/Library.Files.zip)  you shared above with the latest [Firebase-ESP-Client v4.0.3](https://github.com/mobizt/Firebase-ESP-Client/releases/tag/v4.0.3) and ESP32 [Arduino Core SDK v2.0.4](https://github.com/espressif/arduino-esp32/releases/tag/2.0.4) and it works fine.

I successfully test with both VSPI and HSPI on core SD library and [SdFat v2.2.0](https://github.com/greiman/SdFat/releases/tag/2.2.0) in both Arduino IDE and PlatformIO.

The library was intensively test for years and often updated to keep fixing the bugs especially about networks and filesystems.

The changes in ESP32 Arduino Core SDK v2.0.x which related to the ESP-IDF framework affected the SSL client, SPI and filesystems that used widely not only for this library.  Then you need to use the latest version of Firebase-ESP-Client library ([v4.0.3](https://github.com/mobizt/Firebase-ESP-Client/releases/tag/v4.0.3)) that these issues have been fixed.

The old version, v3.3.3 from [this topic](https://github.com/mobizt/Firebase-ESP-Client/discussions/315) which published at that time is to fix this Core SDK v2.0.x internal changes issues.

That means you can use the library version earlier before v3.3.3 for ESP32 Arduino Core SDK v1.0.6 and earlier.

The OTA, flash program space and memory allocation are library unrelated.

I recommend you to reduce the SPI clock frequency depending on your wiring and SD adapter conditions.

closed 2022-07-11 by mobizt

rt2gaurav, 2022-07-11:
@mobizt I am still facing the same issue. I removed all my previous library from my C:\Users\xx\xxx\Arduino\libraries and also from C:\Users\xx\AppData\Local\Arduino15\staging\libraries locations just to give fresh start. I reinstalled 4.0.3 Firebase-ESP-Client library via Arduino IDE Library Manager. What I noticed it upon installing Arduino IDE didn't renamed the downloaded folder name to "Firebase_Arduino_Client_Library_for_ESP8266_and_ESP32" and the contents of the folder were from Firebase-32. May be it is a separate issue so I deleted the files, restarted my computer and then re-installed the library for the 3rd time via manual process. I made sure that the folder name is changed after library installation. I modified the header files after library installation and then reopened Arduino IDE. So I think the libraries are downloaded and installed correctly.

I am attaching a sample json file which I am trying to upload to Firebase Storage. Can you try it on your end to see if the files gets uploaded. This data is ADC output.
[testgb.zip](https://github.com/mobizt/Firebase-ESP-Client/files/9081109/testgb.zip)

I tried slowing down the HSPI clock to 8MHz which is half of what I was using but still it doesn't work. I have tested it before I can write and read the file from SD card but when it comes to interfacing SD to Firebase Storage it gives me payload error. I want to be clear that I have tried uploading the file to Firebase RTDB server from same SD card at 16MHz clock speed which worked but Firebase Storage doesn't work.

mobizt, 2022-07-11:
The file size and file type are not related.

Here is the test result.

```yml
Connecting to Wi-Fi.
Connected with IP: 192.168.43.145

Firebase Client v4.0.3

Token info: type = id token, status = on request
Token info: type = id token, status = ready

Mounting SD Card... success

Upload file...

Uploading file /testgb.json (184002) to testgb.json
Uploaded 0%
Uploaded 2%
Uploaded 4%
Uploaded 6%
Uploaded 8%
Uploaded 10%
Uploaded 12%
Uploaded 14%
Uploaded 16%
Uploaded 18%
Uploaded 20%
Uploaded 22%
Uploaded 24%
Uploaded 26%
Uploaded 28%
Uploaded 30%
Uploaded 32%
Uploaded 34%
Uploaded 36%
Uploaded 38%
Uploaded 40%
Uploaded 42%
Uploaded 44%
Uploaded 46%
Uploaded 48%
Uploaded 50%
Uploaded 52%
Uploaded 54%
Uploaded 56%
Uploaded 58%
Uploaded 60%
Uploaded 62%
Uploaded 64%
Uploaded 66%
Uploaded 68%
Uploaded 70%
Uploaded 72%
Uploaded 74%
Uploaded 76%
Uploaded 78%
Uploaded 80%
Uploaded 82%
Uploaded 84%
Uploaded 86%
Uploaded 88%
Uploaded 90%
Uploaded 92%
Uploaded 94%
Uploaded 96%
Uploaded 98%
Uploaded 100%
Upload completed

Name: testgb.json
Bucket: xxxxxxxxxxx.appspot.com
contentType: application/json
Size: 184002
Generation: 4294967295
Metageneration: 0
ETag: CNSBioWR8PgCEAE=
CRC32:
Token: 1fae9e65-b611-46c1-b1d9-6a184475ea8f
Download URL: https://firebasestorage.googleapis.com/v0/b/xxxxxxxxxxx.appspot.com/o/testgb.json?alt=media&token=1fae9e65-b611-46c1-b1d9-6a184475ea8f
```

As I said the library tested intensively for years.

Large file uploads and downloads tested with size up to 10 MB. Many ESP32 variant boards and SD adapter module was tested.

As mentioned above some internal changes that I don't know in ESP32 Core SDK v2.0.x affected to the filesystems memory allocation (may be related to task stack memory reserved) that causes the library v3.3.2 and earlier don't work with ESP32 Core v2.0.x unless work fine for Core v1.0.6 and earlier version.

The different I have made to solve the ESP32 Core V2.0.x issue about filesystem memory allocation is to open file inside the scope that it reads or writes instead of open file once to standby and then read or write it later.

The SSL issue is not affected unless the filesystem access for read and write are always failed when file is standby to open before usage.

When the file read or write always return 0 length instead of data length that read or written that means failed to read or write and read or write loop task will be timed out and payload read or write timed out error will be printed out because it is in the request/response payload handling task.

rt2gaurav, 2022-07-11:
I switched to another board which works with the code. So may be it is hardware issue as I tried multiple times  to upload the file from non working board which never uploads while with the other board it always uploads. Since my SD card interface works i.e. I can read and write the data on both boards so it must be something else which happens on hardware when I try to upload the file. 

I am trying to upload 30MB of JSON file which takes about 10 mins to upload. I am using  `config.fcs.upload_buffer_size = 4096` command to define buffer length. I found that with buffer size more than 4096 bytes takes similar amount of time for my application. I am using SdFat with exFAT and using dedicated SPI with max clock speed in my case is 30MHz for HSPI CLK pin. Please suggest any advise to increase the speed of upload. Have I reached the bottleneck for ESP32? As per your comments in the code for SSL client:
`fbdo.setBSSLBufferSize(1024 /* Rx buffer size in bytes from 512 - 16384 */, 1024 /* Tx buffer size in bytes from 512 - 16384 */)`

for ESP8266 one needs to define Rx and Tx buffer width but what about ESP32 is it dynamic based on this command `config.fcs.upload_buffer_size = 4096` can I change the configuration for ESP32 to increase Tx buffer? Is there any other trick to increase the upload speed? 

I wanted to ask help from you as well community if there is any example of data logger with FIFO implementation for writing data from SD card file to Firebase storage. 
I am thinking of implementing software architecture like this where I want to develop code on ESP Core 0 for data sampling and writing the data from ADC to SD card and ESP32 Core 1 for reading and uploading data from SD card to Firebase Storage. I am not sure if by doing that there will be any issue as SD card's hspi pins and memory resources would be shared between two cores of ESP32. Do you see any potential issue with this architecture?

mobizt, 2022-07-12:
The config is applied to all devices, you can use this in ESP32 too

`config.fcs.upload_buffer_size = 16384;`

The buffer size 16384 is the maximum content length (max fragment size) to send via the mbedTLS engine.
The length is limited as defined in the mbedTLS related config file as used [here](https://github.com/espressif/arduino-esp32/blob/c93bf11f1e3c3ca8088b837525dcdd78c8579a7a/tools/sdk/esp32/dout_qspi/include/sdkconfig.h#L462).

Library limits this `config.fcs.upload_buffer_size` to 16384 too.

From my experience, ESP8266 is faster than ESP32 in secure TCP data transmission speed which I guess it may be related to SSL engines used to encrypt the data before sending to the IO buffer which BearSSL is constantly memory reserved for use without dynamic memory allocation code inside.

You have to benchmark this by yourself.

mobizt, 2022-07-12:
You said true about increasing the buffer size is not improve the speed due to the fixed size of partial length to send in the client code is limited to 1024 which I need to change fix it.

mobizt, 2022-07-12:
You can update library of the git version that write buffer issue fixed.

From my test, the upload speed depends on network status rather than SPI speed.

I can upload 53 MB file using SdFat and ESP32 in 9 minutes. With different network, the time used for upload 53 MB file increased to 15 minutes.