mobizt/Firebase-ESP-Client #360: HELP bad request from getJSON with query

CakUrang-dev, 2022-07-20:
this library is so helpful for me but I can't query my data this is my data structure

[image]

and this is my code

```
void getStudentCardHex() {
  Firebase.RTDB.setQueryIndex(&fbdo, "/RFID1/data_siswa", "nama", DATABASE_SECRET);
  
  QueryFilter query;

  query.orderBy("card_hex");
  query.equalTo(" 75 42 A0 AC");
  query.limitToLast(1);

  // Get filtered data
  Serial.printf("Get json... %s\n", Firebase.RTDB.getJSON(&fbdo, "/RFID1/data_siswa", &query) ? "ok" : fbdo.errorReason().c_str());

  if (fbdo.httpCode() == FIREBASE_ERROR_HTTP_CODE_OK)
    printResult(fbdo); // see addons/RTDBHelper.h

  // Clear all query parameters
  query.clear();
} 

void loop() {
  getStudentCardHex();
}
 ```
and this is my serial error

14:43:30.502 -> ................Connected with IP: 
14:43:36.037 -> 192.168.1.21
14:43:36.037 -> Firebase Client v4.0.3
14:43:36.037 -> 
14:43:36.037 -> Token info: type = id token, status = on request
14:43:37.738 -> Token info: type = id token, status = ready
14:43:37.738 -> 
14:43:40.366 -> Get json... bad request
14:43:42.974 -> Get json... bad request
14:43:45.443 -> Get json... bad request

is only getting bad requests, I thought my code was the same as your example DataFilter.ino

I hope you can help me

mobizt, 2022-07-20:
You should follow the [example](https://github.com/mobizt/Firebase-ESP-Client/blob/main/examples/RTDB/DataFilter/DataFilter.ino). 

There are many things that need to prepare and how to use library properly.

closed 2022-07-20 by mobizt

CakUrang-dev, 2022-07-20:
Ok, I put all code here

Main File

```
// Include WiFi Function
#include "wifi_function.h"

// Include RFID Library
#include <SPI.h>
#include <MFRC522.h>

// Include Firebase Library
#include <Firebase_ESP_Client.h>

// Provide the token generation process info
#include <addons/TokenHelper.h>

// Provide the RTDB payload printing info
#include <addons/RTDBHelper.h>

// Define API KEY
#define API_KEY "PUT YOUR API KEY"

// Define RTDB Url
#define DATABASE_URL "PUT YOUR DATABASE URL"

// Define Database Secret
#define DATABASE_SECRET "PUT YOUR DATABASE SECRET"

// Define User Email and Password for auth
#define USER_EMAIL "PUT YOUR EMAIL"
#define USER_PASSWORD "PUT YOUR PASSWORD"

// Define Firebase Data Object
FirebaseData fbdo;

FirebaseAuth auth;
FirebaseConfig config; 

unsigned long sendDataPrevMillis = 0;

// Declare RFID Pin
#define SS_PIN 2 // D4
#define RST_PIN 0 // D3
#define LED_PIN 4 // D2

MFRC522 mfrc522(SS_PIN, RST_PIN); // Create MFRC522 Instance

// Declaring Variable
String content;  // Storing Card Data Hex
int id = 0;
bool taskCompleted = false;

// Firebase init
void firebaseInit() {
  Serial.printf("Firebase Client v%s\n\n", FIREBASE_CLIENT_VERSION);

  // Assign the api key
  config.api_key = API_KEY;

  // Assign the user sign in in credentials
  auth.user.email = USER_EMAIL;
  auth.user.password = USER_PASSWORD;

  // Assign the RTDB url
  config.database_url = DATABASE_URL;

  // Assign the callback function for the long running token generation task
  config.token_status_callback = tokenStatusCallback;
[... 160 lines cut ...]
    if (fbdo.httpCode() == FIREBASE_ERROR_HTTP_CODE_OK)
      printResult(fbdo); // see addons/RTDBHelper.h
  
    // Clear all query parameters
    query.clear();
}
'''

it will get the data from /RFID1/data_siswa/

17:36:25.300 -> Get json... ok
17:36:25.300 -> Pretty printed JSON data:
17:36:25.339 -> {
17:36:25.339 -> 	"-N7PIl-pKl4EBGnl6vx2": {
17:36:25.339 -> 		"card_hex": " 75 42 A0 AC",
17:36:25.339 -> 		"nama": "siswa1"
17:36:25.339 -> 	}
17:36:25.339 -> }
17:36:25.339 -> Iterate JSON data:
17:36:25.339 -> 
17:36:25.339 -> 0, Type: object, Name: -N7PIl-pKl4EBGnl6vx2, Value: {"card_hex":" 75 42 A0 AC","nama":"siswa1"}
17:36:25.339 -> 1, Type: object, Name: card_hex, Value: " 75 42 A0 AC"
17:36:25.339 -> 2, Type: object, Name: nama, Value: "siswa1"

but that's not what I want, the code will query with card_hex for getting data, I don't know if that's a bug or not

mobizt, 2022-07-20:
### You MUST READ the Firebase REST API document from Google for the concept and how it works.

The query index is required by Google for Firebase REST API only for better performance on the backend.

If you want to query the the result for any key, you must add the index on that query in the database rules.

Then you need to use [this function](https://github.com/mobizt/Firebase-ESP-Client/tree/main/src#set-the-query-index-to-the-database-rules) for all the keys you want to query.

Please post such basic concept of Firebase question on community forum instead.

mobizt, 2022-07-20:
The function [setQueryIndex](https://github.com/mobizt/Firebase-ESP-Client/tree/main/src#set-the-query-index-to-the-database-rules) modified the database rules for you and you should learn from Firebase document or Firebase weblog for why you should set `.indexOn` key in the database rules.