mobizt/Firebase-ESP-Client #361: Bad request

CakUrang-dev, 2022-07-20:
Main File

```
// Include WiFi Function
#include "wifi_function.h"

// Include RFID Library
#include <SPI.h>
#include <MFRC522.h>

// Include Firebase Library
#include <Firebase_ESP_Client.h>

// Provide the token generation process info
#include <addons/TokenHelper.h>

// Provide the RTDB payload printing info
#include <addons/RTDBHelper.h>

// Define API KEY
#define API_KEY "PUT YOUR API KEY"

// Define RTDB Url
#define DATABASE_URL "PUT YOUR DATABASE URL"

// Define Database Secret
#define DATABASE_SECRET "PUT YOUR DATABASE SECRET"

// Define User Email and Password for auth
#define USER_EMAIL "PUT YOUR EMAIL"
#define USER_PASSWORD "PUT YOUR PASSWORD"

// Define Firebase Data Object
FirebaseData fbdo;

FirebaseAuth auth;
FirebaseConfig config; 

unsigned long sendDataPrevMillis = 0;

// Declare RFID Pin
#define SS_PIN 2 // D4
#define RST_PIN 0 // D3
#define LED_PIN 4 // D2

MFRC522 mfrc522(SS_PIN, RST_PIN); // Create MFRC522 Instance

// Declaring Variable
String content;  // Storing Card Data Hex
int id = 0;
bool taskCompleted = false;

// Firebase init
void firebaseInit() {
  Serial.printf("Firebase Client v%s\n\n", FIREBASE_CLIENT_VERSION);

  // Assign the api key
  config.api_key = API_KEY;

  // Assign the user sign in in credentials
  auth.user.email = USER_EMAIL;
  auth.user.password = USER_PASSWORD;

  // Assign the RTDB url
  config.database_url = DATABASE_URL;

  // Assign the callback function for the long running token generation task
  config.token_status_callback = tokenStatusCallback;

  // Only in ESP8266
  fbdo.setBSSLBufferSize(2048, 2048);
  // Limit the size of response payload
  fbdo.setResponseSize(2048);
  Firebase.begin(&config, &auth);

  // Reconnecting Wifi
  Firebase.reconnectWiFi(true);
  Firebase.setDoubleDigits(5);

  // WiFi reconnect timeout (interval) in ms (10 sec - 5 min) when WiFi disconnected
  config.timeout.serverResponse = 10 * 1000;
}

// Reading Card Function
void readCard() {
  // Look for new cards
  if(!mfrc522.PICC_IsNewCardPresent())
    return;
  // Select one of the card
  if(!mfrc522.PICC_ReadCardSerial())
    return;

  // Showing UID on serial monitor
  Serial.println();
  Serial.print("UID Tag : ");
  
  byte letter;
  for(byte i = 0; i < mfrc522.uid.size; i++) {
    Serial.print(mfrc522.uid.uidByte[i] < 0x10 ? " 0" : " ");
    Serial.print(mfrc522.uid.uidByte[i], HEX);
    content.concat(String(mfrc522.uid.uidByte[i] < 0x10 ? " 0" : " "));
    content.concat(String(mfrc522.uid.uidByte[i], HEX));
  }
  content.toUpperCase();
  content = ""; // Resetting Content
  id++;
  delay(1000);
}

// Function to Create Data
void createCardData() {
  FirebaseJson cardData; // JSON init
  
  cardData.add("card_hex", content);
  cardData.add("nama", "siswa" + String(id + 1));
  
  if(Firebase.RTDB.pushJSONAsync(&fbdo, "RFID1/data_siswa/" , &cardData)) {
    Serial.println();
    cardData.toString(Serial, true);
  } else {
    fbdo.errorReason().c_str();
  }

  cardData.clear();
}

// The main problem
void getStudentCardHex() {
   Firebase.RTDB.setQueryIndex(&fbdo, "/RFID1/data_siswa", "card_hex", DATABASE_SECRET);
    
    QueryFilter query;
  
    query.orderBy("card_hex");
    query.equalTo(" 75 42 A0 AC");
    query.limitToLast(1);
  
    // Get filtered data
    Serial.printf("Get json... %s\n", Firebase.RTDB.getJSON(&fbdo, "/RFID1/data_siswa", &query) ? "ok" : fbdo.errorReason().c_str());
  
    if (fbdo.httpCode() == FIREBASE_ERROR_HTTP_CODE_OK)
      printResult(fbdo); // see addons/RTDBHelper.h
  
    // Clear all query parameters
    query.clear();
}

void setup() {
  Serial.begin(115200); // Initiate serial communication
  SPI.begin(); // Initiate SPI bus
  mfrc522.PCD_Init(); // Initiate MFRC522
  
  // Function to connect to WiFi
  wifiConnect();

  // Function to connect to Firebase
  firebaseInit();
  Serial.println();

  
}

void loop() {
  if (Firebase.ready() && (millis() - sendDataPrevMillis > 1500 || sendDataPrevMillis == 0) && !taskCompleted) {
    sendDataPrevMillis = millis();
    
    // Reading Card Function
    readCard();

    getStudentCardHex();
  }
}
```
WiFI File

`wifi_function.h`

```
#ifndef WIFI_FUNCTION_H
#define WIFI_FUNCTION_H

// Include arduino library
#include <Arduino.h>

// Include WiFi Library
#include <ESP8266WiFi.h>

// Define the WiFi credentials
#define WIFI_SSID "SSID_WIFI"
#define WIFI_PASSWORD "PASSWORD_WIFI"

void wifiConnect();

#endif
```
`wifi_function.cpp`
```
// Inlude WiFi Function
#include "wifi_function.h"

void wifiConnect() {
  WiFi.begin(WIFI_SSID, WIFI_PASSWORD); // WiFi initiate
  Serial.print("Connecting to Wi-Fi");
  while(WiFi.status() != WL_CONNECTED) {
    Serial.print(".");
    delay(300);
  }
  Serial.println("Connected with IP: ");
  Serial.print(WiFi.localIP());
  Serial.println();
}
```

But, if I change the orderBy with another key nothing error appears and the code works normally and returns some value
Example :

'''
void getStudentCardHex() {
   Firebase.RTDB.setQueryIndex(&fbdo, "/RFID1/data_siswa", "card_hex", DATABASE_SECRET);
    
    QueryFilter query;
  
    query.orderBy("nama");
    query.equalTo("siswa1");
    query.limitToLast(1);
  
    // Get filtered data
    Serial.printf("Get json... %s\n", Firebase.RTDB.getJSON(&fbdo, "/RFID1/data_siswa", &query) ? "ok" : fbdo.errorReason().c_str());
  
    if (fbdo.httpCode() == FIREBASE_ERROR_HTTP_CODE_OK)
      printResult(fbdo); // see addons/RTDBHelper.h
  
    // Clear all query parameters
    query.clear();
}
'''

it will get the data from /RFID1/data_siswa/

17:36:25.300 -> Get json... ok
17:36:25.300 -> Pretty printed JSON data:
17:36:25.339 -> {
17:36:25.339 -> 	"-N7PIl-pKl4EBGnl6vx2": {
17:36:25.339 -> 		"card_hex": " 75 42 A0 AC",
17:36:25.339 -> 		"nama": "siswa1"
17:36:25.339 -> 	}
17:36:25.339 -> }
17:36:25.339 -> Iterate JSON data:
17:36:25.339 -> 
17:36:25.339 -> 0, Type: object, Name: -N7PIl-pKl4EBGnl6vx2, Value: {"card_hex":" 75 42 A0 AC","nama":"siswa1"}
17:36:25.339 -> 1, Type: object, Name: card_hex, Value: " 75 42 A0 AC"
17:36:25.339 -> 2, Type: object, Name: nama, Value: "siswa1"

but that's not what I want, the code will query with card_hex for getting data, I don't know if that's a bug or not

and this the rule of it

```
{
  "rules": {
    ".read": "now < 1660237200000",  // 2022-8-12
    ".write": "now < 1660237200000",  // 2022-8-12
    "RFID" : {
			"data_siswa": {
        ".read": true,
        ".write": true,
        ".indexOn": [
          "card_hex"
        ],
      }
    },
    "RFID1" : {
			"data_siswa": {
				".read": true,
        ".write": true,
        ".indexOn": [
          "card_hex",
          "nama"
        ]
      },
    },
  }
}
```

CakUrang-dev, 2022-07-20:
i know how firebase work but I new on arduino please help me

mobizt, 2022-07-20:
Do not open the issue again the post is temporary lock while I'm editing.

closed 2022-07-20 by mobizt

mobizt, 2022-07-20:
Answer what?

I already said that you must look at Firebase REST API document.

Library just convert the REST API to the functions and everything is the same functionalities.

Library functions cover all Firebase REST APIs.

mobizt, 2022-07-20:
Some comment in your rules e.g. // 2022-8-12 cause the invalid JSON and cannot parse by JSON library.

You should replace your database rules with this clean basic authentication

```yml
{
  "rules": {
    ".read": "auth != null", 
    ".write": "auth != null"
  }
}
```

And add the `.indexOn` key by using the function `setQueryIndex` again.

After you set all the keys's index to database rules, you don't need to call `setQueryIndex`  any more.

Read the Firebase REST API document for the meaning of the query options e.g. orderBy, equalTo, limitToLast etc.

Then try to play with it and see the result.

Please remember, all errors returned are from server directly. 

I'm busy and can't code for anyone based on user requirement.  The repository is for library related issue and questions for the library usage.

You can post the error or the requirement result on the community forum e.g. stackoverflow about Firebase REST API for help.