mobizt/Firebase-ESP-Client #368: [ssl_client.cpp] UNKNOWN ERROR CODE (004C)

ivodem, 2022-08-05:
**Hardware**
ESP32-S3-DevkitC-1

**platformio.ini**
```[env:esp32-s3-devkitc-1]
platform =  https://github.com/platformio/platform-espressif32.git
board = esp32-s3-devkitc-1
board_build.partitions = default_8MB.csv
board_build.f_cpu = 240000000L
board_upload.flash_size = 8MB
framework = arduino
monitor_speed = 115200
build_flags = -DCORE_DEBUG_LEVEL=5
lib_deps = mobizt/Firebase Arduino Client Library for ESP8266 and ESP32@^4.0.3
```
**Bug description**
When running a simple `RTDB.setInt()` example with a frequency of ~5Hz to a Real-time Database, I receive the error `UNKNOWN ERROR CODE (004C)` from `ssl_client.cpp`. Giving the following monitor output every **32 seconds** on average:
```
[344664][V][ssl_client.cpp:369] send_ssl_data(): Writing HTTP request with 24 bytes...
[344706][V][ssl_client.cpp:369] send_ssl_data(): Writing HTTP request with 905 bytes...
[344715][V][ssl_client.cpp:369] send_ssl_data(): Writing HTTP request with 244 bytes...
[344718][V][ssl_client.cpp:369] send_ssl_data(): Writing HTTP request with 4 bytes...
[344789][E][ssl_client.cpp:37] _handle_error(): [data_to_read():361]: (-76) UNKNOWN ERROR CODE (004C)
[344792][V][ssl_client.cpp:324] stop_ssl_socket(): Cleaning SSL connection.
ok
[345013][V][ssl_client.cpp:324] stop_ssl_socket(): Cleaning SSL connection.
[345107][E][WiFiClientSecure.cpp:383] setSocketOption(): 1006 : 9
[345108][V][ssl_client.cpp:62] start_ssl_client(): Free internal heap before TLS 266264
[345116][V][ssl_client.cpp:68] start_ssl_client(): Starting socket
[345153][V][ssl_client.cpp:149] start_ssl_client(): Seeding the random number generator
[345157][V][ssl_client.cpp:158] start_ssl_client(): Setting up the SSL/TLS structure...
[345158][D][ssl_client.cpp:179] start_ssl_client(): WARNING: Skipping SSL Verification. INSECURE!
[345167][V][ssl_client.cpp:257] start_ssl_client(): Setting hostname for TLS session...
[345210][V][ssl_client.cpp:272] start_ssl_client(): Performing the SSL/TLS handshake...
[346014][V][ssl_client.cpp:293] start_ssl_client(): Verifying peer X.509 certificate...
[346015][V][ssl_client.cpp:301] start_ssl_client(): Certificate verified.
[346018][V][ssl_client.cpp:316] start_ssl_client(): Free internal heap after TLS 222596
[346026][V][ssl_client.cpp:369] send_ssl_data(): Writing HTTP request with 24 bytes...
[346036][V][ssl_client.cpp:369] send_ssl_data(): Writing HTTP request with 905 bytes...
[346048][V][ssl_client.cpp:369] send_ssl_data(): Writing HTTP request with 244 bytes...
[346052][V][ssl_client.cpp:369] send_ssl_data(): Writing HTTP request with 4 bytes...
ok
```

Transactions are delivered successfully and 'on-time' between error messages. Error duration and reset of SSL take around 500 to 1000ms thus causing a delay in data transactions from ESP32 to Firebase.

**To Reproduce**
The function `writeToFirebase()` running as a task.
```
void writeToFirebase(void *parameter)
{
  for (;;)
  {
    if (Firebase.ready())
    {
      Firebase.RTDB.setInt(&fbdo, "/test/int", count);
      count++;
    }
    Serial.println("ok");
    vTaskDelay(200 / portTICK_PERIOD_MS); // ~5Hz
  }
}
```
Together with a task to keep WiFi connected.
```
xTaskCreatePinnedToCore(
      wifiConnect,
      "Keep alive WiFi", // Task name
      4069,              // Stack size (bytes)
      NULL,              // Parameter
      1,                 // Task priority
      NULL,              // Task handle
      ARDUINO_RUNNING_CORE);

    xTaskCreatePinnedToCore( // FreeRTOS task creation
        writeToFirebase,     // Function to be called
        "Write data",        // Name of task
        10240,                // Stack size
        NULL,                // Parameter to pass
        1,                   // Task priority
        NULL,                // Task handle
        0);
```
**IDE and its version:**
 - PlatformIO Core, version 6.1.3

mobizt, 2022-08-05:
The error is from the core sdk library i.e. WiFiClientSecure and is not related to the library. 

The most possible cause is device hardware related and you should try other devices.

The SSL/TLS handshake takes time and already stated in Readme [here](https://github.com/mobizt/Firebase-ESP-Client#speed-of-data-transfer).

Before open the issue, please make sure it is the actual library issue and consider to use discussion instead.

closed 2022-08-05 by mobizt

mobizt, 2022-08-06:
I test your code for many hours and everything is ok.

And I confirm that this error is related to hardware and you can found the same issue error "[ssl_client.cpp] UNKNOWN ERROR CODE (004C)" from search. Some workarounds may work for someone but not work for others.

Try other good quality boards instead.

ivodem, 2022-08-06:
Sorry for opening an unrelated issue. I was fairly certain this was a library issue. 
As you mentioned you used the same configuration, do you mean you have tried running my code on the same hardware, i.e., ESP32-S3-DevkitC-1?

Do you suggest opening this issue under [arduino-esp32](https://github.com/espressif/arduino-esp32)?

mobizt, 2022-08-06:
This is not even ESP32 Arduino core issue too because all of my ESP32 devices are running fine.

From my past experiences based on this such rarely case errors are from hardware.

mobizt, 2022-08-06:
I don't think it is related to board variant because this is about the WiFi functionality.