mobizt/Firebase-ESP-Client #370: Ethernet  enc28j60 firestore

KJPOUNDY132, 2022-08-09:
Hi, is  there support or example for FireStore with ethernet (especially enc28j60)

mobizt, 2022-08-09:
There is no relevant between Firestore and network interfaces.

Currently library provided 100+ examples and it's not possible to create examples cover all network interfaces, Arduino devices and all Firebase services. 

### YOU MUST LEARN THE BASIC CONCEPT AND ADAPT IT YOUSELF

You need to adapt from this [example](https://github.com/mobizt/Firebase-ESP-Client/blob/main/examples/RTDB/BasicEthernet/ESP8266/ESP8266.ino).

For other Arduino boards with ethernet, you should adapt [this example](https://github.com/mobizt/Firebase-ESP-Client/blob/main/examples/ExternalClient/RTDB/Basic/Basic.ino).

closed 2022-08-09 by mobizt

KJPOUNDY132, 2022-08-10:
maybe i dont know but