mobizt/Firebase-ESP-Client #374: ID Token getting expired in only 5 min

spsarolkar, 2022-08-17:
**Just ask something**
If you have questions, use [Discussions](https://github.com/mobizt/Firebase-ESP-Client/discussions) instead. 

**Describe the bug**
I am trying to use ID token and refresh token to connect to Firebase from ESP8266, but the id token is getting expired every 5 min

**To Reproduce**
```
#if defined(ESP32)
#include <WiFi.h>
#elif defined(ESP8266)
#include <ESP8266WiFi.h>
#endif

#include <Firebase_ESP_Client.h>

// Provide the token generation process info.
#include <addons/TokenHelper.h>

// Provide the RTDB payload printing info and other helper functions.
#include <addons/RTDBHelper.h>

/* 1. Define the WiFi credentials */
#define WIFI_SSID "**"
#define WIFI_PASSWORD "**"

/** 2. Define the API key
 *
 * The API key can be obtained since you created the project and set up
 * the Authentication in Firebase console.
 *
 * You may need to enable the Identity provider at https://console.cloud.google.com/customer-identity/providers
 * Select your project, click at ENABLE IDENTITY PLATFORM button.
 * The API key also available by click at the link APPLICATION SETUP DETAILS.
 *
 */
#define API_KEY "**"

/* 3. If work with RTDB, define the RTDB URL */
#define DATABASE_URL "**.firebasedatabase.app/" //<databaseName>.firebaseio.com or <databaseName>.<region>.firebasedatabase.app

/* 4. Define the Firebase Data object */
FirebaseData fbdo;

/* 5. Define the FirebaseAuth data for authentication data */
FirebaseAuth auth;

/* 6. Define the FirebaseConfig data for config data */
FirebaseConfig config;

unsigned long dataMillis = 0;
int count = 0;

void setup()
{

    Serial.begin(115200);

    WiFi.begin(WIFI_SSID, WIFI_PASSWORD);
    Serial.print("Connecting to Wi-Fi");
    while (WiFi.status() != WL_CONNECTED)
    {
        Serial.print(".");
        delay(300);
    }
    Serial.println();
    Serial.print("Connected with IP: ");
    Serial.println(WiFi.localIP());
    Serial.println();

[... 729 lines cut ...]
 NOW -1660762299Current time: Wed Aug 17 18:51:39 2022
 NOW -1660762300Current time: Wed Aug 17 18:51:40 2022
 NOW -1660762301Current time: Wed Aug 17 18:51:41 2022
 NOW -1660762302Current time: Wed Aug 17 18:51:42 2022
 NOW -1660762303Current time: Wed Aug 17 18:51:43 2022
 NOW -1660762304Current time: Wed Aug 17 18:51:44 2022
 NOW -1660762305Current time: Wed Aug 17 18:51:45 2022
 NOW -1660762306Current time: Wed Aug 17 18:51:46 2022
 NOW -1660762307Current time: Wed Aug 17 18:51:47 2022
 NOW -1660762308Current time: Wed Aug 17 18:51:48 2022
 NOW -1660762309Current time: Wed Aug 17 18:51:49 2022
 NOW -1660762310Current time: Wed Aug 17 18:51:50 2022
 NOW -1660762311Current time: Wed Aug 17 18:51:51 2022

```

**IDE and its version:**
 - Arduino 
 - Version 1.8.15

**ESP8266 Arduino Core SDK version**
 - Firebase Arduino Client Library for ESP8266 and ESP32  3.3.7

**Additional context**
Add any other context about the problem here.

mobizt, 2022-08-17:
Did you read and understand this line?

```
 // To refresh the token 5 minutes before expired
  config.signer.preRefreshSeconds = 3600 - 5 * 60;
```

mobizt, 2022-08-17:
You are trying to set the time which out of library control.

`configTime("IST-5:30", 0, "pool.ntp.org", "time.nist.gov");`

This is already addressing the [issue](https://github.com/mobizt/Firebase-ESP-Client/issues/290) when you do this.

mobizt, 2022-08-17:
Ok I may miss some point and will investigate this later when I have time.

closed 2022-08-20 by mobizt

spsarolkar, 2022-08-20:
Hi,

I see the defect closed without any comment, can you please help what is
the exact cause I am getting this error, anything wrong I am doing?

mobizt, 2022-08-20:
Sorry, if you follow the example, this line is wrong

https://github.com/mobizt/Firebase-ESP-Client/blob/60680080caf053db16106f63b20df1163324527b/examples/Authentications/SignInWithIDToken/SignInWithIDToken.ino#L110-L111

It will refresh token every 5 minutes, to refresh token 5 minutes before it expired, the config should be

```cpp
config.signer.preRefreshSeconds = 5 * 60;
```