mobizt/Firebase-ESP-Client #376: Firebase.patchDocument() takes approx 36 seconds to complete.

samkhan2050, 2022-08-22:
Hi, I am using an M5Stack ATOM DTU with ESP32 for communicating with Firebase Firestore and also RTDB as well. The reason for using RTDB is that i am connecting to stream of boolean values and based on the status of the boolean value, send an corresponding information to the Firestore. The problem i have is that that it take 36 seconds approx to complete a patch operation which i think is too long. Is it due to the handshake time of SSL connection? Or due to a simultaneous HTTP stream connection with the RTDB? My code is as below. 

```
#if defined(ESP32)
#include <WiFi.h>
#elif defined(ESP8266)
#include <ESP8266WiFi.h>
#endif

#include <M5Atom.h>
#include <Firebase_ESP_Client.h>
#include <ArduinoJson.h>
#include <ArduinoJson.hpp>

// Provide the token generation process info.
#include <addons/TokenHelper.h>

// Provide the RTDB payload printing info and other helper functions.
#include <addons/RTDBHelper.h>

/* 1. Define the WiFi credentials */
#define WIFI_SSID "osamakhan"
#define WIFI_PASSWORD "dubai0614"

/* 2. Define the API Key */
#define API_KEY "************" // HIDDEN DUE TO PRIVACY REASONS
#define FIREBASE_PROJECT_ID "*********" // HIDDEN DUE TO PRIVACY REASONS
/* 3. Define the RTDB URL */
#define DATABASE_URL "********************" // HIDDEN DUE TO PRIVACY REASONS

/* 4. Define the user Email and password that alreadey registerd or added in your project */
#define USER_EMAIL "********************" // HIDDEN DUE TO PRIVACY REASONS
#define USER_PASSWORD "********************" // HIDDEN DUE TO PRIVACY REASONS

//Serial Data Pins
#define RX_PIN      33
#define TX_PIN      23

// Define Firebase Data object
FirebaseData fbdoRTDB;
FirebaseJsonData RTVal;
FirebaseData fbdo; //Firestore Instance

FirebaseAuth auth;
FirebaseConfig config;

unsigned long sendDataPrevMillis = 0;

int count = 0;
unsigned long dataMillis = 0;

uint32_t idleTimeForStream = 15000;

// The Firestore payload upload callback function
void fcsUploadCallback(CFS_UploadStatusInfo info)
{
  if (info.status == fb_esp_cfs_upload_status_init)
  {
    Serial.printf("\nUploading data (%d)...\n", info.size);
  }
  else if (info.status == fb_esp_cfs_upload_status_upload)
  {
    Serial.printf("Uploaded %d%s\n", (int)info.progress, "%");
  }
  else if (info.status == fb_esp_cfs_upload_status_complete)
  {
    Serial.println("Upload completed ");
  }
  else if (info.status == fb_esp_cfs_upload_status_process_response)
  {
    Serial.print("Processing the response... ");
  }
  else if (info.status == fb_esp_cfs_upload_status_error)
  {
    Serial.printf("Upload failed, %s\n", info.errorMsg.c_str());
  }
}

void setup()
{

  Serial.begin(115200);
  Serial2.begin(115200, SERIAL_8N1, RX_PIN, TX_PIN);
  M5.begin(true, true, true);
  WiFi.begin(WIFI_SSID, WIFI_PASSWORD);
  Serial.print("Connecting to Wi-Fi");
  while (WiFi.status() != WL_CONNECTED)
  {
    Serial.print(".");
    delay(300);
  }
  Serial.println();
  Serial.print("Connected with IP: ");
  Serial.println(WiFi.localIP());
  Serial.println();

  Serial.printf("Firebase Client v%s\n\n", FIREBASE_CLIENT_VERSION);

  // For the following credentials, see examples/Authentications/SignInAsUser/EmailPassword/EmailPassword.ino

  /* Assign the api key (required) */
  config.api_key = API_KEY;

  /* Assign the user sign in credentials */
  auth.user.email = USER_EMAIL;
  auth.user.password = USER_PASSWORD;

  /* Assign the RTDB URL (required) */
  config.database_url = DATABASE_URL;

  /* Assign the callback function for the long running token generation task */
  config.token_status_callback = tokenStatusCallback; // see addons/TokenHelper.h

  // Or use legacy authenticate method
  // config.database_url = DATABASE_URL;
  // config.signer.tokens.legacy_token = "<database secret>";
#if defined(ESP8266)
  // In ESP8266 required for BearSSL rx/tx buffer for large data handle, increase Rx size as needed.
  fbdo.setBSSLBufferSize(2048 /* Rx buffer size in bytes from 512 - 16384 */, 2048/* Tx buffer size in bytes from 512 - 16384 */);
#endif
  // Limit the size of response payload to be collected in FirebaseData
  fbdo.setResponseSize(2048);

  // To connect without auth in Test Mode, see Authentications/TestMode/TestMode.ino

  Firebase.begin(&config, &auth);

  Firebase.reconnectWiFi(true);

  // The data under the node being stream (parent path) should keep small
  // Large stream payload leads to the parsing error due to memory allocation.
  if (!Firebase.RTDB.beginStream(&fbdoRTDB, "/ActivityFlags"))
    Serial.printf("stream begin error, %s\n\n", fbdoRTDB.errorReason().c_str());

}
void loop()
{
  M5.update();
  if (Firebase.ready()) {
     if (!Firebase.RTDB.readStream(&fbdoRTDB))
      Serial.printf("stream read error, %s\n\n", fbdoRTDB.errorReason().c_str());
    if (fbdoRTDB.streamTimeout()){
      Serial.println("stream timed out, resuming...\n");
    if (!fbdoRTDB.httpConnected())
        Serial.printf("error code: %d, reason: %s\n\n", fbdoRTDB.httpCode(), fbdoRTDB.errorReason().c_str());
    }
     if (fbdoRTDB.streamAvailable()){
      FirebaseJson json = fbdoRTDB.to<FirebaseJson>();
      json.get(RTVal, "RTVal");
     if (RTVal.to<bool>()==true) {
               
        String docPathUpdate = "Customer/lCaUvtZJ4apDywoJ4NUv/Cranes/uT2fcoj8vSy6XOouLSVb/";
        
        FirebaseJson content;
        
        content.set("fields/LoadAngleInfo/arrayValue/values/[0]/mapValue/fields/H1MeasuredLoad/integerValue", "10");
        content.set("fields/LoadAngleInfo/arrayValue/values/[1]/mapValue/fields/H2MeasuredLoad/integerValue", "20");
        content.set("fields/LoadAngleInfo/arrayValue/values/[2]/mapValue/fields/TotalMeasuredLoad/integerValue", "30");
         
        Serial.println("Content Created");
        content.toString(Serial, true);
        Serial.println();
        Serial.println();
        
        updateDocFireStore(docPathUpdate, content, "LoadAngleInfo");
        Serial.println("Firestore updated");
      } 
    }
  } 
}

void updateDocFireStore(String docPathUpdate,FirebaseJson content, String fieldsUpdated){
  Serial.println("Update FireStore Started");
  Serial.println(millis());
  if (Firebase.Firestore.patchDocument(&fbdo, FIREBASE_PROJECT_ID, "" , docPathUpdate.c_str(), content.raw(), fieldsUpdated))
        Serial.printf("ok");//\n%s\n\n", fbdo.payload().c_str());
      else{
       Firebase.Firestore.createDocument(&fbdo, FIREBASE_PROJECT_ID, "" , docPathUpdate.c_str(), content.raw());
      Serial.println(millis());
      }
        Serial.println(millis());
  
}
```

`

mobizt, 2022-08-22:
it depends on your network.

closed 2022-08-22 by mobizt

samkhan2050, 2022-08-22:
But when i use the same patch command without the RTDB stream on the same network, it takes less than 1 seconds.

mobizt, 2022-08-22:
The Firebase REST API is the blocking operation, you should use other tasks in case multitasking.

samkhan2050, 2022-08-22:
Thanks a lot, i will try running the patch command on the other core

mobizt, 2022-08-22:
This [stream callback](https://github.com/mobizt/Firebase-ESP-Client/blob/main/examples/RTDB/DataChangesListener/Callback/Callback.ino) creates the RTOS task for stream.

You can follow the example by avoiding to run any blocking function in the callback function.