mobizt/Firebase-ESP-Client #377: response payload read timed out due to network issue or too large data size

mej558, 2022-08-24:
I feel very close to getting the ESP32 - Firebase firmware to work reliably. It seems to run fine, but will periodically have the following:

**"response payload read timed out due to network issue or too large data size"**  The only data exchanged with the RTDB is a single integer.

Based on the lack of serial output, the WIFI connection appears to be connected and not the cause of the problem.

The best solution I found is a restart...

I don't understand all the options available. Is there something I'm missing to keep the connection to the database?

**Arduino 1.8.19 
Adafruit (PID 3591) HUZZAH32 – ESP32 Feather Board**

2022-08-23 18:56:47 --> ets Jul 29 2019 12:21:46
2022-08-23 18:56:47 --> 
2022-08-23 18:56:47 --> rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
2022-08-23 18:56:47 --> configsip: 0, SPIWP:0xee
2022-08-23 18:56:47 --> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
2022-08-23 18:56:47 --> mode:DIO, clock div:1
2022-08-23 18:56:47 --> load:0x3fff0030,len:1184
2022-08-23 18:56:47 --> load:0x40078000,len:12812
2022-08-23 18:56:47 --> load:0x40080400,len:3032
2022-08-23 18:56:47 --> entry 0x400805e4
2022-08-23 18:56:47 --> Connecting to Wi-Fi...
2022-08-23 18:56:48 --> 
2022-08-23 18:56:48 --> Connected with IP: 192.168.86.238
2022-08-23 18:56:48 --> 
2022-08-23 18:56:48 --> **Firebase Client v4.0.3**
2022-08-23 18:56:48 --> 
2022-08-23 18:56:49 --> Token info: type = id token, status = on request
2022-08-23 18:56:50 --> Token info: type = id token, status = ready
2022-08-23 18:56:50 --> Get integer... 0
2022-08-23 18:56:56 --> Get integer... 255 //works as expected
2022-08-23 18:57:04 --> Get integer... 128 //works as expected
2022-08-23 18:57:10 --> Get integer... 4 //works as expected
2022-08-23 18:57:18 --> Get integer... 1 //works as expected... Leave ESP32 running...
2022-08-23 19:33:53 --> Free Heap Space: 252932
2022-08-23 19:33:53 --> Get int... **response payload read timed out due to network issue or too large data size**

```
#include <WiFi.h>
#include <Firebase_ESP_Client.h>
#include "addons/TokenHelper.h"
#include "addons/RTDBHelper.h"

#define ledPin 13   // LED on Feather32
unsigned long sendDataPrevMillis = 0;
unsigned long count = 0;
int ALARM_STATUS = 0;  //integer value used to determine current alarm state

/**********************************************
 * FIREBASE SETUP
 **********************************************/
// Define Firebase objects
FirebaseData fbdo;
FirebaseAuth auth;
FirebaseConfig config;

/********************************************************************************************************************************************* 
 * For the following credentials, see examples/Authentications/SignInAsUser/EmailPassword/EmailPassword.ino 
 *********************************************************************************************************************************************/
#define API_KEY "AI---"
#define DATABASE_URL "ea---.firebaseio.com"
#define USER_EMAIL "ea---@gmail.com"
#define USER_PASSWORD "rd---"
#define WIFI_SSID "GO---"
#define WIFI_PASSWORD "br---"

/**********************************************
 * Firebase Variables to save database paths
 **********************************************/
String listenerPath = "/3423---/";
String dataPath = "/3423---/status";

void setup()
{

  Serial.begin(115200);

  WiFi.begin(WIFI_SSID, WIFI_PASSWORD);
  Serial.print("Connecting to Wi-Fi");
  while (WiFi.status() != WL_CONNECTED)
  {
    Serial.print(".");
    delay(300);
  }
  Serial.println();
  Serial.print("Connected with IP: ");
  Serial.println(WiFi.localIP());
  Serial.println();

  Serial.printf("Firebase Client v%s\n\n", FIREBASE_CLIENT_VERSION);

  pinMode(ledPin, OUTPUT);    // sets the digital pin 13 as output
  digitalWrite(ledPin, HIGH); // sets the digital pin 13 on
  delay(500);
  digitalWrite(ledPin, LOW); // sets the digital pin 13 off

  /* Assign the api key (required) */
  config.api_key = API_KEY;

  /* Assign the user sign in credentials */
  auth.user.email = USER_EMAIL;
  auth.user.password = USER_PASSWORD;

  /* Assign the RTDB URL (required) */
  config.database_url = DATABASE_URL;

  /* Assign the callback function for the long running token generation task */
  config.token_status_callback = tokenStatusCallback; // see addons/TokenHelper.h

  // To connect without auth in Test Mode, see Authentications/TestMode/TestMode.ino

  //////////////////////////////////////////////////////////////////////////////////////////////
  // Please make sure the device free Heap is not lower than 80 k for ESP32 and 10 k for ESP8266,
  // otherwise the SSL connection will fail.
  //////////////////////////////////////////////////////////////////////////////////////////////

  // Limit the size of response payload to be collected in FirebaseData
  fbdo.setResponseSize(8);

  Firebase.begin(&config, &auth);

  // Comment or pass false value when WiFi reconnection will control by your code or third party library
  Firebase.reconnectWiFi(true);

  Firebase.setDoubleDigits(5);

  config.timeout.serverResponse = 10 * 1000;

  Serial.printf("Get integer... %d\n", ALARM_STATUS);
}

void loop()
{
  // Firebase.ready() should be called repeatedly to handle authentication tasks.
  if (Firebase.ready() && (millis() - sendDataPrevMillis > 500 || sendDataPrevMillis == 0))
  {
    sendDataPrevMillis = millis();

    if(Firebase.RTDB.getInt(&fbdo, dataPath)){
      ALARM_STATUS = fbdo.to<int>();
      if(ALARM_STATUS > 0){
        Serial.printf("Get integer... %d\n", ALARM_STATUS); 
        ALARM_STATUS = 0;
        Firebase.RTDB.setInt(&fbdo, dataPath, 0);
        digitalWrite(ledPin, HIGH); // sets the digital pin 13 on
        delay(5000);
        digitalWrite(ledPin, LOW); // sets the digital pin 13 off
      }
    }else{
      Serial.printf("Free Heap Space: %d\n", ESP.getFreeHeap());
      Serial.printf("Get int... %s\n", fbdo.errorReason().c_str());
      if(WiFi.status() != WL_CONNECTED){
        Serial.printf("WiFi not connected... Restarting...");
      }
      ESP.restart();
    }
  }
}

```

mobizt, 2022-08-24:
If you are sure that the size of data at database node  to read is small, the most possible cause is network issue e.g. interruption between device and server.

This is rarely case that related to your device especially WiFi functionality.

You should try other device and powered it with low noise, enough current and stable voltage supply.

If you supply it via PC usb, try to use shorter cable. If it is not work, change your device orWiFi AP or router.

mobizt, 2022-08-24:
I see you are use this incorrectly.

`fbdo.setResponseSize(8);`

That is only 8 bytes reserved memory which not fit with header + payload.

closed 2022-08-24 by mobizt

mobizt, 2022-08-24:
You should not use this to limit data bandwidth because it can't help as all header and payload are transferred in/out the device already.

mej558, 2022-08-24:
Thanks! I will set it back to the default: fbdo.setResponseSize(2048).  I did have it on 2048 and lowered it.  I thought lowering might help...

I am using a long-ish USBcable to power the device from a Laptop. I will change that to a dedicated power supply. I've got two more modules on order that should be here tomorrow.

Letting it run overnight it seems to reset almost hourly. Here is more data. I put lines in to highlight the timing of the events. 

I see it is sometimes "refreshing", which is awesome because it doesn't trigger a restart. Restarting is a problem for me because my real program takes 13seconds to reboot but is supposed to be checking the RTDB every 500ms.

btw, I don't really understand it all yet, but it is really cool that you built this library and shared it! Thanks so much!!

2022-08-23 20:08:13 --> ets Jul 29 2019 12:21:46
2022-08-23 20:08:13 --> 
2022-08-23 20:08:13 --> rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
2022-08-23 20:08:13 --> configsip: 0, SPIWP:0xee
2022-08-23 20:08:13 --> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
2022-08-23 20:08:13 --> mode:DIO, clock div:1
2022-08-23 20:08:13 --> load:0x3fff0030,len:1184
2022-08-23 20:08:13 --> load:0x40078000,len:12812
2022-08-23 20:08:13 --> load:0x40080400,len:3032
2022-08-23 20:08:13 --> entry 0x400805e4
2022-08-23 20:08:14 --> Connecting to Wi-Fi.
2022-08-23 20:08:14 --> 
2022-08-23 20:08:14 --> Connected with IP: 192.168.86.238
2022-08-23 20:08:14 --> 
2022-08-23 20:08:14 --> Firebase Client v4.0.3
2022-08-23 20:08:14 --> 
2022-08-23 20:08:15 --> Token info: type = id token, status = on request
2022-08-23 20:08:16 --> Token info: type = id token, status = ready
2022-08-23 20:08:16 --> Get integer... 0
2022-08-23 20:08:27 --> Get integer... 255
2022-08-23 20:08:32 --> Get integer... 128
2022-08-23 20:08:37 --> Get integer... 4
2022-08-23 20:08:45 --> Get integer... 1

2022-08-23 20:49:52 --> Free Heap Space: 256248
2022-08-23 20:49:52 --> Get int... send request failed
2022-08-23 20:49:52 --> WiFi connectedRestarting...ets Jul 29 2019 12:21:46
2022-08-23 20:49:52 --> 
2022-08-23 20:49:52 --> rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
2022-08-23 20:49:52 --> configsip: 0, SPIWP:0xee
2022-08-23 20:49:52 --> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
2022-08-23 20:49:52 --> mode:DIO, clock div:1
2022-08-23 20:49:52 --> load:0x3fff0030,len:1184
2022-08-23 20:49:52 --> load:0x40078000,len:12812
2022-08-23 20:49:52 --> load:0x40080400,len:3032
2022-08-23 20:49:52 --> entry 0x400805e4
2022-08-23 20:49:52 --> Connecting to Wi-Fi.
2022-08-23 20:49:52 --> .
2022-08-23 20:49:53 --> 
2022-08-23 20:49:53 --> Connected with IP: 192.168.86.238
2022-08-23 20:49:53 --> 
2022-08-23 20:49:53 --> Firebase Client v4.0.3
2022-08-23 20:49:53 --> 
2022-08-23 20:49:53 --> Token info: type = id token, status = on request
2022-08-23 20:49:55 --> Token info: type = id token, status = ready
2022-08-23 20:49:55 --> Get integer... 0

2022-08-23 21:48:55 --> Token info: type = id token, status = on refreshing
2022-08-23 21:48:57 --> Token info: type = id token, status = ready

2022-08-23 22:47:58 --> Token info: type = id token, status = on refreshing
2022-08-23 22:48:00 --> Token info: type = id token, status = ready

2022-08-23 23:46:59 --> Token info: type = id token, status = on refreshing
2022-08-23 23:47:01 --> Token info: type = id token, status = ready

2022-08-24 00:43:19 --> Free Heap Space: 255968
2022-08-24 00:43:19 --> Get int... response payload read timed out due to network issue or too large data size
2022-08-24 00:43:19 --> WiFi connectedRestarting...
2022-08-24 00:43:19 --> ets Jul 29 2019 12:21:46
[... 57 lines cut ...]

2022-08-24 07:45:36 --> Free Heap Space: 255676
2022-08-24 07:45:36 --> Get int... send request failed
2022-08-24 07:45:36 --> WiFi connectedRestarting...
2022-08-24 07:45:36 --> ets Jul 29 2019 12:21:46
2022-08-24 07:45:36 --> 
2022-08-24 07:45:36 --> rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
2022-08-24 07:45:36 --> configsip: 0, SPIWP:0xee
2022-08-24 07:45:36 --> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
2022-08-24 07:45:36 --> mode:DIO, clock div:1
2022-08-24 07:45:36 --> load:0x3fff0030,len:1184
2022-08-24 07:45:36 --> load:0x40078000,len:12812
2022-08-24 07:45:36 --> load:0x40080400,len:3032
2022-08-24 07:45:36 --> entry 0x400805e4
2022-08-24 07:45:37 --> Connecting to Wi-Fi.
2022-08-24 07:45:37 --> .
2022-08-24 07:45:37 --> .
2022-08-24 07:45:38 --> .
2022-08-24 07:45:38 --> Connected with IP: 192.168.86.238
2022-08-24 07:45:38 --> 
2022-08-24 07:45:38 --> Firebase Client v4.0.3
2022-08-24 07:45:38 --> 
2022-08-24 07:45:39 --> Token info: type = id token, status = on request
2022-08-24 07:45:40 --> Token info: type = id token, status = ready
2022-08-24 07:45:40 --> Get integer... 0

mobizt, 2022-08-24:
Reading data from database too often is not the proper use because it do internal query on the server backend.

And you should learn about the different between Firebase REST API Client and Firebase SDK client. The Firebase SDK client runs under web socket technology.

In RTDB REST API, it provides [stream](https://github.com/mobizt/Firebase-ESP-Client/tree/main/examples/RTDB/DataChangesListener) to avoid inefficient and reduce the data bandwidth for polling read the database.

mej558, 2022-08-24:
Thanks. I started with the stream version and was getting the disconnection problem. I implemented the basic example to try to get to the bottom of the disconnection issue... Since it seems like it might be more of an electrical issue than a problem with the code or network, I will reimplement the stream version of the code, with your suggested electrical improvements and see how that works. Thanks again!