mobizt/Firebase-ESP-Client #378: ESP doesnt update Firebase

topslave, 2022-08-26:
I am writing to an array on the Firebase Database. ESP sometimes updates the database and sometimes doesnt. The log result is the same and as printed below.  Am I doing something wrong?

16:16:38.341 ->   "writeResults": [
16:16:38.341 ->     {
16:16:38.341 ->       "updateTime": "2022-08-26T13:04:24.214621Z",
16:16:38.341 ->       "transformResults": [
16:16:38.341 ->         {
16:16:38.341 ->           "nullValue": null
16:16:38.341 ->         }
16:16:38.341 ->       ]
16:16:38.341 ->     }
16:16:38.341 ->   ],
16:16:38.341 ->   "commitTime": "2022-08-26T13:16:37.623049Z"
16:16:38.341 -> }
16:16:38.341 -> 
16:16:38.341 -> 
16:17:35.227 -> Commit a document (append array)...  path is : /users/VvgCck6vCcNu4bNlaReMTTbZpvg1
16:17:36.723 -> ok
16:17:36.723 -> {
16:17:36.723 ->   "writeResults": [
16:17:36.723 ->     {
16:17:36.723 ->       "updateTime": "2022-08-26T13:04:24.214621Z",
16:17:36.723 ->       "transformResults": [
16:17:36.723 ->         {
16:17:36.723 ->           "nullValue": null
16:17:36.723 ->         }
16:17:36.723 ->       ]
16:17:36.723 ->     }
16:17:36.723 ->   ],
16:17:36.723 ->   "commitTime": "2022-08-26T13:17:36.220208Z"
16:17:36.723 -> }

topslave, 2022-08-26:
Looks like it is the array size,   the size of the array when it reaches a certain amount it doesnt update anymore.

mobizt, 2022-08-26:
It is not related to library.

The Firestore array will be appended only new value only, if that value is already existed in array, it cannot append.

closed 2022-08-26 by mobizt

topslave, 2022-08-26:
ok thank you, but the value doesnt exist. I mean this is an auto generated array with keys like below,  after 22 items it stops updating

appended_data
(array)
0 1085377743
1 "Hello World! 2"
2 1270216262
3 "Hello World! 3"
4 1191391529
5 "Hello World! 4"
6 812669700
7 "Hello World! 5"
8 553475508
9 "Hello World! 6"
10 445349752
11 "Hello World! 7"
12 1344887256
13 "Hello World! 8"
14 730417256
15 "Hello World! 9"
16 1812158119
17 "Hello World! 10"

topslave, 2022-08-26:
sorry on this run it updated 25 items maybe I am doing something wrong

mobizt, 2022-08-26:
I don't know your device. If ESP8266, at least you should read the comment in the examples.
https://github.com/mobizt/Firebase-ESP-Client/blob/0b49417b5ef3f56e60b316e7afd2ba1b9926d1b3/examples/Firestore/CommitDocument/AppendArray/AppendArray.ino#L80

topslave, 2022-08-26:
I played with that once but than set back to normal the chip I am using is  ESP32-D0WDQ6-V3

topslave, 2022-08-26:
ok here is the issue 
When I loaded again the hotpspot shows 2 devices connected than it didnt update for awhile. When I deleted the array from Firestore it started updated again, what do you think?

mobizt, 2022-08-26:
Please follow the examples.

Read the related documentation on Firestore REST API.
I can't debug or verify what's wrong with your code.

The payload is what Firestore returned.

mobizt, 2022-08-26:
If you provide the simple and clean test code that reproduces the issue which I can test, may be I can verify it.

topslave, 2022-08-26:
how can I email you the code? Thank you

mobizt, 2022-08-27:
**Just ask something**
If you have questions, use [Discussions](https://github.com/mobizt/Firebase-ESP-Client/discussions) instead. 

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Create minimal, clean and clear testing code that addressed the issue.
The third party libraries should be removed.
And post the testing code here.

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**IDE and its version:**
 - Arduino, PlatformIO
 - Version [e.g. 1.8.9]

**ESP8266 Arduino Core SDK version**
 - Version [e.g. 2.5.1]

**Additional context**
Add any other context about the problem here.