mobizt/Firebase-ESP-Client #397: FB_ENABLE_EXTERNAL_CLIENT network status callback doesn't seem to work

nickjsanders, 2022-09-15:
Adding this call in the callback seems to be the recommendation: 
`fbdo.setNetworkStatus(WiFi.status() == WL_CONNECTED);`

However, this sets the network status in the FirebaseData instance of tcpClient, but not the Firebase_Signer instance of tcpClient, 
causing auth to fail when failing to connect via that tcpClient. 

This seems to be fixed by:
```
void Firebase_Signer::setNetworkStatus(bool status)
{
    networkStatus = status;
    tcpClient->setNetworkStatus(status);    // add this line to update the signer's tcpClient state
}
```

**Expected behavior**
Is there another mechanism to set the Signer's tcpClient network status that i've missed?

**IDE and its version:**
 - Arduino teesny4+esp-at, PlatformIO

mobizt, 2022-09-16:
Thanks for reporting this. I will fix it soon.

mobizt, 2022-09-19:
Now issue fixed in [v4.1.0](https://github.com/mobizt/Firebase-ESP-Client/releases/tag/v4.1.0).

closed 2022-09-19 by mobizt