mobizt/Firebase-ESP-Client #415: Firebase Configuration Issue When Used in Header File

MatthewGerges2, 2022-10-26:
Hi, I was trying to use firebase in my latest project. When I add the Firebase_ESP_Client library to my project and use it in a .cpp file it works fine. However, when I try making the FirebaseData object a global variable by placing it in a header file, I run into multiple problems:

Here are the output errors:

```
.pio/libdeps/esp32dev/Firebase Arduino Client Library for ESP8266 and ESP32/src/rtdb/QueryFilter.h:74:35: error: expected initializer before 'enable_if'
.pio/libdeps/esp32dev/Firebase Arduino Client Library for ESP8266 and ESP32/src/rtdb/QueryFilter.h:77:37: error: expected nested-name-specifier before 'enable_if'
     auto equalTo(T val) -> typename enable_if<is_num_int<T>::value, QueryFilter &>::type { return mEqualTo(toStringPtr(val), false); }
                                     ^~~~~~~~~
.pio/libdeps/esp32dev/Firebase Arduino Client Library for ESP8266 and ESP32/src/rtdb/QueryFilter.h:77:37: error: expected initializer before 'enable_if'
.pio/libdeps/esp32dev/Firebase Arduino Client Library for ESP8266 and ESP32/src/rtdb/QueryFilter.h:80:37: error: expected nested-name-specifier before 'enable_if'
     auto equalTo(T val) -> typename enable_if<is_string<T>::value, QueryFilter &>::type { return mEqualTo(toStringPtr(val), true); }
                                     ^~~~~~~~~
.pio/libdeps/esp32dev/Firebase Arduino Client Library for ESP8266 and ESP32/src/rtdb/QueryFilter.h:80:37: error: expected initializer before 'enable_if'
In file included from .pio/libdeps/esp32dev/Firebase Arduino Client Library for ESP8266 and ESP32/src/Firebase.h:65,
                 from .pio/libdeps/esp32dev/Firebase Arduino Client Library for ESP8266 and ESP32/src/Firebase_ESP_Client.h:42,
                 from src/firebase.h:4,
                 from src/main.cpp:25:
.pio/libdeps/esp32dev/Firebase Arduino Client Library for ESP8266 and ESP32/src/session/FB_Session.h:615:25: error: expected nested-name-specifier before 'enable_if'
   auto to() -> typename enable_if<is_num_int<T>::value || is_num_float<T>::value || is_bool<T>::value, T>::type
                         ^~~~~~~~~
.pio/libdeps/esp32dev/Firebase Arduino Client Library for ESP8266 and ESP32/src/session/FB_Session.h:615:25: error: expected initializer before 'enable_if'
.pio/libdeps/esp32dev/Firebase Arduino Client Library for ESP8266 and ESP32/src/session/FB_Session.h:668:25: error: expected nested-name-specifier before 'enable_if'
   auto to() -> typename enable_if<is_const_chars<T>::value || is_std_string<T>::value || is_arduino_string<T>::value || is_mb_string<T>::value, T>::type
                         ^~~~~~~~~
.pio/libdeps/esp32dev/Firebase Arduino Client Library for ESP8266 and ESP32/src/session/FB_Session.h:668:25: error: expected initializer before 'enable_if'
.pio/libdeps/esp32dev/Firebase Arduino Client Library for ESP8266 and ESP32/src/session/FB_Session.h:676:25: error: expected nested-name-specifier before 'enable_if'
   auto to() -> typename enable_if<is_same<T, FirebaseJson *>::value, FirebaseJson *>::type
                         ^~~~~~~~~
.pio/libdeps/esp32dev/Firebase Arduino Client Library for ESP8266 and ESP32/src/session/FB_Session.h:676:25: error: expected initializer before 'enable_if'
.pio/libdeps/esp32dev/Firebase Arduino Client Library for ESP8266 and ESP32/src/session/FB_Session.h:692:25: error: expected nested-name-specifier before 'enable_if'
   auto to() -> typename enable_if<is_same<T, FirebaseJsonData *>::value, FirebaseJsonData *>::type
                         ^~~~~~~~~
.pio/libdeps/esp32dev/Firebase Arduino Client Library for ESP8266 and ESP32/src/session/FB_Session.h:692:25: error: expected initializer before 'enable_if'
.pio/libdeps/esp32dev/Firebase Arduino Client Library for ESP8266 and ESP32/src/session/FB_Session.h:700:25: error: expected nested-name-specifier before 'enable_if'
   auto to() -> typename enable_if<is_same<T, FirebaseJson>::value, FirebaseJson &>::type
                         ^~~~~~~~~
.pio/libdeps/esp32dev/Firebase Arduino Client Library for ESP8266 and ESP32/src/session/FB_Session.h:700:25: error: expected initializer before 'enable_if'
.pio/libdeps/esp32dev/Firebase Arduino Client Library for ESP8266 and ESP32/src/session/FB_Session.h:706:25: error: expected nested-name-specifier before 'enable_if'
   auto to() -> typename enable_if<is_same<T, FirebaseJsonArray *>::value, FirebaseJsonArray *>::type
                         ^~~~~~~~~
.pio/libdeps/esp32dev/Firebase Arduino Client Library for ESP8266 and ESP32/src/session/FB_Session.h:706:25: error: expected initializer before 'enable_if'
.pio/libdeps/esp32dev/Firebase Arduino Client Library for ESP8266 and ESP32/src/session/FB_Session.h:723:25: error: expected nested-name-specifier before 'enable_if'
   auto to() -> typename enable_if<is_same<T, FirebaseJsonArray>::value, FirebaseJsonArray &>::type
                         ^~~~~~~~~
.pio/libdeps/esp32dev/Firebase Arduino Client Library for ESP8266 and ESP32/src/session/FB_Session.h:723:25: error: expected initializer before 'enable_if'
.pio/libdeps/esp32dev/Firebase Arduino Client Library for ESP8266 and ESP32/src/session/FB_Session.h:729:25: error: expected nested-name-specifier before 'enable_if'
   auto to() -> typename enable_if<is_same<T, MB_VECTOR<uint8_t> *>::value, MB_VECTOR<uint8_t> *>::type
                         ^~~~~~~~~
.pio/libdeps/esp32dev/Firebase Arduino Client Library for ESP8266 and ESP32/src/session/FB_Session.h:729:25: error: expected initializer before 'enable_if'
.pio/libdeps/esp32dev/Firebase Arduino Client Library for ESP8266 and ESP32/src/session/FB_Session.h:736:25: error: expected nested-name-specifier before 'enable_if'
   auto to() -> typename enable_if<is_same<T, fs::File>::value, fs::File>::type
                         ^~~~~~~~~
.pio/libdeps/esp32dev/Firebase Arduino Client Library for ESP8266 and ESP32/src/session/FB_Session.h:736:25: error: expected initializer before 'enable_if'
In file included from .pio/libdeps/esp32dev/Firebase Arduino Client Library for ESP8266 and ESP32/src/Firebase.h:74,
                 from .pio/libdeps/esp32dev/Firebase Arduino Client Library for ESP8266 and ESP32/src/Firebase_ESP_Client.h:42,
                 from src/firebase.h:4,
                 from src/main.cpp:25:
.pio/libdeps/esp32dev/Firebase Arduino Client Library for ESP8266 and ESP32/src/rtdb/FB_RTDB.h:1888:106: error: expected nested-name-specifier before 'enable_if'
   auto dataPushHandler(FirebaseData *fbdo, T1 path, T2 value, int priority_addr, bool async) -> typename enable_if<is_string<T1>::value && is_num_int<T2>::value, bool>::type
                                                                                                          ^~~~~~~~~
.pio/libdeps/esp32dev/Firebase Arduino Client Library for ESP8266 and ESP32/src/rtdb/FB_RTDB.h:1888:106: error: expected initializer before 'enable_if'
.pio/libdeps/esp32dev/Firebase Arduino Client Library for ESP8266 and ESP32/src/rtdb/FB_RTDB.h:1894:106: error: expected nested-name-specifier before 'enable_if'
   auto dataPushHandler(FirebaseData *fbdo, T1 path, T2 value, int priority_addr, bool async) -> typename enable_if<is_string<T1>::value && is_bool<T2>::value, bool>::type
                                                                                                          ^~~~~~~~~
.pio/libdeps/esp32dev/Firebase Arduino Client Library for ESP8266 and ESP32/src/rtdb/FB_RTDB.h:1894:106: error: expected initializer before 'enable_if'
.pio/libdeps/esp32dev/Firebase Arduino Client Library for ESP8266 and ESP32/src/rtdb/FB_RTDB.h:1900:106: error: expected nested-name-specifier before 'enable_if'
   auto dataPushHandler(FirebaseData *fbdo, T1 path, T2 value, int priority_addr, bool async) -> typename enable_if<is_string<T1>::value && is_same<T2, float>::value, bool>::type
                                                                                                          ^~~~~~~~~
.pio/libdeps/esp32dev/Firebase Arduino Client Library for ESP8266 and ESP32/src/rtdb/FB_RTDB.h:1900:106: error: expected initializer before 'enable_if'
[... 242 lines cut ...]
        if (fbdo.dataType() == "float") {
          floatValue = fbdo.floatData();
          Serial.println(floatValue);
        }
      }
      else {
        Serial.println(fbdo.errorReason());
      }
    }
  }
}

void firebaseWriteData(String topic, void* value, char* datatype)
{
    if (Firebase.ready() && signupOK)
    {
        if (strcmp(datatype, "String") == 0)
        {
            Firebase.RTDB.setString(&fbdo, topic, (char*)value);
        }
    }

}
```
Just to reiterate, if I remove the line #include <Firebase_ESP_Client.h> from my header file, I don’t get these errors. However, I need this line to make the FirebaseData fbdo object global (so that I can read and write data to the database from other files). How can I get around this issue?

mobizt, 2022-10-26:
You can't use firebase.h and firebnase.cpp because Firebase.h and Firebase.cpp included in the library.
No matter what the first letter of your filename is small letter.

You should try other file names to prevent ambiguous.

closed 2022-10-26 by mobizt

MatthewGerges2, 2022-10-26:
I changed the file names to databaseCom.h and databaseCom.cpp and I'm still getting the same output errors even after doing a full clean on platformIO and trying to reupload the code. I don't think the file names were the issue. Do you have any other idea on how I can try to fix it?

mobizt, 2022-10-26:
I can't help. You have to debug your code yourself as it may beak the C++ rules