mobizt/Firebase-ESP-Client #422: Incomplete files when downloading from Firebase Storage to SPIFFS

lsalin, 2022-11-07:
**Describe the bug**
When downloading files from Firebase Storage to ESP32 the resulting file has a size that is a multiple of the chunk size and the rest of the file is discarded. Sometimes the resulting file is 0 bytes long.

**To Reproduce**
```c++
void fcsDownloadCallback(FCS_DownloadStatusInfo info)
{
    if (info.status == fb_esp_fcs_download_status_init)
    {
        Serial.printf("Downloading file %s (%d) to %s\n", info.remoteFileName.c_str(), info.fileSize, info.localFileName.c_str());
    }
}

void loop()
{
    if (Firebase.ready() && !taskCompleted)
    {
        taskCompleted = true;
        Serial.println("\nDownload file...\n");

        if (!Firebase.Storage.download(&fbdo, STORAGE_BUCKET_ID , "test.txt" , "test.txt" , mem_storage_type_flash, fcsDownloadCallback))
          Serial.println(fbdo.errorReason());
        
        File file = SPIFFS.open("/test.txt");
        if(!file){
          Serial.println("Failed to open file for reading");
        } else {
          Serial.print("File size: ");
          Serial.println(file.size());
        }
    }
}
```
usually prints  
`Downloading file test.txt (17896) to /test.txt`  
`File size: 16384`  

(16384 = 8 * 2048)

or sometimes 

`Downloading file test.txt (17896) to /test.txt`  
`File size: 17792`

**Expected behavior**
File sizes should match.

mobizt, 2022-11-07:
I'm pretty sure that there are no such bugs about the filesystems which intensively tested for several months before publishing almost years ago with the well managed internal file wrapper class called MB_FS.h.

I did not see such issue even I test today.

mobizt, 2022-11-07:
You should debug WiFi and SSL in Arduino IDE by configure the debug options.
Any internet interruption will cause the session terminated.

mobizt, 2022-11-07:
The byte that you can be download can't be exceed your device free space of flash filesystem.

The library was able to upload and download 50 to 100 MB file in SD card without errors.

mobizt, 2022-11-07:
The MB_FS class also used in other my libraries included [this library](https://github.com/mobizt/ESP-Mail-Client) which always work on filesystem storage.

That is why I'm pretty sure bugs free.

lsalin, 2022-11-07:
The strange thing is that OTA updates work perfectly. Regarding the file size and available free flash space, I have tested larger files to work but smaller ones to fail. (for example a 4096 byte long file worked every time but 147 byte long file got cut off to 128 bytes long). This seems to be related to the connection packet size rather than the filesystem itself. I'll do some digging around to see if the issue is somehow related to the network reliability but that seems unlikely since everything else works perfectly. Thanks for the help!

mobizt, 2022-11-07:
It's weird as I use it for upload/download various size of email attachment in Email library included few bytes to megabytes without issue.

mobizt, 2022-11-07:
You are right. I'll keep the issue open.

The filesystem wrapper class is ok but something wrong in the Firebase library which I'm checking.

mobizt, 2022-11-08:
The issue related to the ESP32 Core SDK v2.0.2 and newer related to the filesystem internal changes which causes the memory issue when File class was used in other functions.

The library works well in ESP3232 Core SDK v2.0.1 and older.

The issue is now fixed, and I have committed the code change to v4.2.4 which you can try.

 The library released version 4.2.4 will be published after I finish adding new authentication method support.

mobizt, 2022-11-09:
[v4.2.4](https://github.com/mobizt/Firebase-ESP-Client/releases/tag/v4.2.4) was released.

closed 2022-11-09 by mobizt