mobizt/Firebase-ESP-Client #434: MB_ESP32_SSLClient Error

SweiLz, 2022-11-12:
I am trying to use an external client with ethernet from this example. I tried running `ExternalClient/RTDB/Ethernet/ESP32/Basic/Basic.ino` but found below problem.

```
Resetting WIZnet W5500 Ethernet Board...  
Starting Ethernet connection...
Connected with IP 10.42.0.30
Firebase Client v4.2.6

Token info: type = id token (GITKit token), status = on request
Connecting to server via external Client... [ 10294][I][ESP32_SSL_Client.cpp:201] connect_ssl(): WARNING: Skipping SSL Verification. INSECURE!
[ 10296][E][ESP32_SSL_Client.cpp:61] _esp32_ssl_handle_error(): [connect_ssl():341]: (-1) ERROR - Generic error
[ 10301][E][MB_ESP32_SSLClient.cpp:219] connect(): MB_ESP32_SSLClient Error: upgrade connection, -1
failed.
```

This is my code
```
#include <Arduino.h>
#include <Firebase_ESP_Client.h>
#include <addons/TokenHelper.h>
#include <addons/RTDBHelper.h>

#include <Ethernet.h>
#include "sslclient/esp32/MB_ESP32_SSLClient.h"
#include "MB_NTP.h"

uint8_t Eth_MAC[] = {0x02, 0xF0, 0x0D, 0xBE, 0xEF, 0x01};

FirebaseData fbdo;
FirebaseAuth auth;
FirebaseConfig config;

unsigned long sendDataPrevMillis = 0;
int count = 0;
volatile bool dataChanged = false;

EthernetClient basic_client;
MB_ESP32_SSLClient ssl_client;

EthernetUDP udpClient;
MB_NTP ntpClient(&udpClient, "pool.ntp.org", 123 , 0 );

uint32_t timestamp = 0;

void ResetEthernet()
{
    Serial.println("Resetting WIZnet W5500 Ethernet Board...  ");
    pinMode(21, OUTPUT);
    digitalWrite(21, HIGH);
    delay(200);
    digitalWrite(21, LOW);
    delay(50);
    digitalWrite(21, HIGH);
    delay(200);
}

void networkConnection()
{
    Ethernet.init(5);
    ResetEthernet();
    Serial.println("Starting Ethernet connection...");
    Ethernet.begin(Eth_MAC);

    unsigned long to = millis();
    while (Ethernet.linkStatus() == LinkOFF || millis() - to < 2000)
    {
        delay(100);
    }
    if (Ethernet.linkStatus() == LinkON)
    {
        Serial.print("Connected with IP ");
        Serial.println(Ethernet.localIP());
    }
    else
    {
        Serial.println("Can't connected");
    }
}

void networkStatusRequestCallback()
{
    fbdo.setNetworkStatus(Ethernet.linkStatus() == LinkON);
}

void tcpConnectionRequestCallback(const char *host, int port)
{
    if (timestamp == 0)
    {
        timestamp = ntpClient.getTime(2000 /* wait 2000 ms */);

        if (timestamp > 0)
            Firebase.setSystemTime(timestamp);
    }

    Serial.print("Connecting to server via external Client... ");
    if (!ssl_client.connect(host, port))
    {
        Serial.println("failed.");
        return;
    }
    Serial.println("success.");
}

void setup()
{
    Serial.begin(115200);
    networkConnection();

    Serial_Printf("Firebase Client v%s\n\n", FIREBASE_CLIENT_VERSION);

    ssl_client.setClient(&basic_client);
    ssl_client.setInsecure();

    config.api_key = FIREBASE_API_KEY;
    auth.user.email = FIREBASE_USER_EMAIL;
    auth.user.password = FIREBASE_USER_PASSWORD;
    config.database_url = FIREBASE_DATABASE_URL;
    config.token_status_callback = tokenStatusCallback; 

    fbdo.setExternalClient(&ssl_client);
    fbdo.setExternalClientCallbacks(tcpConnectionRequestCallback, networkConnection, networkStatusRequestCallback);

    Firebase.reconnectWiFi(true);
    Firebase.setDoubleDigits(5);
    Firebase.begin(&config, &auth);
}

void loop()
{
    if (Firebase.ready() && (millis() - sendDataPrevMillis > 15000 || sendDataPrevMillis == 0))
    {
        sendDataPrevMillis = millis();
        Serial_Printf("Set bool... %s\n", Firebase.RTDB.setBool(&fbdo, F("/test/bool"), count % 2 == 0) ? "ok" : fbdo.errorReason().c_str());
        count++;
    }
}
```

**IDE and its version:**
 - PlatformIO
 - Version 1.73.1
 - ESP32 Espressif 32 Version 5.2.0

mobizt, 2022-11-12:
No error like that from intensively tested.

Check your internet connection.

mobizt, 2022-11-12:
Please use the discussion first.

closed 2022-11-12 by mobizt

mobizt, 2022-11-12:
At this time, I am investigating the MB_ESP32_SSLClient.h and openslab SSLClient which have similar issue when they work with Arduino Client which some unexpected behavior of socket close connection in user defined Client class which the SSL engine itself works very well for handling SSL/TLS which will be confirmed when WiFiSSLClient or WiFiClientSecure class was used as external client, 

You can use openslab SSLClient to test in ESP32 at this moment during issue investigation.

And I can confirm that this is not library issue unless external SSL client reliability issue.