mobizt/Firebase-ESP-Client #444: settings for google cloud

madlimez, 2022-11-29:
hi, i try upload file to google cloud(just google cloud not firebase). At google cloud APIs page i created api key, client id and service account but these informations do not match the code.

i have bucket id but just id there is no 'appspot.com' part
/* 2. Define the Firebase storage bucket ID e.g bucket-name.appspot.com */
#define STORAGE_BUCKET_ID "xxxxxxxx"

and how can i get these ones
/* 3 The following Service Account credentials required for OAuth2.0 authen in Google Cloud Storage JSON API upload */
#define FIREBASE_PROJECT_ID "xxxxx"
#define FIREBASE_CLIENT_EMAIL "xxxxx"
const char PRIVATE_KEY[] PROGMEM = "-----BEGIN PRIVATE KEY-----xxxxxxxxx-----END PRIVATE KEY-----\n";

mobizt, 2022-11-29:
As this library was designed for Firebase to extend to use the Google Cloud Storage, that is why you will see `/* 2. Define the Firebase storage bucket ID e.g bucket-name.appspot.com */` in that example.

Which Firebase Storage is actually the Google Cloud Storage on the backend.

As usual, you can assign your Google Cloud Storage bucket id, Project ID and Client email for the credentials.

mobizt, 2022-11-29:
You should post your problem in the discussion section instead.

The issue section is for issue tracking.

closed 2022-11-29 by mobizt