mobizt/Firebase-ESP-Client #454: Two GetString

Peluca86, 2022-12-14:
**Describe the bug**
when I use the getString function it only lets me receive one. Why when I try to use that same function for the second time it appears in the unknown error serial

**Screenshots**
[image]
[image]

I use Esp8266 - 01
My version to arduino is Arduino IDE 2.0.4

How can I solve this problem?

mobizt, 2022-12-15:
You have problem with your logic. When you call  fbdo.errorReason() without anything error, it will give unknow error.

closed 2022-12-15 by mobizt

mobizt, 2022-12-15:
If any function takes the fbdo as the argument and that function returns true from operation, if you call `fbdo.errorReason()` in somewhere, it will give the result of last operation that this FirebaseData object was used before.

That's why it shows `unknown error` as the last use of fbdo was success.

Your program logic on the second inner if clause that checking for `Firebase.ready()` and millis, if the condition doesn't meet you print the `fbdo.errorReason()` on serial which gives `"unknown error"` as the previous use of fbdo has no error.

Do not post this such programming error in the issue topic. You should post on the discussions instead.