mobizt/Firebase-ESP-Client #455: ENHANCEMENT

userdevil, 2022-12-22:
i am currently working in my college project i need to set my database secrets as default (database_url,api_key) how can i change that please help me

closed 2022-12-22 by mobizt

mobizt, 2022-12-23:
Your question is not clear.

If you want to create new project, manage the project included your credentials, you should go to [console](https://console.firebase.google.com/).

If you want to assign the new credentials in your code, change these lines with your credential strings.
https://github.com/mobizt/Firebase-ESP-Client/blob/d44dbb183f17baee95220e899b20cf1fb28eebd8/examples/RTDB/Basic/Basic.ino#L73-L80

You should follow the [examples](https://github.com/mobizt/Firebase-ESP-Client/tree/main/examples) and read all the comments in the examples for usage guidelines.

For basic and general programming questions, you should ask in the community forum instead.

userdevil, 2022-12-23:
config.api_key = API_KEY; 
  
 /* Assign the user sign in credentials */ 
 auth.user.email = USER_EMAIL; 
 auth.user.password = USER_PASSWORD; 
  
 /* Assign the RTDB URL (required) */ 
 config.database_url = DATABASE_URL;

In this example the user is assigning the database_url and api_key. Now I need only the user should assign the user sign in credentials. And the database_url and api_key should be assigned inside the library by default. And the user does not need to assign the  database_url and api_key. The user needs to assign only the user sign in credentials.

mobizt, 2022-12-23:
You should read the Google Firebase documentation for the authentication for the prerequisites instead of asking here.

userdevil, 2022-12-23:
Thank you for your response

mobizt, 2022-12-23:
The API key belongs to your project and it's important.
The database url belongs to your project and it's important too.

If you are new to Firebase services and APIs, you should spend the time to learn it.