mobizt/Firebase-ESP-Client #459: Intellisense Errors with Strings

djmick, 2023-01-01:
I'm using Visual Studio 2022 with [Visual Micro](https://www.visualmicro.com/) which is an Arduino IDE for Visual Studio (not to be confused with VS Code). i'm getting dozens of errors when trying to combine Strings together using +. For example String "s3 = s1 + s2". The error message i'm receiving is "more than one operator "+" matches these operands". This issue only occurs when i add the Firebase-ESP-Client to my project. Based on the error message below, it looks like it has to do with MB_String.

_Severity	Code	Description	Project	File	Line	Suppression State	Detail Description
Error (active)	E0350	more than one operator "+" matches these operands:	MyProject_ESP32_S2_Firebase	C:\Users\XXXX\Documents\Arduino\MyProject_ESP32_S2_Firebase\Firebase.ino	530		function "operator+(MB_String &&lhs, const MB_String &rhs)" (declared at line 2044 of "C:\Users\XXXX\Documents\Arduino\libraries\Firebase_Arduino_Client_Library_for_ESP8266_and_ESP32\src\json\MB_String.h")
function "operator+(const StringSumHelper &lhs, const String &rhs)" (declared at line 186 of "C:\Users\XXXX\AppData\Local\arduino15\packages\esp32\hardware\esp32\2.0.6\cores\esp32\WString.h")
operand types are: const char [2] + String_

Notice the yellow highlights in the screen shot below. Also "fbDbPath" is a global string variable
[image]

**IDE and its version:**
 - Arduino 2
 - Visual Studio 2022 ver 17.4.3 w/ [Visual Micro](https://www.visualmicro.com/) Plug-in

**ESP8266 Arduino Core SDK version**
 - Tried both 4.2.7 & 4.3.0

mobizt, 2023-01-01:
This is not code or library error.

It's not error specific to this library.

It's IDE error related to compiler as you can test with just simple Arduino String concatenation like this.

`String a = "something" + String(" is wrong");`

The above line causes the IDE intelligence error in Visual Studio IDE too.

This intellisense error also existed in VSCode + PlatformIO plugin too.

You can ignore this error in your IDE.

closed 2023-01-01 by mobizt

mobizt, 2023-01-01:
You can test with this simple sketch

```cpp
void setup()
{
  Serial.begin(115200);
  Serial.println("something" + String(" is wrong"));
}

void loop()
{
}
```

mobizt, 2023-01-01:
This intellisense issue also discussed [here](https://github.com/mobizt/Firebase-ESP-Client/discussions/440) (in VSCode not Visual Studio).

AndhikaWB, 2023-06-12:
It can be fixed by using `typedef` alias, but just use the alias for `+` operation only, other normal string should still be written as `String`.

```c++
typedef StringSumHelper str;
#define SSID "abcd"
#define KEY "1234"

void setup()
{
  Serial.begin(115200);

  Serial.println("SSID:" + (str) SSID);
  Serial.println("Password:" + (str) KEY);
}

void loop()
{
}
```