mobizt/Firebase-ESP-Client #461: Username & Password Refresh Token Issue in 4.3.x

djmick, 2023-01-07:
In versions 4.30 & 4.3.1 config.signer.preRefreshSeconds is not working with username & password authentication with RTDB. Then after 1 hour when the current token expires, config.token_status_callback is never called and i get permission denied errors. if i reboot, it works fine for the next hour.

If i rollback to V4.2.7, everything works as expected. I currently have config.signer.preRefreshSeconds to 3400 so the token refreshes after a couple of minutes. 

To summarize, same code in 4.2.7 has no issue but when upgrading to 4.3.x, preRefreshSeconds doesn't work and after 1 hour i get access denied errors.

mobizt, 2023-01-07:
It depends on how `Firebase.ready()` was called.
It should run in main loop, if some of your code blocks the `Firebase.ready()` to run in that loop then the token will not refresh in time and expired.

If you set `config.signer.preRefreshSeconds = 3600 - 60;` the token will be expired in 1 minute.

I am running the test intensively all the time to make sure the library is working fine, and I don't see the issue on this.

mobizt, 2023-01-07:
You should clean install the library to leave only one Firebase library in libraries folder, The manual zip installation and then using Arduino IDE Library Manager to update causes unexpected result at runtime due to the new version installed in folder `"Firebase Arduino Client Library for ESP8266 and ESP32"` while the manual zip installation stays in the folder `"Firebase-ESP-Client-main"` which is my fault in library properties file.

The problem occurred when compiler found the same file in multiples folders and choose the wrong old files for compilation

mobizt, 2023-01-07:
I confirm that everything works fine as I currently testing. If you get some strange behavior in such key function of this library (authentication) after library updated, please clean install the library.

djmick, 2023-01-07:
Yes Firebase.ready() is always being called.

I was running v4.2.7 and used the arduino library update feature to upgrade to 4.3.1. I complied the code and ran into the issues described. I then manually uninstalled 4.3.1 and installed 4.2.7 via arduino library install and then compiled. Same code started working.

The only thing i didn't do was install 4.3.1 from clean. I will try that now and follow up.

mobizt, 2023-01-07:
The code in v4.3.x is almost completely rewrite for optimization which the file name was unchanged from v4.2.7 unless the code inside.

djmick, 2023-01-07:
Here is my code i used for this test. Its using 4.3.1

`/**
 * Created by K. Suwatchai (Mobizt)
 *
 * Email: k_suwatchai@hotmail.com
 *
 * Github: https://github.com/mobizt/Firebase-ESP-Client
 *
 * Copyright (c) 2023 mobizt
 *
 */
#include <HttpsOTAUpdate.h>
#include <Update.h>
#include <esp_crt_bundle.h>
#include <ssl_client.h>
#include <WiFiClientSecure.h>
#include <ETH.h>
#include <SPI.h>
#include <SD.h>
#include <sd_defines.h>
#include <sd_diskio.h>
#include <FS.h>
#include <FSImpl.h>
#include <vfs_api.h>
#include <SPIFFS.h>
#include <FB_Const.h>
#include <FB_Error.h>
#include <FB_Network.h>
#include <FB_Utils.h>
#include <Firebase.h>
#include <FirebaseFS.h>
#include <MB_NTP.h>
#include <Arduino.h>
#if defined(ESP32) || defined(PICO_RP2040)
#include <WiFi.h>
#elif defined(ESP8266)
#include <ESP8266WiFi.h>
#endif

#include <Firebase_ESP_Client.h>

 // Provide the token generation process info.
#include <addons/TokenHelper.h>

// Provide the RTDB payload printing info and other helper functions.
#include <addons/RTDBHelper.h>

/* 1. Define the WiFi credentials */
#define WIFI_SSID "xxxxx"
#define WIFI_PASSWORD "xxxxx"

// For the following credentials, see examples/Authentications/SignInAsUser/EmailPassword/EmailPassword.ino

/* 2. Define the API Key */
#define API_KEY "xxxxx"

/* 3. Define the RTDB URL */
#define DATABASE_URL "xxxxx" //<databaseName>.firebaseio.com or <databaseName>.<region>.firebasedatabase.app

/* 4. Define the user Email and password that alreadey registerd or added in your project */
#define USER_EMAIL "xxxxx"
#define USER_PASSWORD "xxxxx"

// Define Firebase Data object
FirebaseData stream;
FirebaseData fbdo;

FirebaseAuth auth;
FirebaseConfig config;
[... 359 lines cut ...]
01/07/2023 02:57:18 PM - Stream Timeout Callback

01/07/2023 02:57:48 PM - Stream Timeout Callback

01/07/2023 02:58:18 PM - Stream Timeout Callback

01/07/2023 02:58:48 PM - Stream Timeout Callback

01/07/2023 02:59:18 PM - Stream Timeout Callback

01/07/2023 02:59:36 PM - Stream Callback
sream path, /users/MY_FIREBASE_USER_ID/app
event path, /
data type, json
event type, patch

Pretty printed JSON data:
{
	"d2_isLocked": true
}
Iterate JSON data:

0, Type: object, Name: d2_isLocked, Value: true

Received stream payload size: 48 (Max. 780)

mobizt, 2023-01-08:
Thanks for addressing this bug as it related to internal timestamp checking.

These lines cause the unchanged timestamp obtained from the `getTime` function.
https://github.com/mobizt/Firebase-ESP-Client/blob/04eff0e3ad5713a3dac336b91ad389673c8ff4c9/src/FB_Utils.h#L1691-L1692

The library [v4.3.2](https://github.com/mobizt/Firebase-ESP-Client/releases/tag/v4.3.2) is now available which fixed this bug, please update.

djmick, 2023-01-08:
Thanks. I'm trying it now

djmick, 2023-01-08:
So far everything seems to be working. Thank you for the prompt fix & updated release.

closed 2023-01-08 by djmick