mobizt/Firebase-ESP-Client #468: Repeating Error Code -3, Not Connected using v4.3.3

djmick, 2023-01-21:
After initial connection to RTDB using v4.3.3 i'm repeatedly  getting "stream timed out, resuming..." followed by "error code: -3, reason: not connected" every 3 seconds. As you'll see in my example code, Firebase.ready() is being called repeatedly.

Opening port
Port open
Connecting to Wi-Fi.
Connected with IP: 10.0.0.121

--------------------
Current Boot Time (EDT): 01/21/2023 04:11:37 PM
--------------------

Firebase Client v4.3.3

***************
01/21/2023 04:11:37 PM
Token info: type = id token (GITKit token), status = on request
***************

***************
01/21/2023 04:11:38 PM
Token info: type = id token (GITKit token), status = ready
***************

RTDB User Path: /users/xxxxxxxxxxxxxxxxxxxxxxxxxxxx
01/21/2023 04:11:38 PM
***************
01/21/2023 04:11:39 PM
stream timed out, resuming...

***************
sream path, /users/xxxxxxxxxxxxxxxxxxxxxxxxxxxx
event path, /
data type, json
event type, put

Pretty printed JSON data:
{
	"dht": {
		"humitity": "19%",
		"lastUpdated": 1674330519425,
		"temperature": "71°F"
	},
	"serverTimeStamp": 1674333984382
}
Iterate JSON data:

0, Type: object, Name: dht, Value: {"humitity":"19%","lastUpdated":1674330519425,"temperature":"71°F"}
1, Type: object, Name: humitity, Value: "19%"
2, Type: object, Name: lastUpdated, Value: 1674330519425
3, Type: object, Name: temperature, Value: "71°F"
4, Type: object, Name: serverTimeStamp, Value: 1674333984382

Received stream payload size: 147 (Max. 147)

***************
01/21/2023 04:11:43 PM
stream timed out, resuming...

error code: -3, reason: not connected
***************

***************
01/21/2023 04:11:46 PM
stream timed out, resuming...

error code: -3, reason: not connected
***************

***************
01/21/2023 04:11:50 PM
[... 259 lines cut ...]
        Serial.println(&timeinfo, "%m/%d/%Y %I:%M:%S %p");
        Serial.println("--------------------");
        Serial.println();
    }
}

void printCurrentTime(bool newLine) {
    time_t now;
    struct tm timeinfo;

    //Serial.println();
   
    if (!getLocalTime(&timeinfo)) {
        Serial.println("Failed to obtain time");
        return;
    }
    else if (newLine) {
        Serial.println(&timeinfo, "%m/%d/%Y %I:%M:%S %p");
    }
    else {
        Serial.print(&timeinfo, "%m/%d/%Y %I:%M:%S %p");
    }
}

`

mobizt, 2023-01-21:
The library works normally and there is nothing updated that involved this issue.
This is not the library issue unless your network issue which you should use other AP to access the internet or reset your router or change your device or power supply.

mobizt, 2023-01-21:
I will check again and update.

djmick, 2023-01-21:
I'm seeing a clear difference between 4.3.2 & 4.3.3

**here's serial log using 4.3.2**
Opening port
Port open

Connected with IP: 10.0.0.121

--------------------
Current Boot Time (EDT): 01/21/2023 05:37:03 PM
--------------------

Firebase Client v4.3.2

***************
01/21/2023 05:37:03 PM
Token info: type = id token (GITKit token), status = on request
***************

***************
01/21/2023 05:37:08 PM
Token info: type = id token (GITKit token), status = ready
***************

RTDB User Path: /users/xxxxxxxxxxxxxxxxxxxxxxxxxxxx
01/21/2023 05:37:08 PM
***************
01/21/2023 05:37:09 PM
stream timed out, resuming...

***************

sream path, /users/xxxxxxxxxxxxxxxxxxxxxxxxxxxx
event path, /
data type, json
event type, put

Pretty printed JSON data:
{
	"dht": {
		"humitity": "19%",
		"lastUpdated": 1674330519425,
		"temperature": "71°F"
	},
	"serverTimeStamp": 1674333984382
}
Iterate JSON data:

0, Type: object, Name: dht, Value: {"humitity":"19%","lastUpdated":1674330519425,"temperature":"71°F"}
1, Type: object, Name: humitity, Value: "19%"
2, Type: object, Name: lastUpdated, Value: 1674330519425
3, Type: object, Name: temperature, Value: "71°F"
4, Type: object, Name: serverTimeStamp, Value: 1674333984382

Received stream payload size: 147 (Max. 147)

***************
01/21/2023 05:37:39 PM
***************

***************
01/21/2023 05:38:09 PM
Token info: type = id token (GITKit token), status = on refreshing
***************

***************
01/21/2023 05:38:14 PM
Token info: type = id token (GITKit token), status = ready
***************

***************
01/21/2023 05:38:14 PM
***************

***************
01/21/2023 05:38:39 PM
***************

***************
01/21/2023 05:39:09 PM
***************

***************
01/21/2023 05:39:15 PM
Token info: type = id token (GITKit token), status = on refreshing
***************

***************
01/21/2023 05:39:20 PM
Token info: type = id token (GITKit token), status = ready
***************

**same code after clean install of 4.3.3**
Opening port
Port open

Connected with IP: 10.0.0.121

--------------------
Current Boot Time (EDT): 01/21/2023 05:42:07 PM
--------------------

Firebase Client v4.3.3

***************
01/21/2023 05:42:07 PM
Token info: type = id token (GITKit token), status = on request
***************

***************
01/21/2023 05:42:08 PM
Token info: type = id token (GITKit token), status = ready
***************

RTDB User Path: /users/xxxxxxxxxxxxxxxxxxxxxxxxxxxx
01/21/2023 05:42:08 PM
***************
01/21/2023 05:42:09 PM
stream timed out, resuming...

***************

sream path, /users/xxxxxxxxxxxxxxxxxxxxxxxxxxxx
event path, /
data type, json
event type, put

Pretty printed JSON data:
{
	"dht": {
		"humitity": "19%",
		"lastUpdated": 1674330519425,
		"temperature": "71°F"
	},
	"serverTimeStamp": 1674333984382
}
Iterate JSON data:

0, Type: object, Name: dht, Value: {"humitity":"19%","lastUpdated":1674330519425,"temperature":"71°F"}
1, Type: object, Name: humitity, Value: "19%"
2, Type: object, Name: lastUpdated, Value: 1674330519425
3, Type: object, Name: temperature, Value: "71°F"
4, Type: object, Name: serverTimeStamp, Value: 1674333984382

Received stream payload size: 147 (Max. 147)

***************
01/21/2023 05:42:13 PM
stream timed out, resuming...

error code: -3, reason: not connected
***************

***************
01/21/2023 05:42:19 PM
stream timed out, resuming...

error code: -3, reason: not connected
***************

Port open
***************
01/21/2023 05:42:30 PM
Token info: type = id token (GITKit token), status = ready
***************

RTDB User Path: /users/xxxxxxxxxxxxxxxxxxxxxxxxxxxx
01/21/2023 05:42:30 PM
sream path, /users/xxxxxxxxxxxxxxxxxxxxxxxxxxxx
event path, /
data type, json
event type, put

Pretty printed JSON data:
{
	"dht": {
		"humitity": "19%",
		"lastUpdated": 1674330519425,
		"temperature": "71°F"
	},
	"serverTimeStamp": 1674333984382
}
Iterate JSON data:

0, Type: object, Name: dht, Value: {"humitity":"19%","lastUpdated":1674330519425,"temperature":"71°F"}
1, Type: object, Name: humitity, Value: "19%"
2, Type: object, Name: lastUpdated, Value: 1674330519425
3, Type: object, Name: temperature, Value: "71°F"
4, Type: object, Name: serverTimeStamp, Value: 1674333984382

Received stream payload size: 147 (Max. 147)

***************
01/21/2023 05:42:31 PM
stream timed out, resuming...

error code: -3, reason: not connected
***************

***************
01/21/2023 05:42:31 PM
stream timed out, resuming...

error code: -3, reason: not connected
***************

***************
01/21/2023 05:42:36 PM
stream timed out, resuming...

error code: -3, reason: not connected
***************

***************
01/21/2023 05:42:39 PM
stream timed out, resuming...

error code: -3, reason: not connected
***************

***************
01/21/2023 05:42:45 PM
stream timed out, resuming...

error code: -3, reason: not connected
***************

***************
01/21/2023 05:42:49 PM
stream timed out, resuming...

error code: -3, reason: not connected
***************

***************
01/21/2023 05:42:53 PM
stream timed out, resuming...

error code: -3, reason: not connected
***************

***************
01/21/2023 05:42:58 PM
stream timed out, resuming...

error code: -3, reason: not connected
***************

***************
01/21/2023 05:43:01 PM
stream timed out, resuming...

error code: -3, reason: not connected
***************

mobizt, 2023-01-22:
The [library update](https://github.com/mobizt/Firebase-ESP-Client/releases/tag/v4.3.4) is available, please update.

closed 2023-01-22 by mobizt