mobizt/Firebase-ESP-Client #473: NTP server time synching failed

abdelrahman99999, 2023-01-29:
I use the RTDB and firebase storage.
after using server root certificate, I get this error 

Token info: type = id token (GITKit token), status = error
Token error: code: -136, message: NTP server time synching failed

then delay some time. then continue normally. 

Token info: type = id token (GITKit token), status = on request
Token info: type = id token (GITKit token), status = ready

why this error happens, how can i fix it?

mobizt, 2023-01-29:
It's normal information what library works behind for authentication process.

The valid time is required by ESP8266 WiFiClientSecure library for SSL certificate validation for certificate expiration, that is why the NTP server time synching is started first.

In ESP32, the WiFiClientSecure library does not check for SSL validation, then the NTP time synching process will be ignored.

If the time was not synched in specific waiting period, that error information will show, and the time synching process will start over.

The NTP server time synching can be done by user code outside the library too by using `configTime` function and user needs to wait until the response from NTP server was received.

mobizt, 2023-01-29:
If you have the question or problem, please post in discussion instead.

closed 2023-01-29 by mobizt