mobizt/Firebase-ESP-Client #474: DownloadFileOTA returns "Download firmware failed, Not Found."

bjdshp, 2023-01-30:
Attempted to incorporate in my code, but failed.  Using example provided, same error without modifications.  I placed a file named "firmware.bin" in the root folder of Firebase Storage.

 The file location is at gs://MYBUCKET-NAME.appspot.com/firmware.bin.

I opened permissions completely with the following rule.

match /{allPaths=**} {
       allow read;
       }

Here is response.

15:50:57.203 -> .................
15:51:02.287 -> Connected with IP: xxx.xxx.x.xxx
15:51:02.287 -> 
15:51:02.287 -> Firebase Client v4.3.5
15:51:02.287 -> 
15:51:02.287 -> Token info: type = id token (GITKit token), status = on request
15:51:09.178 -> Token info: type = id token (GITKit token), status = ready
15:51:09.178 -> 
15:51:09.178 -> Download firmware file...
15:51:09.178 -> 
15:51:12.648 -> Download firmware failed, Not Found.
15:51:12.648 -> Not Found.

mobizt, 2023-01-30:
It's not related to library but user side error.

The library works normally for both Google Cloud Storage and Firebase Storage.

The error "not found" is HTTP 404 Error returns from Google storage server.

It's not about permission, unless the file is not found. 

You should check your file name and extension or make sure you provide the valid storage bucket name or id that belong to your project.

closed 2023-01-30 by mobizt

mobizt, 2023-01-30:
Make sure, the scheme `gs://` was not included in your storage bucket name or id
https://github.com/mobizt/Firebase-ESP-Client/blob/b79a7412cc9b5c15893503a4304640221a14196c/examples/Storage/FirebaseStorage/DownloadFileOTA/DownloadFileOTA.ino#L37-L38

bjdshp, 2023-01-30:
confirmed it is not.  Tested on separate Firebase project with same results.

[image]

mobizt, 2023-01-30:
The result from my test now.

```yaml
Connecting to Wi-Fi..........................................
Connected with IP: 192.168.137.50

Firebase Client v4.3.5

Token info: type = id token (GITKit token), status = on request
Token info: type = id token (GITKit token), status = ready

Download firmware file...

Downloading firmware esp32.bin (671408)
Downloaded 0%, Elapsed time 18277 ms
Downloaded 2%, Elapsed time 18917 ms
Downloaded 4%, Elapsed time 19830 ms
Downloaded 6%, Elapsed time 20730 ms
Downloaded 8%, Elapsed time 21651 ms
Downloaded 10%, Elapsed time 22296 ms
Downloaded 12%, Elapsed time 23206 ms
Downloaded 14%, Elapsed time 24108 ms
Downloaded 16%, Elapsed time 25032 ms
Downloaded 18%, Elapsed time 25939 ms
Downloaded 20%, Elapsed time 26896 ms
Downloaded 22%, Elapsed time 27802 ms
Downloaded 24%, Elapsed time 29012 ms
Downloaded 26%, Elapsed time 29966 ms
Downloaded 28%, Elapsed time 31166 ms
Downloaded 30%, Elapsed time 32226 ms
Downloaded 32%, Elapsed time 33038 ms
Downloaded 34%, Elapsed time 34239 ms
Downloaded 36%, Elapsed time 35161 ms
Downloaded 38%, Elapsed time 36081 ms
Downloaded 40%, Elapsed time 36827 ms
Downloaded 42%, Elapsed time 37933 ms
Downloaded 44%, Elapsed time 38866 ms
Downloaded 46%, Elapsed time 39632 ms
Downloaded 48%, Elapsed time 40718 ms
Downloaded 50%, Elapsed time 41610 ms
Downloaded 52%, Elapsed time 42291 ms
Downloaded 54%, Elapsed time 43212 ms
Downloaded 56%, Elapsed time 44075 ms
Downloaded 58%, Elapsed time 44976 ms
Downloaded 60%, Elapsed time 45942 ms
Downloaded 62%, Elapsed time 46854 ms
Downloaded 64%, Elapsed time 47763 ms
Downloaded 66%, Elapsed time 48685 ms
Downloaded 68%, Elapsed time 49620 ms
Downloaded 70%, Elapsed time 50520 ms
Downloaded 72%, Elapsed time 51442 ms
Downloaded 74%, Elapsed time 52053 ms
Downloaded 76%, Elapsed time 53016 ms
Downloaded 78%, Elapsed time 53909 ms
Downloaded 80%, Elapsed time 54819 ms
Downloaded 82%, Elapsed time 55744 ms
Downloaded 84%, Elapsed time 56422 ms
Downloaded 86%, Elapsed time 57288 ms
Downloaded 88%, Elapsed time 58202 ms
Downloaded 90%, Elapsed time 58844 ms
Downloaded 92%, Elapsed time 59744 ms
Downloaded 94%, Elapsed time 60681 ms
Downloaded 96%, Elapsed time 61600 ms
Downloaded 98%, Elapsed time 62259 ms
Downloaded 100%, Elapsed time 63189 ms
Update firmware completed.

Restarting...

ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x37 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13104
load:0x40080400,len:3036
entry 0x400805e4
```

bjdshp, 2023-01-30:
thanks....I'm glad it works.  Can't diagnose failure.

mobizt, 2023-01-30:
Create new google account or new project may not solve your problem because the server can't find the file name you provide.

You should try with upload and download examples first to make sure that your information is correct.

mobizt, 2023-01-30:
It works in all my devices i.e., ESP8266, ESP32 and Raspberry Pi Pico W with all my Firebase projects.

bjdshp, 2023-01-30:
Ha, sorry for wasting your time.  Had to take the chevrons out of "<firmware.bin>" to make it "firmware.bin".

Keep you the good work!

mobizt, 2023-01-30:
It's a markup used in the library documentation and examples represents the specific data that user should replace with his own or type of function parameters.