mobizt/Firebase-ESP-Client #475: Esp32 firestore without auth

KJPOUNDY132, 2023-02-04:
Hi i want to use firestore without email auth, is it possible?

mobizt, 2023-02-04:
The authentication methods required to use Google Firebase services in this library are already stated in Readme doc and also in all examples.
https://github.com/mobizt/Firebase-ESP-Client/blob/b79a7412cc9b5c15893503a4304640221a14196c/examples/Firestore/CreateDocuments/CreateDocuments.ino#L13

closed 2023-02-04 by mobizt

mobizt, 2023-02-04:
If you mean no auth or public access, please read [this](https://github.com/mobizt/Firebase-ESP-Client#firebase-cloud-firestore) for the security rules setting for public access.

And set the test mode for no auth in the config.

`config.signer.test_mode = true;`

KJPOUNDY132, 2023-02-04:
U say there is no way to use without auth like Service account keys (like python firestore)

mobizt, 2023-02-04:
Read what I reply again.

KJPOUNDY132, 2023-02-04:
Sorry i dont see it:)