mobizt/Firebase-ESP-Client #481: Compilation error: 'FirebaseStorage' does not name a type

Animesh452, 2023-02-17:
I am using AI Thinker ESP32 Cam board in Arduino IDE and getting the above compilation error. Please help resolve this.
Firebase ESP32 Client by Mobizt v4.3.7
[image]

mobizt, 2023-02-18:
Firebase ESP32 Client library supports RTDB and FCM only.

Firebase Arduino Client Library for ESP8266 and ESP32 library (this repo) supports RTDB, FCM, Firestore, Storage. Google Cloud Storage and Functions for Firebase.

Choose the correct library in Arduino Library Manager to install.

mobizt, 2023-02-18:
This library must be removed.
https://github.com/mobizt/Firebase-ESP32

closed 2023-02-18 by mobizt