mobizt/Firebase-ESP-Client #487: Can't upload file

dnshshkr, 2023-03-04:
This is your code in DownloadFile and it worked fine:
`if (!Firebase.Storage.download(&fbdo,STORAGE_BUCKET_ID , "file1.txt", "/file1.txt", mem_storage_type_flash, fcsDownloadCallback))
      Serial.println(fbdo.errorReason());`

This is your code in UploadFile:
`if (!Firebase.Storage.upload(&fbdo, STORAGE_BUCKET_ID, "/file1.txt", mem_storage_type_flash, "file1.txt", "text/plain", fcsUploadCallback))
      Serial.println(fbdo.errorReason());`

I followed every instruction for every function and change the mime type to "text/plain" but this is the response I got from serial monitor:
[image]

This is the file in my Firebase Storage:
[image]

dnshshkr, 2023-03-04:
Full DownloadFile code:
`/**
   Created by K. Suwatchai (Mobizt)

   Email: k_suwatchai@hotmail.com

   Github: https://github.com/mobizt/Firebase-ESP-Client

   Copyright (c) 2023 mobizt

*/

// This example shows how to download file from Firebase Storage bucket.

// If SD Card used for storage, assign SD card type and FS used in src/FirebaseFS.h and
// change the config for that card interfaces in src/addons/SDHelper.h

#include <Arduino.h>
#if defined(ESP32) || defined(ARDUINO_RASPBERRY_PI_PICO_W)
#include <WiFi.h>
#elif defined(ESP8266)
#include <ESP8266WiFi.h>
#endif

#include <Firebase_ESP_Client.h>

// Provide the token generation process info.
#include <addons/TokenHelper.h>

// Provide the SD card interfaces setting and mounting
#include <addons/SDHelper.h>

/* 1. Define the WiFi credentials */
#define WIFI_SSID "###"
#define WIFI_PASSWORD "###"

/* 2. Define the API Key */
#define API_KEY "###"

/* 3. Define the user Email and password that alreadey registerd or added in your project */
#define USER_EMAIL "###"
#define USER_PASSWORD "###"

/* 4. Define the Firebase storage bucket ID e.g bucket-name.appspot.com */
#define STORAGE_BUCKET_ID "###"

// Define Firebase Data object
FirebaseData fbdo;

FirebaseAuth auth;
FirebaseConfig config;

bool taskCompleted = false;

#if defined(ARDUINO_RASPBERRY_PI_PICO_W)
WiFiMulti multi;
#endif

void setup()
{

  Serial.begin(115200);

#if defined(ARDUINO_RASPBERRY_PI_PICO_W)
  multi.addAP(WIFI_SSID, WIFI_PASSWORD);
  multi.run();
#else
  WiFi.begin(WIFI_SSID, WIFI_PASSWORD);
#endif
  Serial.print("Connecting to Wi-Fi");
[... 247 lines cut ...]
  else if (info.status == fb_esp_fcs_upload_status_error)
  {
    Serial.printf("Upload failed, %s\n", info.errorMsg.c_str());
  }
}

void loop()
{

  // Firebase.ready() should be called repeatedly to handle authentication tasks.

  if (Firebase.ready() && !taskCompleted)
  {
    taskCompleted = true;

    Serial.println("\nUpload file...\n");

    // MIME type should be valid to avoid the download problem.
    // The file systems for flash and SD/SDMMC can be changed in FirebaseFS.h.
    if (!Firebase.Storage.upload(&fbdo, STORAGE_BUCKET_ID /* Firebase Storage bucket id */, "/file1.txt" /* path to local file */, mem_storage_type_flash /* memory storage type, mem_storage_type_flash and mem_storage_type_sd */, "file1.txt" /* path of remote stored in the bucket */, "text/plain" /* mime type */, fcsUploadCallback /* callback function */))
      Serial.println(fbdo.errorReason());
  }
}`
Serial monitor for UploadFile:
[image]

mobizt, 2023-03-04:
The update [v4.3.8](https://github.com/mobizt/Firebase-ESP-Client/releases/tag/v4.3.8) is available, please download.

dnshshkr, 2023-03-04:
Hi, it works now! You're wonderful THANK YOU!

closed 2023-03-04 by dnshshkr