mobizt/Firebase-ESP-Client #488: Getting unknown error from FirebaseData class

simzdev, 2023-03-04:
I wanted to make cleaner code for main class so I made another one and I am getting "unknown error" from instance of FirebaseData class. 
`class fireBase{
    public:
    FirebaseData fbdo;
    FirebaseAuth auth;
    FirebaseConfig config;

    fireBase(){
        config.api_key = API_KEY;
        config.database_url = DATABASE_URL;
        auth.user.email = GMAIL;
        auth.user.password = GMAIL_PASSWORD; 
    }

    void begin(bool reconnectBase=false){
        Firebase.begin(&config, &auth);
        config.token_status_callback = tokenStatusCallback;
        if(reconnectBase) Firebase.reconnectWiFi(true);
    }

    bool write(String path, String value){
        bool state = false;
        fbdo.pauseFirebase(false);
        if(Firebase.RTDB.setString(&fbdo, path, value)) state=true;
        else Serial.println("REASON: " + fbdo.errorReason()); //<<From here
        fbdo.pauseFirebase(true);
        return state;
    }

    bool write(String path, bool value){
        bool state = false;
        fbdo.pauseFirebase(false);
        if(Firebase.RTDB.setBool(&fbdo, path, value)) state=true;
        else Serial.println("REASON: " + fbdo.errorReason()); //<<From here
        fbdo.pauseFirebase(true);
        return state;
    }
};`

simzdev, 2023-03-04:
Problem begins when trying to unpause firebase. I am using ps4 controller in project and wifi and bluetooth can't work simultaneously, with paused wifi firebaseData ps4 controller can connect, although I am not sure why it works from start with only wifi library and webpage.

`fbdo.pauseFirebase(false);`

mobizt, 2023-03-04:
Unknown error means no error.

Call `fbdo.errorReason()` from the last operation that returns true will give unknow error as no error.

The issue occurred when you call pauseFirebase during no network or server connection and the internal pause status will not change after all.

I will fix this soon.

mobizt, 2023-03-04:
The update [v4.3.8](https://github.com/mobizt/Firebase-ESP-Client/releases/tag/v4.3.8) is available, please download.

closed 2023-03-04 by mobizt

simzdev, 2023-03-05:
Thank you, everything works as intended.