mobizt/Firebase-ESP-Client #489: We are having Error Of Missing Insufficiant Permission

moradiyainfo, 2023-03-08:
I m having error as per below while run On NODEMCU
Connecting to Wi-Fi.............
Connected with IP: 192.168.1.101

Firebase Client v4.3.8

Token info: type = id token (GITKit token), status = on request
Token info: type = id token (GITKit token), status = ready
List Collection Ids in a document... **Missing or insufficient permissions.**

Whole Program Is attached hear

Plz Help on that 

ALso Set Permission 

`1
rules_version = '2';
2
service cloud.firestore {
3
  match /databases/{database}/documents {
4
    match /{document=**} {
5
      allow read, write: if true;
6
    }
7
  }
8
}
`

mobizt, 2023-03-08:
It's not library related issue. 

There is something missing or misconfiguring in your Google cloud platform project.

You should make sure that your Firebase credentials that you provided are belong to your project.

You should create new project or even your Google account and try again.

If issue persisted, post your question in community forum or contact Google.

closed 2023-03-08 by mobizt

mobizt, 2023-03-08:
Try this default rules

```yaml
rules_version = '2';
service cloud.firestore {
  match /databases/{database}/documents {
    match /{document=**} {
      allow read, write: if
           request.auth != null;
    }
  }
}
```