mobizt/Firebase-ESP-Client #511: ESP32 Firebase RTDB Crashed after 30 minutes above

mrezzaaa, 2023-05-21:
**Describe the bug**
Hi, i have some errors when ESP32 is running above 30 minutes. Here's some log may be help
```
Received stream payload size: 221 (Max. 14323)

Token info: type = id token (GITKit token), status = on refreshing

abort() was called at PC 0x40193c0b on core 1

Backtrace:0x40083b59:0x3ffce6100x400956c5:0x3ffce630 0x4009b135:0x3ffce650 0x40193c0b:0x3ffce6d0 0x40193c52:0x3ffce6f0 0x4019362b:0x3ffce710 0x40193b3b:0x3ffce730 0x400d2c83:0x3ffce750 0x400d42b9:0x3ffce780 0x400d477c:0x3ffce8d0 0x401029d5:0x3ffce8f0 

  #0  0x40083b59:0x3ffce610 in panic_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/panic.c:402
  #1  0x400956c5:0x3ffce630 in esp_system_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/esp_system.c:128
  #2  0x4009b135:0x3ffce650 in abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/abort.c:46
  #3  0x40193c0b:0x3ffce6d0 in __cxxabiv1::__terminate(void (*)()) at /builds/idf/crosstool-NG/.build/HOST-x86_64-apple-darwin12/xtensa-esp32-elf/src/gcc/libstdc++-v3/libsupc++/eh_terminate.cc:47
  #4  0x40193c52:0x3ffce6f0 in std::terminate() at /builds/idf/crosstool-NG/.build/HOST-x86_64-apple-darwin12/xtensa-esp32-elf/src/gcc/libstdc++-v3/libsupc++/eh_terminate.cc:57
  #5  0x4019362b:0x3ffce710 in __cxa_throw at /builds/idf/crosstool-NG/.build/HOST-x86_64-apple-darwin12/xtensa-esp32-elf/src/gcc/libstdc++-v3/libsupc++/eh_throw.cc:95
  #6  0x40193b3b:0x3ffce730 in std::__throw_logic_error(char const*) at /builds/idf/crosstool-NG/.build/HOST-x86_64-apple-darwin12/xtensa-esp32-elf/src/gcc/libstdc++-v3/src/c++11/functexcept.cc:66 (discriminator 2)
  #7  0x400d2c83:0x3ffce750 in void std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_M_construct<char const*>(char const*, char const*, std::forward_iterator_tag) at /Users/icewalker/.platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2-patch3/xtensa-esp32-elf/include/c++/8.4.0/bits/basic_string.tcc:212
      (inlined by) void std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_M_construct_aux<char const*>(char const*, char const*, std::__false_type) at /Users/icewalker/.platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2-patch3/xtensa-esp32-elf/include/c++/8.4.0/bits/basic_string.h:236
      (inlined by) void std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_M_construct<char const*>(char const*, char const*) at /Users/icewalker/.platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2-patch3/xtensa-esp32-elf/include/c++/8.4.0/bits/basic_string.h:255
      (inlined by) std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::basic_string(char const*, std::allocator<char> const&) at /Users/icewalker/.platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2-patch3/xtensa-esp32-elf/include/c++/8.4.0/bits/basic_string.h:516
  #8  0x400d42b9:0x3ffce780 in EIM_BLE::calculateTime() at src/BLEModule.h:143
  #9  0x400d477c:0x3ffce8d0 in loop() at src/main.cpp:181
  #10 0x401029d5:0x3ffce8f0 in loopTask(void*) at /Users/icewalker/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:50
```
This error happened when stream callback called, when sometimes it's running well below 30 minutes.
**To Reproduce**
Create function that esp32 collecting data using BLE, read sensor data on characteristics and submitting into Realtime Database.
Clearing readed data every read (keeping heap/memory usage low)

**Expected behavior**
Running stream as long as possible ( Keep alive with unlimited time maybe)

**IDE and its version:**
 - PlatformIO with Arduino environment
 - Version [6.1.7]

**ESP32 Arduino Core SDK version**
 - Version [5.0.1]

**Additional context**
All running well under 30 mins. I think maybe some authentication or anything about connection to firebase(?). Just asuming why this happened.

mobizt, 2023-05-21:
It is not related to library issue. It is because your Bluetooth code error.

closed 2023-05-21 by mobizt

mobizt, 2023-05-21:
Checking your device free heap and debug your code first. There are debug options available in Arduino IDE.

mrezzaaa, 2023-05-21:
Free heap is about 30-20%. How you figured out its bluetooth error?

mobizt, 2023-05-21:
The stack information shows that the code stops and device crashed when executing calculateTime function from BLE library.

mrezzaaa, 2023-05-21:
Ok, i'll figured it out first.
I wanna ask to you too: 
Is there any conflicting issue if BLE scanning and connecting to other device at the same time with RTDB stream or streamCallback event?

mobizt, 2023-05-21:
In the calculateTime function, memory allocation may fail because of low memory.

You should not use Bluetooth and WiFi at the same time as both required large memory especially in SSL connection.

ESP32, shared the same RF channel for WiFi and Bluetooth which you can't use it at the same time in Arduino framework.

You should stop (disable) BLE to use WiFi and stop WiFi to use BLE.

Leave both running in the background is not recommend.

mrezzaaa, 2023-05-21:
Ok. I got it. Thank you very much about this information.