mobizt/Firebase-ESP-Client #517: Stream Timeout

pranjul-arctano, 2023-06-05:
Can anyone tell me how to set the stream timeout time, In my case I am getting timeout callback after every 15 seconds for few minutes when internet is not available, and after few minutes the callback comes after every 3 seconds. Is their any direct function to set it??

mobizt, 2023-06-05:
You should ask this such question by posting in discussions topics instead.

Do not touch the stream time out setting as it works properly with Firebase REST API even it can be changed.

https://github.com/mobizt/Firebase-ESP-Client/blob/d2fce37b28a29dfa37fa97e1ef37f87983d90a13/examples/RTDB/DataChangesListener/Callback/Callback.ino#L161-L183

Normally stream time out period is 45 seconds, and you should not get the stream timed out error unless, the internet connection was interrupted or unstable.  In case internet connection problems, you may get the stream timed out error in several seconds and more often in case WiFi or network disconnected.

closed 2023-06-05 by mobizt

pranjul-arctano, 2023-06-05:
But I want stream timeout instantly whenever internet is not their and it's timeout function to be called after every 1 sec if the device has no internet access.

mobizt, 2023-06-05:
You should use other method to test internet connection e.g. pinging or search Google for how to.

This library is Firebase client library using REST API.

pranjul-arctano, 2023-06-05:
ok thanks

mobizt, 2023-06-05:
It's not possible to detect the server connection status after client connected in Server-Sent Events http (stream) method.

It's different from web socket or other protocols e.g., MQTT, in stream, client waits for server sent event data (one way messaging) especially keep-alive event to determine that server was still alive.

If no keep-alive event data sent from server within the specific period, the server connection may be lost, and we need to close the connection to clear resource and start a new connection.

Please learn more about Server-Sent Events API.

mobizt, 2023-06-05:
FYI.
The device disconnected event is only available in Firebase client that runs on device that Firebase SDK was installed as it uses web socket.