mobizt/Firebase-ESP-Client #519: Unable to compile examples: 'TaskHandle_t' does not name a type

rincewind23, 2023-06-06:
Setup:
Arduino IDE 2.1.0
Board: Raspberry Pi Pico W from Earle Philhower's package v3.2.1
Firebase library: Firebase Arduino Client Library for ESP8266 and ESP32 by Mobizt v4.3.10

Trying to compile one of the examples (for example TestMode.ino) and getting this error:
```
In file included from c:\Users\Kev\Documents\Arduino\libraries\Firebase_Arduino_Client_Library_for_ESP8266_and_ESP32\src\Firebase.h:40,
from c:\Users\Kev\Documents\Arduino\libraries\Firebase_Arduino_Client_Library_for_ESP8266_and_ESP32\src\Firebase.cpp:38:
c:\Users\Kev\Documents\Arduino\libraries\Firebase_Arduino_Client_Library_for_ESP8266_and_ESP32\src\FB_Const.h:1033:5: error: 'TaskHandle_t' does not name a type
 1033 |     TaskHandle_t resumable_upload_task_handle = NULL;
      |     ^~~~~~~~~~~~
c:\Users\Kev\Documents\Arduino\libraries\Firebase_Arduino_Client_Library_for_ESP8266_and_ESP32\src\FB_Const.h:1034:5: error: 'TaskHandle_t' does not name a type
 1034 |     TaskHandle_t functions_check_task_handle = NULL;
      |     ^~~~~~~~~~~~
c:\Users\Kev\Documents\Arduino\libraries\Firebase_Arduino_Client_Library_for_ESP8266_and_ESP32\src\FB_Const.h:1035:5: error: 'TaskHandle_t' does not name a type
 1035 |     TaskHandle_t functions_deployment_task_handle = NULL;
      |     ^~~~~~~~~~~~
c:\Users\Kev\Documents\Arduino\libraries\Firebase_Arduino_Client_Library_for_ESP8266_and_ESP32\src\FB_Const.h:1037:5: error: 'TaskHandle_t' does not name a type
 1037 |     TaskHandle_t stream_task_handle = NULL;
      |     ^~~~~~~~~~~~
c:\Users\Kev\Documents\Arduino\libraries\Firebase_Arduino_Client_Library_for_ESP8266_and_ESP32\src\FB_Const.h:1038:5: error: 'TaskHandle_t' does not name a type
 1038 |     TaskHandle_t queue_task_handle = NULL;
      |     ^~~~~~~~~~~~```

I've tried including `FreeRTOS.h` and `task.h` but no success there either.

rincewind23, 2023-06-06:
I downgraded the Board package version to 3.1.0 and it compiles successfully now

closed 2023-06-06 by rincewind23

mobizt, 2023-06-07:
Thanks for reporting this issue. 

You now can use the latest Arduino Pico SDK with the latest version of library [v4.3.11](https://github.com/mobizt/Firebase-ESP-Client/releases/tag/v4.3.11).