mobizt/Firebase-ESP-Client #521: Heap-error calling FirebaseData::setExternalClient()

MartinXBcn, 2023-06-14:
Hello,

I just updated your library from version 4.3.10 to 4.3.12.
I am using an external client (TinyGsm) (ESP32; PlatformIO) which worked fine in version 4.3.10.

With the new version 4.3.12 I get the following heap-error when calling ``fbdoRtdb.setExternalClient(...)``:
```
CORRUPT HEAP: Bad head at 0x3ffd1368. Expected 0xabba1234 got 0x8c8fc300
assert failed: multi_heap_free multi_heap_poisoning.c:253 (head != NULL)

Backtrace: 0x40084055:0x3ffd90e0 0x400963d5:0x3ffd9100 0x4009c811:0x3ffd9120 0x4009c427:0x3ffd9250 0x40084519:0x3ffd9270 0x4009c841:0x3ffd9290 0x401da25d:0x3ffd92b0 0x40109a07:0x3ffd92d0 0x40203e62:0x3ffd92f0 0x4010dd65:0x3ffd9310 0x40109e30:0x3ffd9340 0x4013c096:0x3ffd9360 0x400d640a:0x3ffd9380 0x400d4f12:0x3ffd93c0

  #0  0x40084055:0x3ffd90e0 in panic_abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/panic.c:408
  #1  0x400963d5:0x3ffd9100 in esp_system_abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/esp_system.c:137
  #2  0x4009c811:0x3ffd9120 in __assert_func at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/newlib/assert.c:85
  #3  0x4009c427:0x3ffd9250 in multi_heap_free at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/heap/multi_heap_poisoning.c:253
      (inlined by) multi_heap_free at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/heap/multi_heap_poisoning.c:245
  #4  0x40084519:0x3ffd9270 in heap_caps_free at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/heap/heap_caps.c:361
  #5  0x4009c841:0x3ffd9290 in free at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/newlib/heap.c:39
  #6  0x401da25d:0x3ffd92b0 in operator delete(void*) at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32-elf/src/gcc/libstdc++-v3/libsupc++/del_op.cc:49
  #7  0x40109a07:0x3ffd92d0 in FB_Custom_TCP_Client::~FB_Custom_TCP_Client() at lib/MsFirebaseESPClient/src/./wcs/custom/FB_Custom_TCP_Client.h:50
  #8  0x40203e62:0x3ffd92f0 in Firebase_Signer::freeClient(FB_Custom_TCP_Client**) at lib/MsFirebaseESPClient/src/signer/Signer.cpp:852 (discriminator 1)
  #9  0x4010dd65:0x3ffd9310 in Firebase_Signer::setTCPClient(FB_Custom_TCP_Client*) at lib/MsFirebaseESPClient/src/signer/Signer.cpp:1605
  #10 0x40109e30:0x3ffd9340 in FirebaseData::setExternalClient(Client*) at lib/MsFirebaseESPClient/src/session/FB_Session.cpp:83
```
MsFirebaseESPClient is a fork of your library.

The only relevant difference between 4.3.10 and 4.3.12 I found is the elimination of
```
    if (noClient)
        freeClient(&tcpClient);
```
in ``bool Firebase_Signer::reconnect()``.

With re-inserting these two lines version 4.3.12 works for me with an external client again.

Regards, Martin

mobizt, 2023-06-14:
Thanks for reporting this bug, 

This is because I'm trying to avoid the annoying and unnecessary verbose debug message from ssl_client class "Cleaning the SSL connection" when its reconstructor was called even it does not use. I will fix and inform you soon for the update.

mobizt, 2023-06-15:
I update the library with bug fixes in [v4.3.13](https://github.com/mobizt/Firebase-ESP-Client/releases/tag/v4.3.13).

This version is based on your modification in the fork for header files inclusion/exclusion for reduce program space.

If no filesystems macro defined in `FirebaseFS.h`, the `FS.h` and `SPI.h` will not include.
If external client macro was defined in `FirebaseFS.h`, the `ETH.h` will not include in ESP32.

`FirebaseJson`'s `readClient` function is also updated.

closed 2023-06-15 by mobizt