mobizt/Firebase-ESP-Client #533: Timestamp example code gives me permission denied

ZubinAssadian, 2023-07-19:
When I try using the Timestamp.ino example in the library, I get Set timestamp....permission denied. The relevant output is below:

16:06:05.791 -> Set timestamp... Permission denied
16:06:05.791 -> Get timestamp... connection refused
16:06:05.791 -> Set data with timestamp... connection refused
16:06:05.791 -> Push data with timestamp... connection refused
16:06:05.791 -> Get previous pushed data... connection refused

I am running this on an ESP32 DevKit C.

Am I missing any additional set up that must be done in firebase RTDB rules or elsewhere?

My rules are setup as follows:

// These rules grant access to a node matching the authenticated
// user's ID from the Firebase auth token
{
  "rules": {
    "UsersData": {
      "$uid": {
        ".read": "$uid === auth.uid",
        ".write": "$uid === auth.uid"
      }
    }
  }
}

mobizt, 2023-07-20:
Your log in credentials is not valid.

Post your code and all debug messages.

closed 2023-07-20 by mobizt

archee29, 2024-04-01:
[image]

archee29, 2024-04-01:
how i solved this problem ??

[image]

this my code : 
#include <Arduino.h>
#include <ESP8266WiFi.h>
#include <Firebase_ESP_Client.h>
#include "addons/TokenHelper.h"
#include "addons/RTDBHelper.h"

#define WIFI_SSID ""
#define WIFI_PASSWORD ""
#define API_KEY ""
#define DATABASE_URL ""
#define USER_EMAIL ""
#define USER_PASSWORD ""

FirebaseData firebaseData;
FirebaseAuth auth;
FirebaseConfig config;
FirebaseJson jadwalPagiJson;

unsigned long sendDataPrevMillis = 0;
unsigned long timerDelay = 180000;

int count = 0;

String uid;
String databasePath;
String jadwalPagiNode;

void initWiFi(){
  WiFi.begin(WIFI_SSID, WIFI_PASSWORD);
  Serial.print("Connecting to WiFi.");
  while (WiFi.status() != WL_CONNECTED) {
    Serial.print('.');
    delay(1000);
  }
  Serial.println();
  Serial.print("Terhubung dengan IP ADDRESS : ");
  Serial.println(WiFi.localIP());  
}

void initFirebase(){

  config.api_key = API_KEY;  
  auth.user.email = USER_EMAIL;
  auth.user.password = USER_PASSWORD;
  config.database_url = DATABASE_URL;

  Firebase.reconnectWiFi(true);
  firebaseData.setResponseSize(4096);
  
  config.token_status_callback = tokenStatusCallback;
  config.max_token_generation_retry = 5;
  Firebase.begin(&config, &auth);    

  Serial.println("Mendapatkan User UID");
  while((auth.token.uid) == ""){
    Serial.print(".");
    delay(1000);
  }
  uid = auth.token.uid.c_str();
  Serial.print("User UID : ");
  Serial.print(uid);
  databasePath = "/UsersData" + uid;
}

void initJson(){
  jadwalPagiJson.add("cekOmbak", "masuk");
}

void setup(){
  Serial.begin(115200);

  initWiFi();
     
  Serial.println("MULAI PROGRAM ....");  
  
  initFirebase();
  initJson();
}

void loop(){
   if(Firebase.isTokenExpired()){
    Firebase.refreshToken(&config);
    Serial.println("Memperbarui Token");
   }

   else if (Firebase.ready() && (millis() - sendDataPrevMillis > timerDelay || sendDataPrevMillis == 0)){
     sendDataPrevMillis = millis();

      jadwalPagiNode = databasePath + "/jadwalPagi";

      // Write an Int number on the database path test/int
      if (Firebase.RTDB.setJSON(&firebaseData, jadwalPagiNode.c_str(), &jadwalPagiJson)){
        Serial.println("PASSED");
        Serial.println("PATH: " + firebaseData.dataPath());
        Serial.println("TYPE: " + firebaseData.dataType());
      }
      else {
        Serial.println("FAILED");
        Serial.println("REASON: " + firebaseData.errorReason());
      }          
  }
}

mobizt, 2024-04-01:
@archee29 

You should check

1. The sign in credentials (email and password)
2. The database security rules and set it to default

```
{
  "rules": {
    ".read": "auth != null", 
    ".write": "auth != null"
  }
}
```

3. Your usage quota limit