mobizt/Firebase-ESP-Client #559: Getting compile error with latest Firebase-ESP-client, older version was okat

joopheuvel, 2023-09-11:
Just implemented latest Firebase-ESP-client library on Arduino IDE 221 with
an Olimex-ESP32-EVB board. 4.3.20 compiled okay, latest one 4.4.1 gives an error:

c:\Users\...OneDrive\Arduino\libraries\Firebase_Arduino_Client_Library_for_ESP8266_and_ESP32\src/client/WiFiClientImpl.h: In member function 'int WiFiClientImpl::tcpPeek()':
c:\Users\...OneDrive\Arduino\libraries\Firebase_Arduino_Client_Library_for_ESP8266_and_ESP32\src/client/WiFiClientImpl.h:414:47: error: suggest parentheses around '&&' within '||' [-Werror=parentheses]
         if (!_rxBuff || _fillPos == _fillSize && !fillRxBuffer())
                                               ^
cc1plus.exe: some warnings being treated as errors

Can this be fixed? I gues I can try to find the relevant .h file and edit, but others may also have this issue, so would be good to change repository. 
Thanks.
Joop van den Heuvel

mobizt, 2023-09-11:
It's just compiler warning and peek() or tcpPeek() are not used.

mobizt, 2023-09-11:
Compiler warning cannot turn into error.

closed 2023-09-11 by mobizt

joopheuvel, 2023-09-11:
Thanks.
Indeed, in my case the compilation stops as this one is treated as error not to be ignored, so cannot generate code.
In meantime I put brackets around in the .h file.
Thanks.
Joop

mobizt, 2023-09-11:
This is GCC compiler issue that treats the warning of unused code (never executed) as error.

In Arduino IDE Preferences, Compiler warnings should set to default.

I will fix this in the next update.

mobizt, 2023-09-12:
Now the library was updated with issue fixed.