mobizt/Firebase-ESP-Client #561: "Connection Refused" After a few successful uploads to RTDB

Ripwords, 2023-09-13:
I've tried the examples and have had no issues for the past few months, however recently when running the code, the connection would start getting refused after uploading data for a few attempts and i start getting "connection refused" error.

```c++
#include <SPI.h>
#include <Wire.h>
#include <Adafruit_GFX.h>
#include <Adafruit_SSD1306.h>
#include <DHT.h>
#include <Keypad.h>
// WiFiManager
#include <WiFi.h>
#include <WiFiManager.h>
// Firebase
#include <Firebase_ESP_Client.h>
#include <addons/TokenHelper.h>
#include <addons/RTDBHelper.h>
// Timestamp
#include <time.h>
// ESP32
#include "soc/rtc_wdt.h"
// Credentials
#include "creds.h"

// Firebase Config
#define FIREBASE_HOST "embedded-iot-7ef1f-default-rtdb.asia-southeast1.firebasedatabase.app"

// LED Config
#define RED 2
#define GREEN 16

// DHT Config
#define DHTPIN 4
#define DHTTYPE DHT11

// OLED Display Config
#define S_W 128
#define S_H 64
#define OLED_RESET -1

// KEYPAD Config
const byte ROWS = 4; 
const byte COLS = 4; 

const char hexaKeys[ROWS][COLS] = {
  {'1', '2', '3', 'A'},
  {'4', '5', '6', 'B'},
  {'7', '8', '9', 'C'},
  {'*', '0', '#', 'D'}
};

byte rowPins[ROWS] = {16, 17, 5, 18}; 
byte colPins[COLS] = {32, 33, 25, 26}; 

static WiFiManager wm;
static FirebaseData data;
static FirebaseAuth auth;
static FirebaseConfig config;
static DHT dht(DHTPIN, DHTTYPE);
static Adafruit_SSD1306 display(S_W, S_H, &Wire, OLED_RESET);
static Keypad kp = Keypad(makeKeymap(hexaKeys), rowPins, colPins, ROWS, COLS); 

const unsigned char wifiicon[] PROGMEM  ={ // wifi icon
  0x00, 0xff, 0x00, 0x7e, 0x00, 0x18,0x00, 0x00
};

static volatile unsigned long prev = 0;
static volatile unsigned long prev_2 = 0;
static unsigned long deletePrev = 0;
static unsigned long deleteInterval = 10;
static volatile unsigned long tempLogInterval = 5;
[... 266 lines cut ...]
}

void setFBIndex() {
  if (Firebase.ready()) {
    if (!indexing) {
      Serial.print("Setting query index in database rules... ");
      // If sign in with OAuth2.0 token, the database secret pass to this function can be empty string
      if (Firebase.RTDB.setQueryIndex(&data, String(ACC) + "/log", "ts", DB_SECRET)) {
        Serial.println("ok");
        indexing = true;
      }
      else {
        Serial.println(data.errorReason());
      }
    }
  }
}
```

 
[image]

As you can see in this image, the data is logged and then suddenly the connection starts getting refused. There is a Failed (DHT) error but that's my sensor error unrelated to the Firebase logic. 

Thanks in advance, let me know if there's any information more that you require. I am using an ESP32 DOIT Devkit

Ripwords, 2023-09-13:
Just a note: I have credentials saved in a different file, DB_SECRET, API_KEY and such, so that's not the problem as the data is uploaded successfully for a few attempts, before connection refused starts

mobizt, 2023-09-13:
First, update the library.

I see that you are using Async function, please avoid using it if you are using it together with other functions as I already noted in the example comment (at the bottom of basic example).

The library data transfer is fast enough which async is not needed.

Ripwords, 2023-09-13:
I've updated to the latest version uploaded a few hours ago, and changed the WiFi network to my mobile network and also removed use of async. The behaviour has improved, but after awhile it still returns me "connection refused" i've already set the BSSL size to maximum as well

mobizt, 2023-09-13:
It is your network issue that resist for internet access.

mobizt, 2023-09-13:
It's not necessary to set SSL buffer size as much, 2048 is ok for general use.

I see you reserved the stack for some FreeRTOS task as much as 36k which is not necessary too.

You should debug your code which memory allocation failure can be a problem in your case.

Note that, library was intensively test 24/7 with WiFI, Ethernet and GSM network and it works fine.

closed 2023-09-13 by mobizt

Ripwords, 2023-09-13:
Okay thanks for the suggestions, I will try to debug

Ripwords, 2023-09-13:
[170621][E][WiFiClientImpl.h:468] tcpWrite(): fail on fd 48, errno: 104, "Connection reset by peer"

I turned on Core Debuggin and received this error message

mobizt, 2023-09-13:
That is because server denied to connect or your network interruption.

Another possible issue, when you use WiFiManger to control your WiFi, you should set reconnect WiFi option to false to prevent network reconnection conflicts.

`Firebase.reconnectWiFi(false);`

Ripwords, 2023-09-13:
I think that may be my problem, thanks again, i have also turned on TCP fbdo.keepAlive not sure if that will help

Ripwords, 2023-09-13:
This seems to have solved my issue for now. Thanks again for your help

mobizt, 2023-09-13:
Thanks for feedback and I will update the document too.