mobizt/Firebase-ESP-Client #568: How to check if a directory exists in Firebase Storage?T

Frans400, 2023-09-17:
Is there anyway to check if a file exits in the Firebase storage cloud?

something like:  Firebase.Storage.doesFileExits(&fbdo, STORAGE_BUCKET_ID, "firmware.bin");

any other alternative to check if a file exits will also help me

thank Mosbizt and all other contributors for this super great library

mobizt, 2023-09-17:
There is no folder or file in Firebase Storage and Google Cloud Storage unless objects.

Some valid characters in the path to object, can be use as directory separator but no physical directory created.

The console folder creation and deletion are only for GUI.

mobizt, 2023-09-17:
You can only [list the objects](https://github.com/mobizt/Firebase-ESP-Client/blob/main/examples/Storage/FirebaseStorage/ListFiles/ListFiles.ino) (known as file) and parse the response for folder from `/ `separator.

closed 2023-09-17 by mobizt