mobizt/Firebase-ESP-Client #569: connection refused

Pathornnnnn, 2023-09-20:
thai 
เริ่มต้นจากผมลองใช้ไลบรารี่นี้ กับ esp32 พอเริ่มทำก็ทำงานตามปกติแต่พอผ่านไป 1 นาที Serial port ค้างจากนั้นก็แสดง connection refused

eng (google translate)
To begin with, I tried using this library with esp32. When I started doing it, it worked normally, but after 1 minute the Serial port froze and then showed connection refused.

thai
เบื้องต้นผมศึกษามาเกี่ยว connection refused ถ้าเข้าใจไม่ผิดมันเกี่ยวกับการเชื่อมต่อ internet แต่ผมลองดูแล้วก็ไม่น่าใช่ หรืออาจเป็นเพราะผมใช้ firebasedata object เยอะไปรึเปล่า ฝากพี่ๆช่วยผมหรือแนะนำผมหน่อยครับ T_T

eng (google translate)
Initially, I studied about connection refused. If I'm not mistaken, it's related to internet connection. But I tried it and it doesn't seem right. Or maybe it's because I use too many firebasedata objects? Please brothers help me. T_T

code

#include <Arduino.h>
#if defined(ESP32)
  #include <WiFi.h>
#elif defined(ESP8266)
  #include <ESP8266WiFi.h>
#endif
#include <Firebase_ESP_Client.h>
//Provide the token generation process info.
#include "addons/TokenHelper.h"
//Provide the RTDB payload printing info and other helper functions.
#include "addons/RTDBHelper.h"

// Insert your network credentials
#define WIFI_SSID "xxx"
#define WIFI_PASSWORD "xxx"

// Insert Firebase project API Key
#define API_KEY "xxx"

// Insert RTDB URLefine the RTDB URL */
#define DATABASE_URL "xxx" 

//Define Firebase Data object
FirebaseData fbdo;

FirebaseAuth auth;
FirebaseConfig config;

unsigned long sendDataPrevMillis = 0;
int intValue;
float floatValue;
bool signupOK = false;
int load1=1, load2=1, load3=1, load4=1, Power;
#define Relay1  14 //GPIO 16
#define Relay2  33 //GPIO 5
#define Relay3  26 //GPIO 4
#define Relay4  27 //GPIO 0
void updateValue(){
  if (Firebase.RTDB.getInt(&fbdo, "/devices/food")) {
      if (fbdo.dataType() == "int") {
        load1 = fbdo.intData();
        Serial.println("Food : "+String(load1));
      }
    }
    else {
      Serial.println(fbdo.errorReason());
    }
    
      if (Firebase.RTDB.getInt(&fbdo, "/devices/water")) {
      if (fbdo.dataType() == "int") {
        load2 = fbdo.intData();
        Serial.println("water : "+String(load2));
      }
    }
    else {
      Serial.println(fbdo.errorReason());
    }
    if (Firebase.RTDB.getInt(&fbdo, "/devices/fanl")) {
      if (fbdo.dataType() == "int") {
        load3 = fbdo.intData();
        Serial.println("fanl : "+String(load3));
      }
    }
    else {
      Serial.println(fbdo.errorReason());
    }
    
    if (Firebase.RTDB.getInt(&fbdo, "/devices/fanr")) {
      if (fbdo.dataType() == "int") {
        load4 = fbdo.intData();
        Serial.println("fanr : "+String(load4));
      }
    }
    else {
      Serial.println(fbdo.errorReason());
    }
}
void Relays(){  
  digitalWrite(Relay1, load1);  
  digitalWrite(Relay2, load2);
  digitalWrite(Relay3, load3);
  digitalWrite(Relay4, load4);
}

void sendValue(){
  
  if (Firebase.RTDB.setInt(&fbdo, "foodLevel", random(1,100))){
      Serial.println("PASSED");
      Serial.println("PATH: " + fbdo.dataPath());
      Serial.println("TYPE: " + fbdo.dataType());
    }
    else {
      Serial.println("FAILED");
      Serial.println("REASON: " + fbdo.errorReason());
    }
    
    if (Firebase.RTDB.setInt(&fbdo, "waterLevel", random(1,100))){
      Serial.println("PASSED");
      Serial.println("PATH: " + fbdo.dataPath());
      Serial.println("TYPE: " + fbdo.dataType());
    }
    else {
      Serial.println("FAILED");
      Serial.println("REASON: " + fbdo.errorReason());
    }
    
    if (Firebase.RTDB.setInt(&fbdo, "humidity", random(1,100))){
      Serial.println("PASSED");
      Serial.println("PATH: " + fbdo.dataPath());
      Serial.println("TYPE: " + fbdo.dataType());
    }
    else {
      Serial.println("FAILED");
      Serial.println("REASON: " + fbdo.errorReason());
    }
    
    if (Firebase.RTDB.setInt(&fbdo, "temperature", random(1,100))){
      Serial.println("PASSED");
      Serial.println("PATH: " + fbdo.dataPath());
      Serial.println("TYPE: " + fbdo.dataType());
    }
    else {
      Serial.println("FAILED");
      Serial.println("REASON: " + fbdo.errorReason());
    }
}

void autoMode(){
  if (Firebase.RTDB.getInt(&fbdo, "/devices/AutoMode")) {
    if (fbdo.dataType() == "int") {
        if(fbdo.intData() == 1){
          Serial.println("Auto Mode : Work");
        }
        else{
          Serial.println("Auto Mode : not Work");
        }
    }
    else {
      Serial.println(fbdo.errorReason());
    }
  }
}

void checkwifi(){
  if(WiFi.status() != WL_CONNECTED){
    WiFi.begin(WIFI_SSID, WIFI_PASSWORD);
    Serial.print("Connecting to Wi-Fi");
    while (WiFi.status() != WL_CONNECTED) {
      Serial.print(".");
      delay(300);
    }
  Serial.println();
  Serial.print("Connected with IP: ");
  Serial.println(WiFi.localIP());
  Serial.println();
  }
}

void WiFiStationConnected(WiFiEvent_t event, WiFiEventInfo_t info){
  Serial.println("Connected to AP successfully!");
}

void WiFiGotIP(WiFiEvent_t event, WiFiEventInfo_t info){
  Serial.println("WiFi connected");
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());
}

void WiFiStationDisconnected(WiFiEvent_t event, WiFiEventInfo_t info){
  Serial.println("Disconnected from WiFi access point");
  Serial.print("WiFi lost connection. Reason: ");
  Serial.println(info.wifi_sta_disconnected.reason);
  Serial.println("Trying to Reconnect");
  WiFi.begin(WIFI_SSID, WIFI_PASSWORD);
}

void setup() {
  Serial.begin(115200);
  pinMode(Relay1, OUTPUT);
  pinMode(Relay2, OUTPUT);
  pinMode(Relay3, OUTPUT);
  pinMode(Relay4, OUTPUT);
  WiFi.disconnect(true);

  delay(1000);

  WiFi.onEvent(WiFiStationConnected, WiFiEvent_t::ARDUINO_EVENT_WIFI_STA_CONNECTED);
  WiFi.onEvent(WiFiGotIP, WiFiEvent_t::ARDUINO_EVENT_WIFI_STA_GOT_IP);
  WiFi.onEvent(WiFiStationDisconnected, WiFiEvent_t::ARDUINO_EVENT_WIFI_STA_DISCONNECTED);
  WiFi.begin(WIFI_SSID, WIFI_PASSWORD);
  Serial.print("Connecting to Wi-Fi");
  while (WiFi.status() != WL_CONNECTED) {
    Serial.print(".");
    delay(300);
  }
  Serial.println();
  Serial.print("Connected with IP: ");
  Serial.println(WiFi.localIP());
  Serial.println();

  /* Assign the api key (required) */
  config.api_key = API_KEY;

  /* Assign the RTDB URL (required) */
  config.database_url = DATABASE_URL;

  /* Sign up */
  if (Firebase.signUp(&config, &auth, "", "")) {  
    Serial.println("ok");
    signupOK = true;
  }
  else {
    Serial.printf("%s\n", config.signer.signupError.message.c_str());
  }
  /* Assign the callback function for the long running token generation task */
  config.token_status_callback = tokenStatusCallback; //see addons/TokenHelper.h

  Firebase.begin(&config, &auth);
  Firebase.reconnectWiFi(true);

}

void loop() {
  if (Firebase.ready() && signupOK && (millis() - sendDataPrevMillis > 3000 || sendDataPrevMillis == 0)) {
    sendDataPrevMillis = millis();
    //sendValue();
    updateValue();
    autoMode();
    Relays();
    checkwifi();
  }
}

mobizt, 2023-09-20:
The `connection refused` means that server does not respond to the request in time.

In your case the problem is internet access or network connection was interrupted.

When you manage to connect to network based on WiFi events, like this.

`WiFi.onEvent(WiFiStationDisconnected, WiFiEvent_t::ARDUINO_EVENT_WIFI_STA_DISCONNECTED);`

It's not necessary to use this in your code. 

`Firebase.reconnectWiFi(true);`

The operation behind this library setting is to handle network reconnection for your device and it causes the reconnection conflicts.

closed 2023-09-20 by mobizt

Pathornnnnn, 2023-09-21:
now i try new code is comment WiFi.onEvent and use Firebase.reconnectWiFi(true) 

in esp32 is not still same problem first can read value from RTDB and after 1 min esp32 froze after froze 1 min serial monitor show   connection refused 

but in esp8266 i try same new code but it's work can work over 1 min

Do you have any advice?

mobizt, 2023-09-21:
The problem is in your code that uses WiFi library incorrectly.

You should remove `WiFi.begin(WIFI_SSID, WIFI_PASSWORD);` from the function `WiFiStationDisconnected`.

ESP32 WiFi library and ESP8266 WiFi library are different and may not work the same for all functionalities.

Pathornnnnn, 2023-09-21:
CODED

/*
  Rui Santos
  Complete project details at our blog.
    - ESP32: https://RandomNerdTutorials.com/esp32-firebase-realtime-database/
    - ESP8266: https://RandomNerdTutorials.com/esp8266-nodemcu-firebase-realtime-database/
  Permission is hereby granted, free of charge, to any person obtaining a copy of this software and associated documentation files.
  The above copyright notice and this permission notice shall be included in all copies or substantial portions of the Software.
  Based in the RTDB Basic Example by Firebase-ESP-Client library by mobizt
  https://github.com/mobizt/Firebase-ESP-Client/blob/main/examples/RTDB/Basic/Basic.ino
*/

#include <Arduino.h>
#if defined(ESP32)
  #include <WiFi.h>
#elif defined(ESP8266)
  #include <ESP8266WiFi.h>
#endif
#include <Firebase_ESP_Client.h>
//Provide the token generation process info.
#include "addons/TokenHelper.h"
//Provide the RTDB payload printing info and other helper functions.
#include "addons/RTDBHelper.h"

// Insert your network credentials
#define WIFI_SSID "xxx"
#define WIFI_PASSWORD "xxx"

// Insert Firebase project API Key
#define API_KEY "xxx"

// Insert RTDB URLefine the RTDB URL */
#define DATABASE_URL "xxx" 

//Define Firebase Data object
FirebaseData fbdo;

FirebaseAuth auth;
FirebaseConfig config;
int i = 1;
unsigned long sendDataPrevMillis = 0;
int intValue;
float floatValue;
bool signupOK = false;
int load1=1, load2=1, load3=1, load4=1, Power;
#define Relay1  14 
#define Relay2  33 
#define Relay3  26
#define Relay4  27 
void updateValue(){
  if (Firebase.RTDB.getInt(&fbdo, "/devices/food")) {
      if (fbdo.dataType() == "int") {
        load1 = fbdo.intData();
        Serial.println("Food : "+String(load1));
      }
    }
    else {
      Serial.println(fbdo.errorReason());
    }
    
      if (Firebase.RTDB.getInt(&fbdo, "/devices/water")) {
      if (fbdo.dataType() == "int") {
        load2 = fbdo.intData();
        Serial.println("water : "+String(load2));
      }
    }
    else {
      Serial.println(fbdo.errorReason());
    }
    if (Firebase.RTDB.getInt(&fbdo, "/devices/fanl")) {
      if (fbdo.dataType() == "int") {
        load3 = fbdo.intData();
        Serial.println("fanl : "+String(load3));
      }
    }
    else {
      Serial.println(fbdo.errorReason());
    }
    
    if (Firebase.RTDB.getInt(&fbdo, "/devices/fanr")) {
      if (fbdo.dataType() == "int") {
        load4 = fbdo.intData();
        Serial.println("fanr : "+String(load4));
      }
    }
    else {
      Serial.println(fbdo.errorReason());
    }
}
void Relays(){  
  digitalWrite(Relay1, load1);  
  digitalWrite(Relay2, load2);
  digitalWrite(Relay3, load3);
  digitalWrite(Relay4, load4);
}

void sendValue(){
  
  if (Firebase.RTDB.setInt(&fbdo, "foodLevel", random(1,100))){
      Serial.println("PASSED");
      Serial.println("PATH: " + fbdo.dataPath());
      Serial.println("TYPE: " + fbdo.dataType());
    }
    else {
      Serial.println("FAILED");
      Serial.println("REASON: " + fbdo.errorReason());
    }
    
    if (Firebase.RTDB.setInt(&fbdo, "waterLevel", random(1,100))){
      Serial.println("PASSED");
      Serial.println("PATH: " + fbdo.dataPath());
      Serial.println("TYPE: " + fbdo.dataType());
    }
    else {
      Serial.println("FAILED");
      Serial.println("REASON: " + fbdo.errorReason());
    }
    
    if (Firebase.RTDB.setInt(&fbdo, "humidity", random(1,100))){
      Serial.println("PASSED");
      Serial.println("PATH: " + fbdo.dataPath());
      Serial.println("TYPE: " + fbdo.dataType());
    }
    else {
      Serial.println("FAILED");
      Serial.println("REASON: " + fbdo.errorReason());
    }
    
    if (Firebase.RTDB.setInt(&fbdo, "temperature", random(1,100))){
      Serial.println("PASSED");
      Serial.println("PATH: " + fbdo.dataPath());
      Serial.println("TYPE: " + fbdo.dataType());
    }
    else {
      Serial.println("FAILED");
      Serial.println("REASON: " + fbdo.errorReason());
    }
}

void autoMode(){
  if (Firebase.RTDB.getInt(&fbdo, "/devices/AutoMode")) {
    if (fbdo.dataType() == "int") {
        if(fbdo.intData() == 1){
          Serial.println("Auto Mode : Work ");
        }
        else{
          Serial.println("Auto Mode : not Work");
        }
    }
    else {
      Serial.println(fbdo.errorReason());
    }
  }
}

void setup() {
  Serial.begin(115200);
  pinMode(Relay1, OUTPUT);
  pinMode(Relay2, OUTPUT);
  pinMode(Relay3, OUTPUT);
  pinMode(Relay4, OUTPUT);
  WiFi.disconnect(true);

  delay(1000);

  WiFi.begin(WIFI_SSID, WIFI_PASSWORD);
  Serial.print("Connecting to Wi-Fi");
  while (WiFi.status() != WL_CONNECTED) {
    Serial.print(".");
    delay(300);
  }
  Serial.println();
  Serial.print("Connected with IP: ");
  Serial.println(WiFi.localIP());
  Serial.println();

  /* Assign the api key (required) */
  config.api_key = API_KEY;

  /* Assign the RTDB URL (required) */
  config.database_url = DATABASE_URL;

  /* Sign up */
  if (Firebase.signUp(&config, &auth, "", "")) {  
    Serial.println("ok");
    signupOK = true;
  }
  else {
    Serial.printf("%s\n", config.signer.signupError.message.c_str());
  }
  /* Assign the callback function for the long running token generation task */
  config.token_status_callback = tokenStatusCallback; //see addons/TokenHelper.h

  Firebase.begin(&config, &auth);
  Firebase.reconnectWiFi(true);

}

void loop() {
  if (Firebase.ready() && signupOK && (millis() - sendDataPrevMillis > 3000 || sendDataPrevMillis == 0)) {
    sendDataPrevMillis = millis();
    //sendValue();
    updateValue();
    autoMode();
    Relays();
    Serial.println(String(i));
    i+=1;
  }
}

now i try add i variable to indicator cycle read data from RTDB after test 3 time i see esp32 froze every i = 20  after that froze and connection refused
 i try change wifi to hotspot it's still same problem.

Pathornnnnn, 2023-09-21:
Ok now i can solve this problem i solve this by downgrade libary to 2.3.7 it work with esp32. Thank you so much kub.

mobizt, 2023-09-21:
It's your own risk to used deprecated and known bugs version.