mobizt/Firebase-ESP-Client #587: response payload read timed out && connection refused

MeteOnen, 2023-10-17:
Arduino Version 1.8.19
esp32 Board Version 2.0.14
Firebase-ESP-Client Version 4.4.8

My code
```void setup(){
  Serial.begin(115200);
  WiFi.disconnect(true);
  WiFi.begin(ssid, password);
  Serial.print("Connecting");
  while (WiFi.status() != WL_CONNECTED) {
    Serial.print(".");
    delay(500);
  }
  Serial.println("");
  Serial.println("Wi-Fi connected");
  Serial.print("IP Address : ");
  Serial.println(WiFi.localIP());
  config.api_key = API_KEY;
  auth.user.email = USER_EMAIL;
  auth.user.password = USER_PASSWORD;
  config.database_url = DATABASE_URL;
  config.timeout.serverResponse = 15 * 1000;
  if (Firebase.signUp(&config, &auth, "", "")) {
    Serial.println("Firebase connected.");
  }
  else {
    Serial.printf("Error: %s\n", config.signer.signupError.message.c_str());
  }
  config.token_status_callback = tokenStatusCallback;
  fbdo.setBSSLBufferSize(4096, 1024);
  fbdo.setResponseSize(2048);
  Firebase.begin(&config, &auth);
  Firebase.reconnectWiFi(true);
  Firebase.RTDB.setMaxRetry(&fbdo, 3);
  Firebase.RTDB.setMaxErrorQueue(&fbdo, 10);
}

void loop(){
  if (Firebase.ready() && (millis() - getDataPrevMillis > 5000 || getDataPrevMillis == 0)) {
    getDataPrevMillis = millis();
    if (Firebase.RTDB.getInt(&fbdo, "/r")) {
      reddata = fbdo.to<int>();
      Serial.print("Red data: ");
      Serial.println(reddata);
    }
    else {
      Serial.print("Red error: ");
      Serial.println(fbdo.errorReason());
    }
    delay(1000);
    if (Firebase.RTDB.getInt(&fbdo, "/g")) {
      greendata = fbdo.to<int>();
      Serial.print("Green data: ");
      Serial.println(greendata);
    }
    else {
      Serial.print("Green error: ");
      Serial.println(fbdo.errorReason());
    }
    delay(1000);
    if (Firebase.RTDB.getInt(&fbdo, "/b")) {
      bluedata = fbdo.to<int>();
      Serial.print("Blue data: ");
      Serial.println(bluedata);
    }
    else {
      Serial.print("Blue error: ");
      Serial.println(fbdo.errorReason());
    }
  }
}
```

Coming to the error I encountered. First of all, the code works correctly. It gives the information in rtdb for a while. After a while, I get the "connection refused" error. I added a delay in between. I increased the time. I added the timeout,setMaxRetry,setMaxErrorQueue codes. It didn't get better again. Sometimes it gives a "response payload read timed out" error. What is the mistake I made? How should I solve this problem?

mobizt, 2023-10-17:
Data at the path you want to get is large and you need to increase `BearSSL` rx buffer.
`fbdo.setBSSLBufferSize(4096, 1024);`

MeteOnen, 2023-10-17:
I use exactly the same. You can see it in my codes.

mobizt, 2023-10-17:
I know and post your line of code where to set, and I recommend increasing it.

mobizt, 2023-10-17:
I see that you only get int value which increasing the `BearSSL` buffer is not necessary.

The internet condition is the issue in your case.

MeteOnen, 2023-10-17:
12:00:07.233 -> Firebase connected.
12:02:46.708 -> Red error: connection refused

After 2 minutes, it send a "connection refused" error every second.

MeteOnen, 2023-10-17:
What kind of problem could there be with the internet? I tried this with 2 separate 100mbit modems and my phone's mobile internet. The result did not change.

mobizt, 2023-10-17:
It's not about your internet package but your internet condition that including networking devices.

I can't help on this user specific condition issue which all devices work fine on my testing.