mobizt/Firebase-ESP-Client #591: Token error: code: -1, message: Missing required parameter: assertion` on AUTH 2.0

saadmustafa12, 2023-11-03:
**Describe the bug**
Hey Mobizt!

I am back again with new error.  I have started facing this error recently whenever I let device run for like 12 - 24 hours non stop. After this error it just stuck into infinite loop.

`10:36:15.757 -> Token info: type = OAuth2.0 access token, status = on request
10:36:16.140 -> Token info: type = OAuth2.0 access token, status = error
10:36:16.140 -> Token error: code: -1, message: Missing required parameter: assertion`

**To Reproduce**
I am using default method of AUTH 2.0. It was working fine before but recently it started causing this problem.

**Expected behavior**
It should work without problem and keep running smoothly for as much long time as device is up. In case of infinite loop device should reboot for automatic repair and start working again.

**Screenshots**
[image]

**IDE and its version:**
 - Arduino,
 - Version 2.2.1

mobizt, 2023-11-03:
First, you should update ESP32/ESP8266 Arduino Core SDK and library.

Config and Auth object should be defined globally as in the examples.

There is some usage scenario that you did not tell that causes the invalid time which finally give you the error like this.
 
If you set your device to sleep in ESP32 and woke up, or your device time was changed after calling begin, you should call these to reset the authentication.
https://github.com/mobizt/Firebase-ESP-Client/blob/3dacb1f2ffc2f42ba0ef12edf1fd9d5cba8b2253/examples/Authentications/ReAuthenticate/ReAuthenticate.ino#L88-L91

In case of ESP32 deep sleep, the time zone environment lost (bug in Core), you have to sync your device time with NTP server again.

closed 2023-11-03 by mobizt

saadmustafa12, 2023-11-03:
I was updating system time every 24 hours with Firebase Timestamp. I have removed that function and now time only sets when device reboots. 

I have no deep sleep mode. Let me check and get back to you if problem presists.

Thank you!