mobizt/Firebase-ESP-Client #596: No more processes

sudilacf, 2023-11-18:
`tcpWrite(): fail on fd 59, errno: 11, "No more processes`

how to fix this problem?

mobizt, 2023-11-18:
You should create minimal code that reproduces the issue.

The issue is about the bugs in ESP32 WiFiClient core library when the session was reused while the connection was termonated by server.

1. You should not use async functions.
2. Firebase closes the session in 50 seconds even the session was kept alive for reuse. Then call `fbdo.clear()` after the session was opened (server connected) longer than 50 seconds.

closed 2023-11-18 by mobizt

sudilacf, 2023-11-18:
so I need to call fbdo.clear() after using get/set function?

mobizt, 2023-11-18:
No, it's not necessary, it depends on your code which `FirebaseData` object may be used incorrectly, and I suggest based on the error.

If `FirebaseData` object used for stream with callback, you should not reuse it for other functions.
If you use, xxxAsync, please avoid using it.

sudilacf, 2023-11-18:
Im not using it this is my code

#include <Arduino.h>
#include <LiquidCrystal_I2C.h>
#include <ESP32Servo.h>
#include <WiFi.h>
#include "functions.h"

// Pins number
const int SERVO_PIN = 13;
const int BUZZ_PIN = 33;
const int BTN1_PIN = 2;
const int BTN2_PIN = 4;
const int BTN3_PIN = 0;
const int R_PIN = 27;
const int G_PIN = 14;
const int B_PIN = 12;

const char *FIREBASE_URL = "https://b1g-1a44a-default-rtdb.asia-southeast1.firebasedatabase.app";
const char *FIREBASE_SECRETS = "OlEFCS6pWzkR1RsmcND51vr9L8GfkU4ZK3WRr8IZ";
const int GMT_OFFSET_SECONDS = 3600 * 8;
const int GMT_DAYLIGHT_OFFSET_SECONDS = 0;
const char *NTP_SERVER = "pool.ntp.org";
const char *NTP_SERVER2 = "time.nist.gov";

bool disconnected = true;
bool isReady = false;
bool hasSetup = false;
bool networkSetup = false;
bool isPressed = false; 
unsigned long pressStartTime;
bool firebaseInit = false;
bool streamStarted = false;
bool isFeeding = false;
bool feedingDisplayed = false;
int scheduleOffset;
String scheduleOn;

AsyncWebServer server(80);
LiquidCrystal_I2C lcd(0x27, 16, 2);
Servo servo;

FirebaseData fbdo;
FirebaseData feedStream;
FirebaseData settingsStream;
FirebaseAuth auth;
FirebaseConfig config;
FirebaseJson json;
FirebaseJsonData result;

void setup() {
  
  Serial.begin(115200);
  
  // Initialize pins
  pinMode(BTN1_PIN, INPUT_PULLDOWN);
  pinMode(BTN2_PIN, INPUT_PULLDOWN);
  pinMode(BTN3_PIN, INPUT_PULLUP);
  pinMode(BUZZ_PIN, OUTPUT);
  pinMode(R_PIN, OUTPUT);
  pinMode(G_PIN, OUTPUT);
  pinMode(B_PIN, OUTPUT);
  pinMode(SERVO_PIN, OUTPUT);
  servo.attach(SERVO_PIN);

  // Initialize lcd
  lcd.init();
  lcd.backlight();
  lcd.print("Starting...");
  delay(2000);
[... 399 lines cut ...]
      lcd.print("No data to show");

    }

  }

}

void feed() {

  digitalWrite(R_PIN, LOW);
  digitalWrite(G_PIN, LOW);
  digitalWrite(B_PIN, HIGH);
  isFeeding = true;
  feedingDisplayed = false;
  servo.write(0);
  delay(3000);
  servo.write(90);
  lcd.clear();
  isFeeding = false;
  digitalWrite(R_PIN, LOW);
  digitalWrite(G_PIN, HIGH);
  digitalWrite(B_PIN, LOW);

}

mobizt, 2023-11-18:
Minimal?

sudilacf, 2023-11-18:
> Minimal?

Im using 3 FirebaseData object two for MultiStream and one fbdo

```
void feedStreamCallback(MultiPathStream data) {

  Serial.printf("Feed:\nLength: %d\nMax: %d\n", data.payloadLength(), data.maxPayloadLength());

  if(data.get("/now")) {

    if(data.value.equals("true")) {

      if(Firebase.RTDB.setBool(&fbdo, "/feed/now", false)) {

        if(addToHistory(&fbdo)) {

          if(Firebase.RTDB.setString(&fbdo, "/settings/push_key", fbdo.pushName())) {

            feed();

          }

        }

      }

    }

  }
  
  if(data.get("/offset")) {

    scheduleOffset = data.value.toInt();

  }
  
  if(data.get("/on")) {

    scheduleOn = data.value;

  }

}

void settingsStreamCallback(MultiPathStream data) {

  Serial.printf("Settings:\nLength: %d\nMax: %d\n", data.payloadLength(), data.maxPayloadLength());

  if(data.get("/push_key")) {

    if(Firebase.RTDB.getJSON(&fbdo, "/history/" + data.value)) {

      json = fbdo.jsonObject();
        
    } else {

      lcd.clear();
      lcd.print("Last fed:");
      lcd.setCursor(0, 1);
      lcd.print("No data to show");

    }

  }

}
```

mobizt, 2023-11-18:
No, minimal does not mean the fraction of code. 

You should create minimal, clean without third party libraries but complete that reproduces the issue and post here instead of your code.

I can't verify or debug your complex code.

mobizt, 2023-11-18:
Did you know this will cause memory leaks?

`Firebase.RTDB.setString(new FirebaseData()...`

When you dynamically allocated the `FirebaseData` with `new` without using `delete`.

mobizt, 2023-11-18:
The `FirebaseData` object is not only contains data but SSL Client and its resource which you should defined it locally or globally whenever you used it in usage scope.

sudilacf, 2023-11-18:
this is the summarization of my code:

```
#include <WiFi.h>
#include <TimeLib.h>
#include <WiFiUdp.h>
#include <NTPClient.h>
#include <Firebase_ESP_Client.h>

int scheduleOffset;
String scheduleOn;

FirebaseData fbdo;
FirebaseData feedStream;
FirebaseData settingsStream;
FirebaseAuth auth;
FirebaseConfig config;
FirebaseJson json;
FirebaseJsonData result;

WiFiUDP clientUDP;
NTPClient timeClient(clientUDP, "asia.pool.ntp.org", 3600 * 8);

void feedStreamCallback(MultiPathStream data);
void settingsStreamCallback(MultiPathStream data);
void streamTimeoutCallback(bool);

void setup() {

    Serial.begin(115200);

    WiFi.begin("WIFISSID", "PASSWORD");
    while (WiFi.status() != WL_CONNECTED)
    {
        Serial.print(".");
        delay(500);
    }
    Serial.println("connected");

    config.database_url = "";
    config.signer.tokens.legacy_token = "";

    Firebase.reconnectNetwork(true);
    fbdo.setBSSLBufferSize(2048, 1024);
    feedStream.setBSSLBufferSize(2048, 1024);
    settingsStream.setBSSLBufferSize(2048, 1024);
    Firebase.begin(&config, &auth);
    feedStream.keepAlive(5, 5, 1);
    settingsStream.keepAlive(5, 5, 1);
    
    if(!Firebase.RTDB.beginMultiPathStream(&feedStream, "/feed")) {
        Serial.printf("Failed to begin stream %s\n", feedStream.errorReason().c_str());
        return;
    }

    if(!Firebase.RTDB.beginMultiPathStream(&settingsStream, "/settings")) {
        Serial.printf("Failed to begin stream %s\n", settingsStream.errorReason().c_str());
        return;
    }

    Firebase.RTDB.setMultiPathStreamCallback(&feedStream, feedStreamCallback, streamTimeoutCallback);
    Firebase.RTDB.setMultiPathStreamCallback(&settingsStream, settingsStreamCallback, streamTimeoutCallback);

}

void loop() {

    if (!Firebase.ready())
        return;
      
    timeClient.update();

    json.get(result, "month");
    String monthStr(result.to<int>());
    json.get(result, "day");
    String dayStr(result.to<int>());
    json.get(result, "year");
    String yearStr(result.to<int>());
    json.get(result, "hour");
    String hourStr(result.to<int>());
    json.get(result, "minute");
    String minuteStr(result.to<int>());
    json.get(result, "meridian");
    String meridian(result.to<String>());
    String lastFed = monthStr + "/" + dayStr + "/" + yearStr + " " + hourStr + ":" + minuteStr + ":00 " + meridian;

    if(!scheduleOn.isEmpty() && !scheduleOn.equals(lastFed) && getTimeOffset(0).equals(scheduleOn)) {

        if(addToHistory(&fbdo)) {

            if(Firebase.RTDB.setString(new FirebaseData(), "/settings/push_key", fbdo.pushName())){
            
            if(Firebase.RTDB.setString(new FirebaseData(), "/feed/on", getTimeOffset(scheduleOffset))) {

                feed();

            }

            }
            
        }

    }

}

void feedStreamCallback(MultiPathStream data) {

  Serial.printf("Feed:\nLength: %d\nMax: %d\n", data.payloadLength(), data.maxPayloadLength());

  if(data.get("/now")) {

    if(data.value.equals("true")) {

      if(Firebase.RTDB.setBool(&fbdo, "/feed/now", false)) {

        if(addToHistory(&fbdo)) {

          if(Firebase.RTDB.setString(&fbdo, "/settings/push_key", fbdo.pushName())) {

            feed();

          }

        }

      }

    }

  }
  
  if(data.get("/offset")) {

    scheduleOffset = data.value.toInt();

  }
  
  if(data.get("/on")) {

    scheduleOn = data.value;

  }

}

void settingsStreamCallback(MultiPathStream data) {

  Serial.printf("Settings:\nLength: %d\nMax: %d\n", data.payloadLength(), data.maxPayloadLength());

  if(data.get("/push_key")) {

    if(Firebase.RTDB.getJSON(&fbdo, "/history/" + data.value)) {

      json = fbdo.jsonObject();
        
    }

  }

}

void feed() {

  //rotate servo

}

bool addToHistory(FirebaseData *fbdo) {

    const char *weekday_names[] = {"Sunday", "Monday", "Tuesday", "Wednesday", "Thursday", "Friday", "Saturday"};
    const char *month_names[] = {"January", "February", "March", "April", "May", "June", "July", "August", "September", "October", "November", "December"};
    unsigned long epochTime = timeClient.getEpochTime();
    const char *meridian = isAM(epochTime) ? "AM" : "PM";
    FirebaseJson json;

    json.set("week_day_name", weekday_names[weekday(epochTime) - 1]);
    json.set("year", year(epochTime));
    json.set("month", month(epochTime));
    json.set("month_name", month_names[month(epochTime) - 1]);
    json.set("day", day(epochTime));
    json.set("hour", hourFormat12(epochTime));
    json.set("minute", minute(epochTime));
    json.set("second", second(epochTime));
    json.set("meridian", meridian);
    json.set("epoch_time", epochTime);

    return Firebase.RTDB.pushJSON(fbdo, "/history", &json);

}
```

mobizt, 2023-11-18:
Please read my previous comment, your code causes memory leaks and will not work properly.

sudilacf, 2023-11-18:
> The `FirebaseData` object is not only contains data but SSL Client and its resource which you should defined it locally or globally whenever you used it in usage scope.

which FirebaseData do you mean?

mobizt, 2023-11-18:
Already told in my previous comment.

mobizt, 2023-11-18:
> Did you know this will cause memory leaks?
> 
> `Firebase.RTDB.setString(new FirebaseData()...`
> 
> When you dynamically allocated the `FirebaseData` with `new` without using `delete`.

sudilacf, 2023-11-18:
> > Did you know this will cause memory leaks?
> > `Firebase.RTDB.setString(new FirebaseData()...`
> > When you dynamically allocated the `FirebaseData` with `new` without using `delete`.

Ow I see, I'll try to remove it and test again. thanks

sudilacf, 2023-11-19:
sir I got another problem

`tcpWrite(): fail on fd 51, errno: 128, "Socket is not connected"`

what causes this?

mobizt, 2023-11-19:
The same reason, the connection terminated by server which `WiFiClient` does not handle this properly as a bug in ESP32 `WiFiClient` library.

To be fix, the `WiFiClient` should be closed the connection on its end as server is gone instead of just give the error and still vaiting the data in the infinite loop without closing the session.

Now you should update the library by download the zip file from library repository.

sudilacf, 2023-11-19:
I think I fixed it after removing `Firebase.reconnectNetwork(true)` cuz I have code for internet reconnection.

btw, can you send me the link of WiFiClient lib pls, I cant find it.

mobizt, 2023-11-19:
No, it's internally implemented, and you should update the library.

Anyway, you should follow the example and read the comment first.
https://github.com/mobizt/Firebase-ESP-Client/blob/a3701a04f0d1f82955a75140729100554e5fc0a2/examples/RTDB/Basic/Basic.ino#L118-L119

sudilacf, 2023-11-19:
But I'm using the latest version of library 4.4.8

And yes, already read that.

mobizt, 2023-11-19:
I just update the library and it takes several hours for Arduino Library Manager to update.

In PlatformIO, you can update it now.

sudilacf, 2023-11-19:
Thanks bro