mobizt/Firebase-ESP-Client #598: Non-blocking token request method

import-tiago, 2023-11-19:
I have a project in which the ESP32 can eventually be configured on a Wi-Fi network without internet access (and it's fine for my application);

But, when using Firebase-ESP-Client, the tokenProcessingTask() will freeze the execution of the entire application because it has a while() that infinitely tries to obtain the token.

My suggestion is to add an option like "config.abort_connection_if_token_fail = true" and "config.token_request_attempts = 5" (examples only), to allow use cases like mine to continue working (even if it means being without Firebase services available).

I will modify for my own use, but the official solution may help others.

mobizt, 2023-11-19:
It is about http request-response process which of course it is blocking for a while which is actually done once at a time.

The process utilized the states in a predetermined condition loop instead of infinite loop as you may feel that is why the callback is presented.

The time out is not necessary as long as you can control the execution of Firebase.ready().

FYi, this is not async operation library, the time out was done internally with TCP client and you can change it (see basic example comment).

mobizt, 2023-11-19:
The Async Firebase Client library is under development.

closed 2023-11-19 by mobizt